Koenkk/zigbee2mqtt #24612: Bare metal upgrade failed!

thargy, 2024-11-02:
### What happened?

I attempted to upgrade on bare metal, and followed the instructions [in this discussion](https://github.com/Koenkk/zigbee2mqtt/discussions/24198)

Specifically:
```bash
mv data/configuration.yaml data/configuration.yaml.bak
./update.sh # (update as you usually do)
mv data/configuration.yaml.bak data/configuration.yaml
```

This didn't work as it first required me to enter my git credentials and then attempt a merge, leaving my branch in a bad state so I had to do a hard reset

```bash
git reset --hard origin/master
```

Thankfully, I keep a backup of my `configuration.yaml`, which I restored.  I then had to re-run the update, which succeeded.

However, on starting the service, Z2m came up but all my devices were missing!

The error logs  were full of logs like this for each device:

```
[2024-11-02 21:38:27] error: 	z2m: Cannot find '0x8c65a3fffe55bccf/1' of group 'Kitchen Lightstrips'
[2024-11-02 21:38:27] error: 	z2m: Cannot find endpoint '1' of device '0x8c65a3fffe55bccf'
[2024-11-02 21:38:27] error: 	z2m: Cannot find '0x4c5bb3fffed6c5ae/1' of group 'Kitchen Lightstrips'
[2024-11-02 21:38:27] error: 	z2m: Cannot find endpoint '1' of device '0x4c5bb3fffed6c5ae'
[2024-11-02 21:38:27] error: 	z2m: Cannot find '0xa4c1387da31bc73e/1' of group 'Kitchen Lightstrips'
[2024-11-02 21:38:27] error: 	z2m: Cannot find endpoint '1' of device '0xa4c1387da31bc73e'
...
```

### What did you expect to happen?

I expected the update to work without breaking my install.

### How to reproduce it (minimal and precise)

I followed the published instructions.

### Zigbee2MQTT version

 1.41.0

### Adapter firmware version

 20240710

### Adapter

Sonoff 

### Setup

Pi

thargy, 2024-11-02:
Note, I can see '0x8c65a3fffe55bccf' is defined in the `configuration.yaml`:
```yaml
devices:
...
  "0x8c65a3fffe55bccf":
    friendly_name: Kitchen Bar Lightstrip
...
```

And in the `coordinator_backup.json`:
```json
    {
      "nwk_address": "2708",
      "ieee_address": "8c65a3fffe55bccf",
      "is_child": false,
      "link_key": {
        "key": "e1770df239c2c0f81b1c1addba22f2db",
        "rx_counter": 0,
        "tx_counter": 7978
      }
    },
```

As are all the other devices in the logs (~90 devices).

My `database.db` looks suspiciously empty though:
```json
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0029e8c733","nwkAddr":0,"manufId":7,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{}}
{"id":2,"type":"Group","groupID":1,"members":[],"meta":{}}
{"id":3,"type":"Group","groupID":2,"members":[],"meta":{}}
{"id":4,"type":"Group","groupID":3,"members":[],"meta":{}}
{"id":5,"type":"Group","groupID":4,"members":[],"meta":{}}
{"id":6,"type":"Group","groupID":5,"members":[],"meta":{}}
{"id":7,"type":"Group","groupID":6,"members":[],"meta":{}}
{"id":8,"type":"Group","groupID":7,"members":[],"meta":{}}
{"id":9,"type":"Group","groupID":8,"members":[],"meta":{}}
{"id":10,"type":"Group","groupID":9,"members":[],"meta":{}}
{"id":11,"type":"Group","groupID":10,"members":[],"meta":{}}
{"id":12,"type":"Group","groupID":11,"members":[],"meta":{}}
{"id":13,"type":"Group","groupID":12,"members":[],"meta":{}}
{"id":14,"type":"Group","groupID":13,"members":[],"meta":{}}
{"id":15,"type":"Group","groupID":14,"members":[],"meta":{}}
{"id":16,"type":"Group","groupID":15,"members":[],"meta":{}}
{"id":17,"type":"Group","groupID":16,"members":[],"meta":{}}
{"id":18,"type":"Group","groupID":17,"members":[],"meta":{}}
{"id":19,"type":"Group","groupID":18,"members":[],"meta":{}}
{"id":20,"type":"Group","groupID":19,"members":[],"meta":{}}
{"id":21,"type":"Group","groupID":20,"members":[],"meta":{}}
{"id":22,"type":"Group","groupID":21,"members":[],"meta":{}}
{"id":23,"type":"Group","groupID":22,"members":[],"meta":{}}
{"id":24,"type":"Group","groupID":23,"members":[],"meta":{}}
```

As does `state.json`
```json
{}
```

I'm desperately hoping I don't have to somehow re-add 90+ devices!

thargy, 2024-11-02:
Note, updating my HAOS addon (which has 120 devices) didn't fail.

thargy, 2024-11-02:
Here is a full start-up debug log.

[log.log](https://github.com/user-attachments/files/17608560/log.log)

thargy, 2024-11-02:
I suspect I'm ****ed as I have lost my `database.db` and it cannot be rebuilt.

thargy, 2024-11-03:
So, after many hours of testing and digging into z2m and z2m-herdsman, (all of which is preferrable to having to somehow rebuild my network yet again), I have managed to make a custom extension which runs and checks for missing devices.

The following is my understanding so far:

* The `settings` has the IEEE address and friendly name of each device, as read from the `configuration.yaml`
* The `adapter` has the IEEE address and the associated network address, at least for ZStack which supports backups, this can be seen in `coordinator_backup.json`.
* When a device joins the network, all it needs is the IEEE address, and the network address, which are available from the config, and the coordinator respectively. At this point, the device is interviewed for the rest of the information and the response is stored in `database.db`.
* There is already a process to support re-interviewing a failed device from the UI, or via MQTT (using the 'zigbee2mqtt/bridge/response/device/interview' topic)

Given the above, when the `database.db` is lost, it should be possible to rebuild it by taking the IEEE address and the network address and triggering a re-interview, as if it had just attempted to rejoin the network.

Sadly, due to the way Z2m is architected, and the associated SOC, I've yet to find the right way to safely create the device - something like `device = model_1.Device.create('Unknown', payload.ieeeAddr, payload.networkAddress, undefined, undefined, undefined, undefined, false);`, and trigger an interview.

Before I go too far, I was wondering if you know any reason it might not work?

thargy, 2024-11-03:
Success!! (Well at least partial)

Inspired by the above, I took a somewhat _easier_ approach and wrote a quick python script to parse the configuration file, the `database.db` (which just has empty groups and, importantly, a coordinator) and the coordinator.backup.

It then rebuilds the database.db, populating the items with an 'Unknown' type, and the `IEEEAddr` and `nwkAddr`.  As a bonus, it also updates the group membership.

On first run, I still got all the errors, as before and non of the groups or devices appeared. But then, suddenly all my devices were there!

I could then slowly go through them, clicking on the 'i' in the UI to re-interview them.

I then stopped Z2m and reran the script, re-added the groups, and, this time, Z2m recognised the groups on restart. However, I did notice that the python json serializer encoded some of the special characters differently, this didn't seem to be a problem though.

Here's the script:
```python
import json
import yaml
import shutil

PATH='/opt/zigbee2mqtt/data/'

DB_FILE=PATH+'database.db'
BACKUP_FILE=PATH+'coordinator_backup.json'
CONFIG_FILE=PATH+'configuration.yaml'
DB_BACKUP_FILE=PATH+'database_backup.db'

# Holds database data by relevant type
dbDevices: dict[str, object] = {}
dbGroups: dict[int, object] = {}
coordinator: object = None

# Read database
dbLines: list[str]
with open(DB_FILE, 'r') as dbFile:
    dbLines = dbFile.readlines()

# Parse database.db to find coordinator
print('Loading existing database.db file')
for line in dbLines:
    DBdata:dict[str, object] = json.loads(line)
    if not "type" in DBdata:
        continue
    type = DBdata["type"]
    
    if type == "Group":
        if not "groupID" in DBdata:
            continue
        
        dbGroups[DBdata["groupID"]] = DBdata
        continue
        
    if not "ieeeAddr" in DBdata:
        continue
    ieeeAddr = DBdata["ieeeAddr"]
    
    if type == "Coordinator":
        coordinator = DBdata
    else:        
        dbDevices[ieeeAddr] = DBdata
print(f"Found {'coordinator, ' if coordinator else ''}{str(len(dbDevices))+' devices, ' if len(dbDevices)>0 else ''}{str(len(dbGroups))+' groups' if len(dbGroups)>0 else ''}.")
if not coordinator:
    print("The coordinator was not found in the existing database, aborting! (Try running Z2m first)")
    exit

# Parse coordinator_backup.json to retreive all IEEE present in coordinator and get their current nwkAddr
print('Loading coordinator backup.')
added=0
updated=0
found=0
with open(BACKUP_FILE,"r") as backupfile:
    data = json.loads(backupfile.read())
[... 150 lines cut ...]
        
    # Then unknown groups
    for group in dbGroups.values():
        id = id + 1
        group["id"] = id
        json.dump(group, dbFile)
        dbFile.write('\n')
        
print("Done!")

if lastGoodId < id:
    print(f"Note: all ids above {lastGoodId} are unknown to the configuration and may be removed!")
```

WARNING: I'm not a Python developer, and I wrote this at a fever pitch at 5AM while trying to restore my house lights before the wife woke up. i.e., it's warranted as "Works For Me" and is _not_ recommended.

That said, I tried to make it reasonably safe and robust, but if you run it, update the path constants at the top and _stop z2m first_!

However, if you're reasonably proficient and looking for an alternative to re-pairing 90+ devices (94 in my case), this might save your neck.  Even if all you have is your coordinator backup, it can recreate the database, but your friendly names will be missing!

This proves my hypothesis above. Basically, Z2m has all the information it needs to try automatically rebuilding the `database.db` if it becomes out-of-sync with the coordinator backup.

The tricky bit is handling the re-interviews, which should be done in small batches until successful. That's also how I've coded my OTA updates to work, as I have nearly 200 Philips hue bulbs, and when an update comes out, you have to do it in small batches of 4-5 devices.

Would this be something worth looking at implementing? When z2m notices the config/db/coordinator is out of sync, should it do its best to recover information so long as it has IEEE and Network addresses?  Looking through the issues on here and the herdsman, I'm certainly not the first unlucky sod to get in this state.  Many devices, like hue bulbs, are challenging to re-pair when they think they are still connected to the coordinator. I've found I either need to use the remote control trick, or find the Serial number and pair with the Philips Hue hub, just so I can un-pair them and restore them to pairing mode.  Neither is a cheap option for someone just using the bulbs.

thargy, 2024-11-04:
Here's some logging of what a message looks like, e.g. for `nwkAddr` `39790`:

```log
[2024-11-02 23:20:17] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd - {"dstaddr":39790,"relaycount":1,"relaylist":[40583]}
[2024-11-02 23:20:17] debug: 	zh:zstack:unpi:parser: --- parseNext [254,35,68,129,0,0,0,0,245,252,1,1,0,21,0,205,155,232,0,0,15,8,58,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,8,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,7,165,232,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,41,254,5,69,196,245,252,1,135,158,149,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,245,252,1,1,0,21,0,15,205,232,0,0,15,8,58,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,156,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,91,213,232,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,5,254,5,69,196,245,252,1,135,158,149,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,245,252,1,1,0,21,0,160,2,233,0,0,15,8,58,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,253,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,195,5,233,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,76,254,5,69,196,110,155,1,135,158,105,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,227,19,234,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,121,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,106,42,234,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,201]
[2024-11-02 23:20:17] debug: 	zh:zstack:unpi:parser: --> parsed 35 - 2 - 4 - 129 - [0,0,0,0,245,252,1,1,0,21,0,205,155,232,0,0,15,8,58,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28] - 8
...
[2024-11-02 23:20:17] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":39790,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":21,"securityuse":0,"timestamp":15246599,"transseqnumber":0,"len":15,"data":{"type":"Buffer","data":[8,19,10,1,0,32,70,226,255,32,31,228,255,32,0]}}
[2024-11-02 23:20:17] debug: 	zh:controller: Trying to identify unknown device with address '39790'
[2024-11-02 23:20:17] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,245,252,1,135,158,149,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,245,252,1,1,0,21,0,15,205,232,0,0,15,8,58,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,156,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,91,213,232,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,5,254,5,69,196,245,252,1,135,158,149,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,245,252,1,1,0,21,0,160,2,233,0,0,15,8,58,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,253,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,195,5,233,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,76,254,5,69,196,110,155,1,135,158,105,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,227,19,234,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,121,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,106,42,234,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,201]
[2024-11-02 23:20:17] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [245,252,1,135,158] - 149
...
[2024-11-02 23:20:17] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd - {"dstaddr":39790,"relaycount":1,"relaylist":[40583]}
[2024-11-02 23:20:17] debug: 	zh:zstack:unpi:parser: --- parseNext [254,35,68,129,0,0,0,0,245,252,1,1,0,21,0,15,205,232,0,0,15,8,58,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,156,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,91,213,232,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,5,254,5,69,196,245,252,1,135,158,149,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,245,252,1,1,0,21,0,160,2,233,0,0,15,8,58,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,253,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,195,5,233,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,76,254,5,69,196,110,155,1,135,158,105,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,227,19,234,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,121,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,106,42,234,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,201]
[2024-11-02 23:20:17] debug: 	zh:zstack:unpi:parser: --> parsed 35 - 2 - 4 - 129 - [0,0,0,0,245,252,1,1,0,21,0,15,205,232,0,0,15,8,58,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28] - 156
...
[2024-11-02 23:20:17] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":39790,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":21,"securityuse":0,"timestamp":15258971,"transseqnumber":0,"len":15,"data":{"type":"Buffer","data":[8,19,10,1,0,32,70,226,255,32,31,228,255,32,0]}}
[2024-11-02 23:20:17] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,245,252,1,135,158,149,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,245,252,1,1,0,21,0,160,2,233,0,0,15,8,58,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,253,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,195,5,233,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,76,254,5,69,196,110,155,1,135,158,105,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,227,19,234,0,0,15,8,19,10,1,0,32,70,226,255,32,31,228,255,32,0,135,158,28,121,254,5,69,196,110,155,1,135,158,105,254,35,68,129,0,0,0,0,110,155,1,1,0,21,0,106,42,234,0,0,15,8,19,10,1,0,3 [... cut ...]
        // This only lasts until next reboot (runtime Set), allowing to 'force' another trigger if necessary.
    }
```

As we see the following, only once as expected:
```log
[2024-11-02 23:20:17] debug: 	zh:controller: Trying to identify unknown device with address '39790'
```

20 seconds later we get the:
```log
[2024-11-02 23:20:37] debug: 	zh:controller: Failed to retrieve IEEE address for device '39790': Error: AREQ - ZDO - ieeeAddrRsp after 10000ms
```

Is this failure expected, or is it as a result of a lack of throttling due to ~30 device messages and, subsequent queries being sent out?

thargy, 2024-11-04:
Some from the same [log.log](https://github.com/user-attachments/files/17617960/log.log) as above, I found a succesful query with `nwkAddr` `44525`:

```log
[2024-11-02 23:23:04] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd - {"dstaddr":44525,"relaycount":1,"relaylist":[44970]}
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: <-- [254,29,68,129,0,0,0,0,237,173,1,1,0,69,0,113,119,26,0,0,9,28,95,17,56,10,0,0,32,3,170,175,28,146]
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: --- parseNext [254,29,68,129,0,0,0,0,237,173,1,1,0,69,0,113,119,26,0,0,9,28,95,17,56,10,0,0,32,3,170,175,28,146]
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: --> parsed 29 - 2 - 4 - 129 - [0,0,0,0,237,173,1,1,0,69,0,113,119,26,0,0,9,28,95,17,56,10,0,0,32,3,170,175,28] - 146
[2024-11-02 23:23:04] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":44525,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":69,"securityuse":0,"timestamp":1734513,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[28,95,17,56,10,0,0,32,3]}}
[2024-11-02 23:23:04] debug: 	zh:controller: Trying to identify unknown device with address '44525'
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:writer: --> frame [254,4,37,1,237,173,0,0,96]
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,1,0,101]
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,1,0,101]
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 1 - [0] - 101
[2024-11-02 23:23:04] debug: 	zh:zstack:znp: <-- SRSP: ZDO - ieeeAddrReq - {"status":0}
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: <-- [254,13,69,129,0,228,27,174,0,16,68,239,84,237,173,27,0,44]
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: --- parseNext [254,13,69,129,0,228,27,174,0,16,68,239,84,237,173,27,0,44]
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: --> parsed 13 - 2 - 5 - 129 - [0,228,27,174,0,16,68,239,84,237,173,27,0] - 44
[2024-11-02 23:23:04] debug: 	zh:zstack:znp: <-- AREQ: ZDO - ieeeAddrRsp - {"zdo":[0,{"eui64":"0x54ef441000ae1be4","nwkAddress":44525,"startIndex":27,"assocDevList":[]}]}
[2024-11-02 23:23:04] debug: 	zh:controller: Received ZDO response: clusterId=IEEE_ADDRESS_RESPONSE, status=SUCCESS, payload={"eui64":"0x54ef441000ae1be4","nwkAddress":44525,"startIndex":27,"assocDevList":[]}
[2024-11-02 23:23:04] debug: 	zh:controller: IEEE address from '0x54ef441000ae1be4:44525'
[2024-11-02 23:23:04] debug: 	zh:controller: IEEE address is from unknown device '0x54ef441000ae1be4:44525'
[2024-11-02 23:23:04] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-11-02 23:23:07] debug: 	zh:zstack:unpi:parser: <-- [254,7,69,196,126,50,2,174,165,162,251,154]
[2024-11-02 23:23:07] debug: 	zh:zstack:unpi:parser: --- parseNext [254,7,69,196,126,50,2,174,165,162,251,154]
[2024-11-02 23:23:07] debug: 	zh:zstack:unpi:parser: --> parsed 7 - 2 - 5 - 196 - [126,50,2,174,165,162,251] - 154
...
[2024-11-02 23:23:14] debug: 	zh:controller: Failed to retrieve IEEE address for device '44525': Error: AREQ - ZDO - ieeeAddrRsp after 10000ms
[2024-11-02 23:23:14] debug: 	zh:controller: Data is from unknown device with address '44525', skipping...
...
[2024-11-02 23:25:33] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd - {"dstaddr":44525,"relaycount":1,"relaylist":[44970]}
[2024-11-02 23:25:33] debug: 	zh:zstack:unpi:parser: --- parseNext [254,98,68,129,0,0,192,252,237,173,1,1,0,69,0,136,36,168,0,0,78,28,95,17,57,10,247,0,65,69,100,16,0,101,16,0,3,40,30,152,57,0,0,0,0,149,57,0,0,0,0,150,57,0,16,22,69,151,57,0,0,0,0,5,33,3,0,154,32,0,9,33,0,21,13,35,23,11,0,0,14,35,0,0,0,0,10,33,170,175,12,32,1,17,35,1,0,0,0,170,175,28,171]
[2024-11-02 23:25:33] debug: 	zh:zstack:unpi:parser: --> parsed 98 - 2 - 4 - 129 - [0,0,192,252,237,173,1,1,0,69,0,136,36,168,0,0,78,28,95,17,57,10,247,0,65,69,100,16,0,101,16,0,3,40,30,152,57,0,0,0,0,149,57,0,0,0,0,150,57,0,16,22,69,151,57,0,0,0,0,5,33,3,0,154,32,0,9,33,0,21,13,35,23,11,0,0,14,35,0,0,0,0,10,33,170,175,12,32,1,17,35,1,0,0,0,170,175,28] - 171
[2024-11-02 23:25:33] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":64704,"srcaddr":44525,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":69,"securityuse":0,"timestamp":11019400,"transseqnumber":0,"len":78,"data":{"type":"Buffer","data":[28,95,17,57,10,247,0,65,69,100,16,0,101,16,0,3,40,30,152,57,0,0,0,0,149,57,0,0,0,0,150,57,0,16,22,69,151,57,0,0,0,0,5,33,3,0,154,32,0,9,33,0,21,13,35,23,11,0,0,14,35,0,0,0,0,10,33,170,175,12,32,1,17,35,1,0,0,0]}}
[2024-11-02 23:25:33] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-11-02 23:25:33] debug: 	zh:controller: Data is from unknown device with address '44525', skipping...
[2024-11-02 23:25:33] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,196,241,147,0,224,254,98,68,129,0,0,192,252,241,147,1,1,0,94,0,224,232,168,0,0,78,28,95,17,213,10,247,0,65,69,100,16,1,101,16,1,3,40,27,152,57,82,32,239,64,149,57,131,36,55,63,150,57,154,153,21,69,151,57,108,8,248,65,5,33,6,0,154,32,16,9,33,0,21,13,35,23,11,0,0,14,35,0,0,0,0,10,33,0,0,12,32,1,17,35,1,0,0,0,241,147,29,22]
[... cut ...]
        logger_1.logger.debug(`IEEE address from '${payload.eui64}:${payload.nwkAddress}'`, NS);
        const device = model_1.Device.byIeeeAddr(payload.eui64);
        if (!device) {
            logger_1.logger.debug(`IEEE address is from unknown device '${payload.eui64}:${payload.nwkAddress}'`, NS);
            return;
        }
        device.updateLastSeen();
        this.selfAndDeviceEmit(device, 'lastSeenChanged', { device, reason: 'networkAddress' });
        this.checkDeviceNetworkAddress(device, payload.eui64, payload.nwkAddress);
    }
```

I hope this helps.

Koenkk, 2024-11-05:
- Is there still a `data-backup` directory with the original database?
- Did you do a `git clean`? I don't see how otherwise the `database.db` could be removed
- With z2m 2.0.0 the update script will be simplified, so these conflicts should not occur anymore.

thargy, 2024-11-05:
> * Is there still a `data-backup` directory with the original database?

Nope, when you try to update again, it complains unless you delete the backup directory. After doing a hard reset and clean, I had to delete the backup to run `.update.sh` again to build the node packages, etc.

> * Did you do a `git clean`? I don't see how otherwise the `database.db` could be removed

I confess I did. I thought I had enough with my backed-up configuration and coordinator. It was Saturday night, I was tired, and mistakes were made. 😮‍💨 

> * With z2m 2.0.0 the update script will be simplified, so these conflicts should not occur anymore.

As I pointed out above, after 40 years of software dev experience, careful backup strategies, and bitter experience, I still managed to **** it up; at least I won't be alone in my ignominy! To be entirely fair, sometimes there have even been bugs, e.g. #19988 and #18842.  

Regardless, the ability to rebuild your network without pairing is well within scope and seems like a valuable feature to pursue.  Hence I changed this from a bug report to a feature request, recognising my own idiocy and providing a workaround for others seeking to avoid re-pairing their entire network.

Have you given any thought to [my suggestion](https://github.com/Koenkk/zigbee2mqtt/issues/24612#issuecomment-2454306303)?

Is this something you would prefer I raise in Zigbee-herdman as a feature request (I note you don't encourage issues there so I asked first)?

Also, what are your thoughts on the potential bug raised in [this comment](https://github.com/Koenkk/zigbee2mqtt/issues/24612#issuecomment-2454583260)?  Woudl you prefer I raise that as a separate issue, and if so, in which repo?

thargy, 2024-11-05:
As an aside, and in the greatest tradition of stable hands the world over, I have, of course, committed my database and state files to backup _and_ source control...  So this particular horse should not recur for me.

Koenkk, 2024-11-06:
> If the coordinator is in 'Permit Join' mode treat any message as a join attempt. I noticed that many of my devices were still talking to the coordinator and the messages were getting thrown away as the device was not in database.db (but was in configuration.yaml and on the coordinator).

We've already been thinking about this, it's also needed for the (semi)automatic ZHA -> Z2M migration (https://github.com/Koenkk/zigbee2mqtt/discussions/24478), so definitely an interesting idea!

thargy, 2025-01-06:
Not stale

closed 2025-03-16 by github-actions[bot]