Koenkk/zigbee2mqtt #24613: WL-19DWZ Wrong message

albertinizao, 2024-11-02:
### What happened?

After the 1.41.0, the contact sensor now detected as WL-19DWZ (previous version was  detected as SNZB-04) works wrong.

Arbitrarily these devices are sending me incorrect messages with “contact=true” without the actual state of the device having changed. In this case, at 23:10:03 the device did not change his real status, without any movement, but sent these messages.

[2024-11-02 23:10:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Silla Bailey', payload '{"battery":100,"battery_low":false,"contact":false,"linkquality":102,"tamper":false,"voltage":3000}'
[2024-11-02 23:10:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Silla Bailey', payload '{"battery":100,"battery_low":false,"contact":false,"linkquality":99,"tamper":false,"voltage":3000}'
[2024-11-02 23:10:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Silla Bailey', payload '{"battery":100,"battery_low":false,"contact":false,"linkquality":96,"tamper":false,"voltage":3000}'
[2024-11-02 23:10:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Silla Bailey', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":99,"tamper":false,"voltage":3000}'

### What did you expect to happen?

The WL-19DWZ should not send any message with "contact=true" when the door is open

### How to reproduce it (minimal and precise)

Version 1.41.0. Put a device WL-19DWZ close and then open it. 
Review the status is now opened (concact=false). 
Wait until a message with contact=true is sent, . Sometimes it happens a few minutes later, sometimes a hour later.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20241002

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Add-on on home assistant at raspbery pi

### Debug log

[2024-11-02 23:10:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Silla Bailey', payload '{"battery":100,"battery_low":false,"contact":false,"linkquality":102,"tamper":false,"voltage":3000}'
[2024-11-02 23:10:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Silla Bailey', payload '{"battery":100,"battery_low":false,"contact":false,"linkquality":99,"tamper":false,"voltage":3000}'
[2024-11-02 23:10:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Silla Bailey', payload '{"battery":100,"battery_low":false,"contact":false,"linkquality":96,"tamper":false,"voltage":3000}'
[2024-11-02 23:10:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Silla Bailey', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":99,"tamper":false,"voltage":3000}'

Oliohly, 2024-11-06:
I got the same problem with any of these devices: they suddenly set 'contact' to 'true' - after long or short duration of time

Koenkk, 2024-11-07:
Should be fixed in the latest  [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Oliohly, 2024-11-14:
I installed the dev branch and now all WL-19DWZ window sensors work like expected. Thanks! Will this be implemented in the stable version?

Koenkk, 2024-11-16:
yes, next release is on 1 December

closed 2024-11-16 by Koenkk