Koenkk/zigbee2mqtt #24615: Working Door/Window Contacts detected as Leak Detectors since 1.41.0

senrabyar, 2024-11-03:
### What happened?

I have a set of 4 Aubess Contact sensors which have been correctly identified and work fine until the release of 1.41.0

### What did you expect to happen?

to be detected as Contact

### How to reproduce it (minimal and precise)

problem is evident immediately on pairing ...

[image]
[image]

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Setup

RPi5

senrabyar, 2024-11-03:
Sorry it's late - perhaps I'm too tired - the devices are actually motion sensors not contact sensors - indeed they are listed by Z2M as SNZB-03 
Here's how some of them appear in HA
[image]

sqorpo, 2024-11-03:
I have the same issue. I think it was fixed in the dev build. I'll have to wait until it gets pushed to the home assistant add-on. until then I have lights controlled by "leak sensors" I guess..

senrabyar, 2024-11-03:
Ahh but some of my lights do flash when my actual leak sensors are working.    No worries here being downstairs whilst the bath upstairs fills !

NicCarlos, 2024-11-04:
Same problem with Tuya Motion Sensor ZMS-102.

Koenkk, 2024-11-06:
Already fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)!

closed 2024-11-06 by Koenkk