Koenkk/zigbee2mqtt #24616: Aqara LLKZMK11LM (lumi.relay.c2acn01) stopped working after update to 1.40.2 or 1.41.0

I-G-1-1, 2024-11-03:
### What happened?

I have two "Aqara LLKZMK11LM (lumi.relay.c2acn01) relay" they work fine untill zigbee2mqtt version 1.40.1.
After update to 1.40.2 (tried also to update to 1.41.0) they stopped working: if I try to change the state of the switch it works the first time then if I try to change the state another time nothing happens.
On zigbee2mqtt log (log level: info) seems like switching the relay the second time crash zigbee2mqtt.
```
[2024-11-03 10:58:19] info: 	z2m:mqtt: MQTT publish: topic 'hass/zigbee2mqtt/bridge/state', payload 'offline'
[2024-11-03 10:58:19] info: 	z2m: Disconnecting from MQTT server
[2024-11-03 10:58:19] info: 	z2m: Stopping zigbee-herdsman...
```

Reverting to zigbee2mqtt 1.40.1 the device works perfectly.

### What did you expect to happen?

on zigbee2mqtt webUI when I switch state on/off I expect the relay to switch on/off

### How to reproduce it (minimal and precise)

1. on zigbee2mqtt webUI go to "Aqara LLKZMK11LM" device
2. go to exposes section
3. go to state and change from _off_ to _on_ or from _on_ to _off_

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20211222

### Adapter

zStack30x (CC2538)

### Setup

Python Virtual Environment (following this guide: https://www.zigbee2mqtt.io/guide/installation/07_python_virtual_environment.html)

### Debug log

[log info.log](https://github.com/user-attachments/files/17610036/log.info.log)

SanderBlom, 2024-11-03:
Same issue here: 

Manufacturer - [Aqara](https://www.zigbee2mqtt.io/supported-devices/#v=Aqara)
Model - [SSM-U01](https://www.zigbee2mqtt.io/devices/SSM-U01.html#aqara-ssm-u01)
Zigbee Model -  lumi.switch.n0agl1
Firmware Build date - Jan 12 2024

Logs when trying to switch it on. 
````
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/rele-vifte', payload '{"auto_off":null,"button_lock":null,"consumer_connected":null,"current":null,"device_temperature":17,"energy":6.86,"led_disabled_night":null,"linkquality":138,"overload_protection":null,"power":8.1,"power_outage_memory":false,"state":"ON","update":{"installed_version":45,"latest_version":45,"state":"idle"},"update_available":null,"voltage":null}'

```

hendrik607, 2024-11-04:
same here, they just stop working randomly, now 2 not working from 4 (Z2M version 1.41.0-1)

carlreid, 2024-11-05:
Also encountering the same kind of problem, where I can't toggle on/off state in most cases. They also don't seem to respond to OTA, as that's also what I was trying.

I-G-1-1, 2024-12-02:
The issue is still present on version 1.42.0

here is the log (debug) when I try to switch the relay:
```
[2024-12-02 11:29:17] debug: 	z2m:mqtt: Received MQTT message on 'hass/zigbee2mqtt/relay_controller_01/set' with data '{"state_l2":"ON"}'
[2024-12-02 11:29:17] debug: 	z2m: Publishing 'set' 'state' to 'relay_controller_01'
[2024-12-02 11:29:17] debug: 	zh:controller:endpoint: ZCL command 0x00158d0006c8f91b/2 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2024-12-02 11:29:19] info: 	z2m:mqtt: MQTT publish: topic 'hass/zigbee2mqtt/motion_sensor_stanzino', payload '{"battery":100,"linkquality":99,"occupancy":false,"power_outage_count":20,"voltage":3015}'
[2024-12-02 11:29:22] debug: 	zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,6,4,171,162,1,1,0,94,0,197,67,112,0,0,7,24,25,10,0,0,24,1,171,162,29,124]
[2024-12-02 11:29:22] debug: 	zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,6,4,171,162,1,1,0,94,0,197,67,112,0,0,7,24,25,10,0,0,24,1,171,162,29,124]
[2024-12-02 11:29:22] debug: 	zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,6,4,171,162,1,1,0,94,0,197,67,112,0,0,7,24,25,10,0,0,24,1,171,162,29] - 124
[2024-12-02 11:29:22] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":1030,"srcaddr":41643,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":94,"securityuse":0,"timestamp":7357381,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,25,10,0,0,24,1]}}
[2024-12-02 11:29:22] debug: 	zh:controller: Received payload: clusterID=1030, address=41643, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=94, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":25,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":24,"attrData":1}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-12-02 11:29:22] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-12-02 11:29:22] debug: 	z2m: Received Zigbee message from 'motion_sensor_camera_mensola', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
[2024-12-02 11:29:22] info: 	z2m:mqtt: MQTT publish: topic 'hass/zigbee2mqtt/motion_sensor_camera_mensola', payload '{"battery":100,"linkquality":94,"occupancy":true,"power_outage_count":72,"voltage":3095}'
[2024-12-02 11:29:30] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,10,0,106,175,1,1,0,55,0,116,161,112,0,0,5,16,132,0,0,0,106,175,29,200]
[2024-12-02 11:29:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,10,0,106,175,1,1,0,55,0,116,161,112,0,0,5,16,132,0,0,0,106,175,29,200]
[2024-12-02 11:29:30] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,10,0,106,175,1,1,0,55,0,116,161,112,0,0,5,16,132,0,0,0,106,175,29] - 200
[2024-12-02 11:29:30] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":10,"srcaddr":44906,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":55,"securityuse":0,"timestamp":7381364,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[16,132,0,0,0]}}
[2024-12-02 11:29:30] debug: 	zh:controller: Received payload: clusterID=10, address=44906, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=55, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":132,"commandIdentifier":0},"payload":[{"attrId":0}],"command":{"ID":0,"name":"read","parameters":[{"name":"attrId","type":33}],"response":1}}
[2024-12-02 11:29:30] debug: 	zh:controller:endpoint: ZCL command 0x00158d0006c8f91b/1 genTime.readRsp({"time":786450570}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":132,"writeUndiv":false})
[2024-12-02 11:29:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-12-02 11:29:30] debug: 	z2m: Received Zigbee message from 'relay_controller_01', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
[2024-12-02 11:29:39] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,10,0,124,23,1,1,0,28,0,129,16,113,0,0,5,16,187,0,0,0,124,23,29,153]
[2024-12-02 11:29:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,10,0,124,23,1,1,0,28,0,129,16,113,0,0,5,16,187,0,0,0,124,23,29,153]
[2024-12-02 11:29:39] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,10,0,124,23,1,1,0,28,0,129,16,113,0,0,5,16,187,0,0,0,124,23,29] - 153
[2024-12-02 11:29:39] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":10,"srcaddr":6012,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":28,"securityuse":0,"timestamp":7409793,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[16,187,0,0,0]}}
[2024-12-02 11:29:39] debug: 	zh:controller: Received payload: clusterID=10, address=6012, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=28, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":187,"commandI [... cut ...] eType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":4,"commandIdentifier":10},"payload":[{"attrId":1285,"dataType":35,"attrData":131072}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
```

I don't have any knwoledge to understand this log, so I hope someone could help to fix this bug. 

For now I'm reverting to 1.40.1.

Thanks

I-G-1-1, 2024-12-02:
The issue is still present on version 1.42.0

here is the log (debug) when I try to switch the relay:
```
[2024-12-02 11:29:17] debug: 	z2m:mqtt: Received MQTT message on 'hass/zigbee2mqtt/relay_controller_01/set' with data '{"state_l2":"ON"}'
[2024-12-02 11:29:17] debug: 	z2m: Publishing 'set' 'state' to 'relay_controller_01'
[2024-12-02 11:29:17] debug: 	zh:controller:endpoint: ZCL command 0x00158d0006c8f91b/2 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2024-12-02 11:29:19] info: 	z2m:mqtt: MQTT publish: topic 'hass/zigbee2mqtt/motion_sensor_03', payload '{"battery":100,"linkquality":99,"occupancy":false,"power_outage_count":20,"voltage":3015}'
[2024-12-02 11:29:22] debug: 	zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,6,4,171,162,1,1,0,94,0,197,67,112,0,0,7,24,25,10,0,0,24,1,171,162,29,124]
[2024-12-02 11:29:22] debug: 	zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,6,4,171,162,1,1,0,94,0,197,67,112,0,0,7,24,25,10,0,0,24,1,171,162,29,124]
[2024-12-02 11:29:22] debug: 	zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,6,4,171,162,1,1,0,94,0,197,67,112,0,0,7,24,25,10,0,0,24,1,171,162,29] - 124
[2024-12-02 11:29:22] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":1030,"srcaddr":41643,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":94,"securityuse":0,"timestamp":7357381,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,25,10,0,0,24,1]}}
[2024-12-02 11:29:22] debug: 	zh:controller: Received payload: clusterID=1030, address=41643, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=94, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":25,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":24,"attrData":1}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-12-02 11:29:22] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-12-02 11:29:22] debug: 	z2m: Received Zigbee message from 'motion_sensor_04', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
[2024-12-02 11:29:22] info: 	z2m:mqtt: MQTT publish: topic 'hass/zigbee2mqtt/motion_sensor_04', payload '{"battery":100,"linkquality":94,"occupancy":true,"power_outage_count":72,"voltage":3095}'
[2024-12-02 11:29:30] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,10,0,106,175,1,1,0,55,0,116,161,112,0,0,5,16,132,0,0,0,106,175,29,200]
[2024-12-02 11:29:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,10,0,106,175,1,1,0,55,0,116,161,112,0,0,5,16,132,0,0,0,106,175,29,200]
[2024-12-02 11:29:30] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,10,0,106,175,1,1,0,55,0,116,161,112,0,0,5,16,132,0,0,0,106,175,29] - 200
[2024-12-02 11:29:30] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":10,"srcaddr":44906,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":55,"securityuse":0,"timestamp":7381364,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[16,132,0,0,0]}}
[2024-12-02 11:29:30] debug: 	zh:controller: Received payload: clusterID=10, address=44906, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=55, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":132,"commandIdentifier":0},"payload":[{"attrId":0}],"command":{"ID":0,"name":"read","parameters":[{"name":"attrId","type":33}],"response":1}}
[2024-12-02 11:29:30] debug: 	zh:controller:endpoint: ZCL command 0x00158d0006c8f91b/1 genTime.readRsp({"time":786450570}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":132,"writeUndiv":false})
[2024-12-02 11:29:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-12-02 11:29:30] debug: 	z2m: Received Zigbee message from 'relay_controller_01', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
[2024-12-02 11:29:39] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,10,0,124,23,1,1,0,28,0,129,16,113,0,0,5,16,187,0,0,0,124,23,29,153]
[2024-12-02 11:29:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,10,0,124,23,1,1,0,28,0,129,16,113,0,0,5,16,187,0,0,0,124,23,29,153]
[2024-12-02 11:29:39] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,10,0,124,23,1,1,0,28,0,129,16,113,0,0,5,16,187,0,0,0,124,23,29] - 153
[2024-12-02 11:29:39] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":10,"srcaddr":6012,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":28,"securityuse":0,"timestamp":7409793,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[16,187,0,0,0]}}
[2024-12-02 11:29:39] debug: 	zh:controller: Received payload: clusterID=10, address=6012, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=28, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":187,"commandIdentifier":0},"payload":[{"att [... cut ...] eType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":4,"commandIdentifier":10},"payload":[{"attrId":1285,"dataType":35,"attrData":131072}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
```

I don't have any knwoledge to understand this log, so I hope someone could help to fix this bug. 

For now I'm reverting to 1.40.1.

Thanks

hendrik607, 2025-02-01:
issue still there do not know how to remove 'stale' label

I-G-1-1, 2025-02-01:
the issue is still here

ADMiNZ, 2025-02-03:
<h3>Rollback Firmware for Zigbee Devices via OTA (Tested on Aqara LLKZMK11LM)</h3>

<p>Successfully rolled back firmware to <code>OTA_lumi_relay_c2acn01_V1.3.35.20180920.20181008170031.ota</code>.</p>

<h4>Implementation Steps:</h4>
<ol>
  <li>Add to <code>configuration.yaml</code>:
    <pre><code>ota:
    zigbee_ota_override_index_location: ota.json</code></pre>
  </li>
  
  <li>Create <code>ota.json</code>:
    <pre><code>[
  {
    "url": "OTA_lumi_relay_c2acn01_V1.3.35.20180920.20181008170031.ota",
    "force": true
  }
]</code></pre>
  </li>

  <li>Upload firmware file to <code>/homeassistant/zigbee2mqtt/</code></li>
</ol>

<h4>Additional Resources:</h4>
<ul>
  <li>🔧 OTA firmware collection: 
    <a href="https://github.com/Koenkk/zigbee2mqtt/files/6240732/ota_files.zip" target="_blank">GitHub Repository</a>
  </li>
  <li>📖 Documentation: 
    <a href="https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files" target="_blank">Zigbee2MQTT OTA Guide</a>
  </li>
</ul>

<p><em>Credits: @sorryusernameisalreadytaken, @timurlutfullin</em></p>

I-G-1-1, 2025-02-03:
Sorry but your problem it's not the issue originally described here.

The relay works well with the firmware 9-22-2021 on zigbee2mqtt 1.40.2 and stopped when using zigbee2mqtt 1.41.0 or newer. Reverting to 1.40.2 it starts working again.

With this behaviour the issue originaly described here seems not a firmware issue.

ADMiNZ, 2025-02-03:
I fully acknowledge that the root issue might lie within Zigbee2MQTT itself. However, diagnosing the exact cause is beyond my current technical expertise. On the other hand, performing a firmware rollback takes mere minutes and, crucially, resolves functionality in my setup. To clarify, I’m not claiming this as a universal solution—I’m simply sharing a workaround that successfully addressed my specific

I-G-1-1, 2025-04-05:
the issue is still there