Koenkk/zigbee2mqtt #24620: [New device support]: TS0601 from _TZE200_9xfjixap

akcsachsen, 2024-11-03:
### Link

https://de.aliexpress.com/item/1005005666920591.html?spm=a2g0o.order_list.order_list_main.4.109b5c5fXmA74l&gatewayAdapt=glo2deu

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0xa4c13861bcadd1e8","nwkAddr":3076,"manufId":4417,"manufName":"_TZE200_9xfjixap","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_9xfjixap","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1730633746302}

### Zigbee2MQTT version

1.41.0

### Comments

please add support for that thermostat.
I found that device, but there is no support until now.

thanks

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_9xfjixap',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

there is no use possible

Schmandre, 2024-11-05:
Same issue here, any idea how to fix that?

akcsachsen, 2024-11-14:
sorry, I cannot "install" the file= avatto_me167.js
I got: z2m: Request 'zigbee2mqtt/bridge/request/extension/save' failed with error: 'ConstructorClass is not a constructor'

der-seemann, 2024-11-15:
Same here, please add support

Elektronix-LTD, 2024-11-16:
Same for me. No temperature entitles, not possible to control temperature as well. I can't use thermostat because of this issue.

Buhhjaa, 2024-12-01:
#24552
I think they talking about the same Device. 
To add more maybe helpful info:
In my case the packaging say **Modell "ME167"** but zb2mqtt shows the same model and vendor described here. I checked the DB and found these 2:
https://www.zigbee2mqtt.io/devices/ME167.html
https://www.zigbee2mqtt.io/devices/TS0601_thermostat_3.html
In my case im running HA on VM therefore in my opinion i can't configure any .js file

Chrisni2, 2024-12-01:
Looks like mine is supported now after today's z2m update 😊

Buhhjaa, 2024-12-01:
> Looks like mine is supported now after today's z2m update 😊

Confirmed :) Great Job! And a BIG Thank you to the Developers!

akcsachsen, 2024-12-03:
after updating it is working now, very thank you !!

just a wish, it would be nice, if I can see the valve data (in %) or even control that instead of the dest. temperature ;)

0verEngineer, 2024-12-05:
There is one problem, the local temperature calibration can only be changed from -9 to +9 instead of -10 to +10 like it should be (tested with tuya gateway and app)

Buhhjaa, 2024-12-27:
After testing the Thermostat for a while, i have to say that it works sadly at the bare minimum. Also the commands do not match actually the value i want to change.
If i want to change the Temp value in heating mode, obviously the calibration value is triggerd.
[image]

The schedule setup is not working at all. If i switch to Auto Mode the Thermostat in the Morning switch to 156°C. 

So what ist actually working? Its On and Off. I currently just switch the Thermostat in the Morning on and in the evening off. Thats it!

0verEngineer, 2024-12-27:
> After testing the Thermostat for a while, i have to say that it works sadly at the bare minimum. Also the commands do not match actually the value i want to change.
> If i want to change the Temp [...]

I don't have these Problems

Buhhjaa, 2024-12-28:
Interesting, im using the actual ember driver with the sonoff stick. Also using a second Stick as repeater (Firmwaredate: 20220515). Log gives me the following errors:

`2024-12-28 09:46:30zh:controller:device: Device '0xa4c1384cde0b6ff7' is only compliant to revision '21' of the ZigBee specification (current revision: 23).`

`[2024-12-28 09:56:59] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "11306"`

`[2024-12-28 09:46:29] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Thermostat_EgFlur1', payload '{"battery_low":false,"child_lock":"UNLOCK","current_heating_setpoint":21,"error":null,"frost_protection":"ON","last_seen":"2024-12-28T09:46:29+01:00","linkquality":240,"local_temperature":21,"local_temperature_calibration":2,"running_state":"heat","scale_protection":"ON","schedule_friday":"07:00/21.0 18:30/22.0 19:30/21.0 22:00/16.0","schedule_monday":"07:00/21.0 18:30/22.0 19:30/21.0 22:00/16.0","schedule_saturday":"07:00/21.0 18:30/22.0 19:30/21.0 22:00/16.0","schedule_sunday":"07:00/21.0 18:30/22.0 19:30/21.0 22:00/16.0","schedule_thursday":"07:00/21.0 18:30/22.0 19:30/21.0 22:00/16.0","schedule_tuesday":"07:00/21.0 18:30/22.0 19:30/21.0 22:00/16.0","schedule_wednesday":"07:00/21.0 18:30/22.0 19:30/21.0 22:00/16.0","system_mode":"heat"}'`

closed 2025-03-06 by github-actions[bot]