Koenkk/zigbee2mqtt #24621: [Feature request]: Reset occupancy on _TZ3000_msl6wxk9 /  _TZ3000_6ygjfyll if no occupancy received for a while

srett, 2024-11-03:
### Is your feature request related to a problem? Please describe

I have a few different battery-powered PIR sensors, and every now and then it appears that a message updating the occupancy status gets lost. So it might happen that I go to bed and the next day I see the heating stayed on in the kitchen all night as z2m never got the clearance message from the PIR.

### Describe the solution you'd like

Now after turning on herdsman debug logs I quickly noticed that two of the sensors, the _TZ3000_msl6wxk9 and _TZ3000_6ygjfyll, do in fact send the occupancy status exactly once a minute as long as it detects motion, and then one final update clearing the occupancy, if no more motion is detected. This should make it trivial and safe to add a timeout, i.e. if no occupancy update is received within 3 or 4 minutes, just clear the occupancy status if it is set.

As I'm not familiar with the code base, I have no clue though at which point this should be handled, or if there's already some infrastructure to handle similar issues with other devices.

### Describe alternatives you've considered

Put 3 PIRs in every room and do a majority vote :>

### Additional context

According to the [converter](https://github.com/Koenkk/zigbee-herdsman-converters/blob/8cd25aae2490e53dda31af595a578503029c3819/src/devices/tuya.ts#L1855), these should all be identical, but I only have two of these: `['_TZ3000_mcxw5ehu', '_TZ3000_6ygjfyll', '_TZ3040_6ygjfyll', '_TZ3000_msl6wxk9']`

I also have a _TZ3040_bb6xaihh and confirmed that it does **not** send repeated updates while occupied, unfortunately.

This is what the log looks like when I'm in a room for some time:

```
[2024-11-03T08:30:49+01:00] debug:         zh:controller: Received payload: clusterID=1280, address=52078, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=65, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":127,"commandIdentifier":0},"payload":{"zonestatus":1,"extendedstatus":0},"command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}}
[2024-11-03T08:31:49+01:00] debug:         zh:controller: Received payload: clusterID=1280, address=52078, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=58, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":128,"commandIdentifier":0},"payload":{"zonestatus":1,"extendedstatus":0},"command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}}
[2024-11-03T08:32:49+01:00] debug:         zh:controller: Received payload: clusterID=1280, address=52078, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=54, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":129,"commandIdentifier":0},"payload":{"zonestatus":1,"extendedstatus":0},"command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}}
[2024-11-03T08:33:49+01:00] debug:         zh:controller: Received payload: clusterID=1280, address=52078, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=58, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":130,"commandIdentifier":0},"payload":{"zonestatus":1,"extendedstatus":0},"command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}}
[2024-11-03T08:34:49+01:00] debug:         zh:controller: Received payload: clusterID=1280, address=52078, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=58, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":131,"commandIdentifier":0},"payload":{"zonestatus":1,"extendedstatus":0},"command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}}
[2024-11-03T08:35:49+01:00] debug:         zh:controller: Received payload: clusterID=1280, address=52078, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=54, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":132,"commandIdentifier":0},"payload":{"zonestatus":1,"extendedstatus":0},"command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}}
[2024-11-03T08:36:49+01:00] debug:         zh:controller: Received payload: clusterID=1280, address=52078, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=54, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":133,"commandIdentifier":0},"payload":{"zonestatus":1,"extendedstatus":0},"command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}}
```

srett, 2024-11-05:
@Koenkk could you give a rough outline how I could go about implementing this, and where it would fit in? I was thinking about adding code in tuya.ts. I already found there is one device that makes use of `setTimeout` and `clearTimeout` to work around an issue.  
So my idea was to (re)start a timeout every time I get an occupancy message and stop the timeout if there is a clearance message. Then once the timer expires I publish a synthetic "no occupancy" event. Does that sound reasonable?

Koenkk, 2024-11-06:
I think you can use https://github.com/Koenkk/zigbee-herdsman-converters/blob/ad4d87dab7bd60010a28152d8ee26d32ef0c5094/src/converters/fromZigbee.ts#L1158

Koenkk, 2024-11-11:
Looks good, if I understand correctly, the keep time makes no difference here? If yes, then I propose to remove it. I'm wondering that also the occupancy timeout doesn't make a difference then? (since it's always 60)

srett, 2024-11-12:
> if I understand correctly, the keep time makes no difference here?

I couldn't test that since I failed to change it. (But maybe we're talking past each other regarding occupancy timeout vs. keep time?)
Keep time is configurable on the device itself, and determines how long the PIR sensor waits after no more activity is detected to send the clearance signal. Occupancy timeout would be the same on the z2m side, it would trigger that long after the last activity update was received, to handle the case where the clearance message of the device got lost, so just a safety measure for this device.
The default keep time is 60 seconds, which matches the activity update interval of 60 seconds. I've no idea if changing the keep time would change the update interval too, or if that's fixed at 60 seconds.
Anyways, since the maximum configurable keep time is 120 seconds, I'd vote for hard-coding this to 122 seconds, so twice the (default) update interval.

So to summarize: Introduce the converter shown above (minus the keep_time stuff, as this is unverified to be related) and change the device to use that one. :)

Koenkk, 2024-11-13:
> So to summarize: Introduce the converter shown above (minus the keep_time stuff, as this is unverified to be related) and change the device to use that one. :)

Sounds good! 122 seconds makes sense indeed, maybe 125 just in case?

closed 2024-12-15 by srett