Koenkk/zigbee2mqtt #24624: [New device support]: TS0601_TZE284_7zazvlyn zigbee screen with scenes

niqu1982, 2024-11-03:
### Link

https://www.aliexpress.com/item/1005007630565315.html?spm=a2g0o.detail.pcDetailBottomMoreOtherSeller.1.45d74cdo4cdobH&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40050.354490.0&scm_id=1007.40050.354490.0&scm-url=1007.40050.354490.0&pvid=3f10e6dc-c7c4-4404-b438-07a31fab2b66&_t=gps-id%3ApcDetailBottomMoreOtherSeller%2Cscm-url%3A1007.40050.354490.0%2Cpvid%3A3f10e6dc-c7c4-4404-b438-07a31fab2b66%2Ctpp_buckets%3A668%232846%238107%231934&pdp_npi=4%40dis%21PLN%21285.66%21244.99%21%21%2169.61%2159.70%21%40211b813b17306556232802301e0114%2112000041571631077%21rec%21PL%21703466389%21X&utparam-url=scene%3ApcDetailBottomMoreOtherSeller%7Cquery_from%3A

### Database entry

friendly_name":"0xf84477fffeab06d5","ieee_address":"0xf84477fffeab06d5"

### Zigbee2MQTT version

1.41.0-1

### Comments

3.5 Inch Smart Wall Switch Touch Screen 

#4-way switch :Effortlessly switch between four customizable scenariosEnhance your home automation with intuitive physical controls
#4-way curtain control :Operate up to 4 curtains to match your daily routines
#4-way lighting adjustment :Adjust color temperature and brightness4-way dimming, support offline work
#8-way scene :Manage up to 8 different scenarios for ultimate conveniencePerfect for creating the ideal ambiance for any occasion
#High-resolution display : 360 x 480 Resolution.Responsive touch controlfor seamless interaction
#Built-in radar sensing :Turn on the switch, turn on the radar bydefault, and when the set screen savertime is up, The main camera willautomatically display a silent screen,which will automatically light up whensomeone approaches

info 2024-11-03 18:32:51z2m: Accepting joining not in blocklist device '0xf84477fffeab06d5'
info 2024-11-03 18:32:51zh:controller: Interview for '0xf84477fffeab06d5' started
info 2024-11-03 18:32:51z2m: Device '0xf84477fffeab06d5' joined
info 2024-11-03 18:32:51z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xf84477fffeab06d5"},"type":"device_connected"}'
info 2024-11-03 18:32:51z2m: Starting interview of '0xf84477fffeab06d5'
info 2024-11-03 18:32:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xf84477fffeab06d5"},"type":"pairing"}'
info 2024-11-03 18:32:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf84477fffeab06d5","ieee_address":"0xf84477fffeab06d5"},"type":"device_joined"}'
info 2024-11-03 18:32:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf84477fffeab06d5","ieee_address":"0xf84477fffeab06d5","status":"started"},"type":"device_interview"}'
info 2024-11-03 18:32:52zh:controller: Succesfully interviewed '0xf84477fffeab06d5'
info 2024-11-03 18:32:52z2m: Successfully interviewed '0xf84477fffeab06d5', device has successfully been paired
warning 2024-11-03 18:32:52z2m: Device '0xf84477fffeab06d5' with Zigbee model 'TS0601' and manufacturer name '_TZE284_7zazvlyn' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info 2024-11-03 18:32:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xf84477fffeab06d5","supported":false},"type":"pairing"}'
info 2024-11-03 18:32:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf84477fffeab06d5","ieee_address":"0xf84477fffeab06d5","status":"successful","supported":false},"type":"device_interview"}'

### External definition

```shell
I cant;t sniff device output cluseters. Can somebody help i trying using support new devices but this is black magic for me :(
https://www.aliexpress.com/item/1005007630487666.html?spm=a2g0o.order_list.order_list_main.11.67931c2431BLHj&gatewayAdapt=glo2pol

I have conbee II
also i can use skyconnect or cc2531 to sniffing communication.
```

### What does/doesn't work with the external definition?

I try but now working :( I don't see exposess

vuk1980, 2024-12-14:
I received same device yesterday. If you didn't find anything out I've managed to get it work for 4 switches and 8 scenes so far.
Curtains and Dimmers switches are still far away as I am strugling with datapoints at this stage.
I also notices that time is synced but date is not... (to be more precise year)

My external definition for this device:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            manufacturerName: '_TZE284_7zazvlyn',
        },
    ],
    model: 'TS0601_F3-PRO',
    vendor: 'Tuya',
    description: '4-gang switch with 8 scenes, 4 dimmers and 4 curtains',
    fromZigbee: [
        tuya.fz.datapoints,
        tuya.fz.indicator_mode,
        fz.tuya_cover_options,
        fz.cover_position_tilt,
    ],
    toZigbee: [
        tuya.tz.datapoints,
        tuya.tz.backlight_indicator_mode_1,
        tz.cover_state,
        tz.cover_position_tilt,
        tz.tuya_cover_calibration,
        tz.tuya_cover_reversal,

    ],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        ...[1, 2, 3, 4].map((i) => tuya.exposes.switch().withEndpoint(`l${i}`)),
        e.action(['scene_1', 'scene_2', 'scene_3', 'scene_4','scene_5', 'scene_6', 'scene_7', 'scene_8']),
    ],
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'action', tuya.valueConverter.static('scene_1')],
            [2, 'action', tuya.valueConverter.static('scene_2')],
            [3, 'action', tuya.valueConverter.static('scene_3')],
            [4, 'action', tuya.valueConverter.static('scene_4')],
            [5, 'action', tuya.valueConverter.static('scene_5')],
            [6, 'action', tuya.valueConverter.static('scene_6')],
            [7, 'action', tuya.valueConverter.static('scene_7')],
            [8, 'action', tuya.valueConverter.static('scene_8')],
            [16, 'backlight_mode', tuya.valueConverter.onOff],
            [121, 'state_l1', tuya.valueConverter.onOff],
            [122, 'state_l2', tuya.valueConverter.onOff],
            [123, 'state_l3', tuya.valueConverter.onOff],
            [124, 'state_l4', tuya.valueConverter.onOff],
        ],
    },
    endpoint: (device) => {
        return {l1: 1,l2: 1,l3: 1,l4: 1};
    },
};

module.exports = definition;
`
If anyone have something related to this device don't be shy to share :)

wangkaidev, 2025-01-04:
Same device here.

SuperWangKai, 2025-01-04:
Got same device here. The device has very smooth UX.

SuperWangKai, 2025-01-05:
Got this device properties description JSON from Tuya cloud developer, hope it helps:
```json
{
  "result": {
    "properties": [
      {
        "code": "scene_1",
        "custom_name": "",
        "dp_id": 1,
        "time": 1736006421342,
        "type": "enum",
        "value": "scene"
      },
      {
        "code": "scene_2",
        "custom_name": "",
        "dp_id": 2,
        "time": 1736006421342,
        "type": "enum",
        "value": "scene"
      },
      {
        "code": "scene_3",
        "custom_name": "",
        "dp_id": 3,
        "time": 1736008696183,
        "type": "enum",
        "value": "scene"
      },
      {
        "code": "scene_4",
        "custom_name": "",
        "dp_id": 4,
        "time": 1736008697584,
        "type": "enum",
        "value": "scene"
      },
      {
        "code": "scene_5",
        "custom_name": "",
        "dp_id": 5,
        "time": 1736008698332,
        "type": "enum",
        "value": "scene"
      },
      {
        "code": "scene_6",
        "custom_name": "",
        "dp_id": 6,
        "time": 1736008699081,
        "type": "enum",
        "value": "scene"
      },
      {
        "code": "scene_7",
        "custom_name": "",
        "dp_id": 7,
        "time": 1736006421342,
        "type": "enum",
        "value": "scene"
      },
      {
        "code": "scene_8",
        "custom_name": "",
        "dp_id": 8,
        "time": 1736008700692,
        "type": "enum",
        "value": "scene"
      },
      {
[... 387 lines cut ...]
        "value": ""
      },
      {
        "code": "backlight_switch",
        "custom_name": "",
        "dp_id": 149,
        "time": 1736009436573,
        "type": "bool",
        "value": true
      },
      {
        "code": "mode_change",
        "custom_name": "",
        "dp_id": 150,
        "time": 1736009037455,
        "type": "enum",
        "value": "switch"
      }
    ]
  },
  "success": true,
  "t": 1736009659907,
  "tid": "8a8636cccabc11efb25b46a4973a7179"
}
```

SuperWangKai, 2025-01-05:
Trying to learn external converter code and make the 4 curtain covers work. Here is the testing code, toZigbee seems working, but fromZigbee is not, no UI update when updated from the device, any ideas?

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets; // Import exposes presets

// Helper function to extract data value
function getTuyaDataValue(datatype, data) {
    switch (datatype) {
        case 0: // raw
        case 1: // boolean
            return data[0] === 1;
        case 2: // value (4 bytes)
            return (data[0] << 24) | (data[1] << 16) | (data[2] << 8) | data[3];
        case 3: // string
            return data.toString();
        case 4: // enum
            return data[0];
        case 5: // bitmap
            return (data[0] << 8) | (data[1]);
        default:
            return null;
    }
}

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_7zazvlyn',
        },
    ],
    zigbeeModel: ['TS0601_F3-Pro'],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Tuya F3-pro smart panel with curtain control',
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    fromZigbee: [
        {
            cluster: 'manuSpecificTuya',
            type: ['commandDataResponse', 'commandSetDataResponse', 'commandMcuSyncTime'],
            convert: (model, msg, publish, options, meta) => {
                const dp = msg.data.dp; // Data point
                const datatype = msg.data.datatype;
                const data = msg.data.data;
                const value = getTuyaDataValue(datatype, data);
                const result = {};

                console.debug(`Processing dp: ${dp}, datatype: ${datatype}, raw data: ${JSON.stringify(data)}, interpreted value: ${value}`);

                if (msg.type === 'commandMcuSyncTime') {
                    console.info('Received commandMcuSyncTime, ignoring as no action required.');
                    return result;
                }

                // Map data points to state and position attributes
                switch (dp) {
                    case 133: result.state = value === 0 ? 'open' : value === 1 ? 'stop' : 'close'; result.endpoint = '1'; break;
                    case 113: result.position = value; result.endpoint = '1'; break;
                    case 134: result.state = value === 0 ? 'open' : value === 1 ? 'stop' : 'close'; result.endpoint = '2'; break;
                    case 114: result.position = value; result.endpoint = '2'; break;
                    case 135: result.state = value === 0 ? 'open' : value === 1 ? 'stop' : 'close'; result.endpoint = '3'; break;
                    case 115: result.position = value; result.endpoint = '3'; break;
                    case 136: result.state = value === 0 ? 'open' : value === 1 ? 'stop' : 'close'; result.endpoint = '4'; break;
                    case 116: result.position = value; result.endpoint = '4'; break;
[... 82 lines cut ...]
                        } else {
                            throw new Error(`Unsupported state value: ${value}`);
                        }
                    }
                }
            },
        },
    ],
    exposes: [
        e.cover_position().withEndpoint('cover_1'),
        e.cover_position().withEndpoint('cover_2'),
        e.cover_position().withEndpoint('cover_3'),
        e.cover_position().withEndpoint('cover_4'),
    ],
    // endpoint: (device) => ({
    //     cover_1: 1,
    //     cover_2: 2,
    //     cover_3: 3,
    //     cover_4: 4,
    // }),
    meta: {multiEndpoint: true},
};

module.exports = definition;
```

devkaiwang, 2025-01-12:
Two cover controls are still not working correct - 
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_7zazvlyn',
        },
    ],
    zigbeeModel: ['TS0601_F3-Pro'],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Tuya F3-pro smart panel with curtain control',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.cover_position().withEndpoint('l1'),
        e.cover_position().withEndpoint('l2')
    ],

    // configure: async (device, coordinatorEndpoint) => {
    //     for (const ep of [1, 2]) {
    //         await reporting.bind(device.getEndpoint(ep), coordinatorEndpoint, ['closuresWindowCovering']);
    //     }
    // },

    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [133, 'state_l1', tuya.valueConverterBasic.lookup({
                'CLOSE': tuya.enum(0),
                'STOP': tuya.enum(1),
                'OPEN': tuya.enum(2)
            })],
            [113, 'position_l1', tuya.valueConverter.coverPosition],

            [134, 'state_l2', tuya.valueConverterBasic.lookup({
                'CLOSE': tuya.enum(0),
                'STOP': tuya.enum(1),
                'OPEN': tuya.enum(2)
            })],
            [114, 'position_l2', tuya.valueConverter.coverPosition]
        ]
    },
    endpoint: (device) => {
        return {
            l1: 1,
            l2: 2
        };
    }
};

module.exports = definition;
```

Would you please help? @Koenkk

ilmarh, 2025-03-18:
got this device today, thought it is less smart. but appeared to be new and not working with z2m. so i begin searching and find this issue, thus decide to bump it
meanwhile will read about converters, and looking for tuya wired hub to check functionality

AlekseyPetrov75, 2025-04-06:
Is there any progress in the converter code? Does it make sense to buy this device?

devkaiwang, 2025-04-18:
For people who cannot access to the AliExress link -

[image]
[image]

devkaiwang, 2025-04-19:
The panel's switches, scenes, and backlight control are working on my side. 
**For curtain, should we use "binding"?** 
https://smarthomescene.com/guides/how-to-bind-zigbee-devices-directly-in-zigbee2mqtt/
https://www.zigbee2mqtt.io/guide/usage/binding.html

I cannot bind the panel to a curtain/window-opener:

[image]

unnamedi, 2025-05-21:
Did you manage to get it working? I'm waiting for mine to arrive and it would be nice to have it working or partially working!

unnamedi, 2025-05-21:
> I received same device yesterday. If you didn't find anything out I've managed to get it work for 4 switches and 8 scenes so far. Curtains and Dimmers switches are still far away as I am strugling with datapoints at this stage. I also notices that time is synced but date is not... (to be more precise year)
> 
> My external definition for this device:
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
> fingerprint: [
> {
> modelID: 'TS0601',
> manufacturerName: '_TZE284_7zazvlyn',
> },
> ],
> model: 'TS0601_F3-PRO',
> vendor: 'Tuya',
> description: '4-gang switch with 8 scenes, 4 dimmers and 4 curtains',
> fromZigbee: [
> tuya.fz.datapoints,
> tuya.fz.indicator_mode,
> fz.tuya_cover_options,
> fz.cover_position_tilt,
> ],
> toZigbee: [
> tuya.tz.datapoints,
> tuya.tz.backlight_indicator_mode_1,
> tz.cover_state,
> tz.cover_position_tilt,
> tz.tuya_cover_calibration,
> tz.tuya_cover_reversal,
> ],
> onEvent: tuya.onEventSetTime,
> configure: tuya.configureMagicPacket,
> exposes: [
> ...[1, 2, 3, 4].map((i) => tuya.exposes.switch().withEndpoint(l${i})),
> e.action(['scene_1', 'scene_2', 'scene_3', 'scene_4','scene_5', 'scene_6', 'scene_7', 'scene_8']),
> ],
> meta: {
> multiEndpoint: true,
> tuyaDatapoints: [
> [1, 'action', tuya.valueConverter.static('scene_1')],
> [2, 'action', tuya.valueConverter.static('scene_2')],
> [3, 'action', tuya.valueConverter.static('scene_3')],
> [4, 'action', tuya.valueConverter.static('scene_4')],
> [5, 'action', tuya.valueConverter.static('scene_5')],
> [6, 'action', tuya.valueConverter.static('scene_6')],
> [7, 'action', tuya.valueConverter.static('scene_7')],
> [8, 'action', tuya.valueConverter.static('scene_8')],
> [16, 'backlight_mode', tuya.valueConverter.onOff],
> [121, 'state_l1', tuya.valueConverter.onOff],
> [122, 'state_l2', tuya.valueConverter.onOff],
> [123, 'state_l3', tuya.valueConverter.onOff],
> [124, 'state_l4', tuya.valueConverter.onOff],
> ],
> },
> endpoint: (device) => {
> return {l1: 1,l2: 1,l3: 1,l4: 1};
> },
> };
> module.exports = definition;
> ```
> 
> If anyone have something related to this device don't be shy to share :)

Were you able to use his MMWAVE?

devkaiwang, 2025-06-11:
My question is how can we connect the controls to another Zigbee device? E.g. how can we connect a curtain control to control another Zigbee curtain motor?

niqu1982, 2025-09-02:
> I have it in testing... I'll try to share this week with everybody!

Any update ? working ?

reineruhry, 2025-09-02:
Here is my testing code!

```
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");
const e = exposes.presets;

const tuyaLocal = {
    dataPoints: {
        name_l1: 125,
        name_l2: 126,
        name_l3: 127,
        name_l4: 128,
    },
};

const convLocal = {
    name: {
        to: (v, meta) => {
            const utf8bytes = new TextEncoder().encode(v);
            return Array.from(utf8bytes, (utf8bytes) => utf8bytes.toString(16).padStart(4, "0")).join("");
        },
        from: (v, meta) => {
            const bytes = [];
            for (let i = 0; i < v.length; i += 4) {
                bytes.push(Number.parseInt(v.slice(i, i + 4), 16));
            }
            const hexToBytes = Uint8Array.from(bytes);
            return new TextDecoder("utf-8").decode(hexToBytes);
        },
    },
};

const definition = {
    fingerprint: [
        {
            modelID: "TS0601",
            manufacturerName: "_TZE284_7zazvlyn",
        },
    ],
    zigbeeModel: ["TS0601_F3-Pro"],
    model: "TS0601",
    vendor: "TuYa",
    description: "Tuya F3-pro smart panel with curtain control",
    fromZigbee: [tuya.fz.datapoints], //fzLocal.switch_name,
    toZigbee: [tuya.tz.datapoints], //tzLocal.switch_name,
    configure: async (device, coordinatorEndpoint) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ["genOnOff"]);
    },
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    extend: [

    ],
    exposes: [
        tuya.exposes.switch().withEndpoint('l1'),
        tuya.exposes.switch().withEndpoint('l2'),
        tuya.exposes.switch().withEndpoint('l3'),
        tuya.exposes.switch().withEndpoint('l4'),
        e.text("name", ea.STATE_SET).withEndpoint("l1").withDescription("Name for Switch 1"),
        e.text("name", ea.STATE_SET).withEndpoint("l2").withDescription("Name for Switch 2"),
        e.text("name", ea.STATE_SET).withEndpoint("l3").withDescription("Name for Switch 3"),
        e.text("name", ea.STATE_SET).withEndpoint("l4").withDescription("Name for Switch 4"),
        e.text("light_switch_name", ea.STATE_SET).withEndpoint("l1").withDescription("Name for Light Switch 1"),
        e.text("light_switch_name", ea.STATE_SET).withEndpoint("l2").withDescription("Name for Light Switch 2"),
        e.text("light_switch_name", ea.STATE_SET).withEndpoint("l3").withDescription("Name for Light Switch 3"),
[... 47 lines cut ...]
            [128, "light_switch_name_l4", convLocal.name],
            [129, "curtain_switch_name_l1", convLocal.name],
            [130, "curtain_switch_name_l2", convLocal.name],
            [131, "curtain_switch_name_l3", convLocal.name],
            [132, "curtain_switch_name_l4", convLocal.name],
            [137, "name_l1", convLocal.name],
            [138, "name_l2", convLocal.name],
            [139, "name_l3", convLocal.name],
            [140, "name_l4", convLocal.name],
            [141, "scene_name_1", convLocal.name],
            [142, "scene_name_2", convLocal.name],
            [143, "scene_name_3", convLocal.name],
            [144, "scene_name_4", convLocal.name],
            [145, "scene_name_5", convLocal.name],
            [146, "scene_name_6", convLocal.name],
            [147, "scene_name_7", convLocal.name],
            [148, "scene_name_8", convLocal.name],
            [150, "mode", tuya.valueConverterBasic.lookup({switch: tuya.enum(0), scene: tuya.enum(1), smart_light: tuya.enum(2), curtain: tuya.enum(3)})],
        ],
    },
};

module.exports = definition;

```

devkaiwang, 2025-09-21:
I have some progress now.

Converter is like 70% done. With some sync logic in Automation, I can sync an attribute of the panel with the according one of the LED. The problem is there are too many attributes from both sides (all attributes of the smart panel, target lights and covers to be controlled), I'm not an expert of Automation, seems we need some blueprint with slots to fill up. I will clean up and paste the converter code when I have some time later.

https://github.com/user-attachments/assets/b07bd03b-e418-4438-a523-c33c339f3803

2137home, 2025-09-26:
ok, it works. 
i can press switches and scenes, changing names, and make an automations as triggers.
But if it were possible to change icons...

devkaiwang, 2025-09-27:
> also found https://github.com/namanbharihoke/-tuya-f3pro-zigbee2mqtt

Great work but seems there is no time sync and text encoding is incorrect in that repo.

namanbharihoke, 2025-09-30:
> also found https://github.com/namanbharihoke/-tuya-f3pro-zigbee2mqtt

this is my convertor, let me try and incorporate the time sync and scene names from [devkaiwang]. maybe we can finally have a completely working one.

devkaiwang, 2025-09-30:
> This is my converter code, I think we have everything Tuya App supports: 
> - 4 physical/touch relay buttons
> - 8 scene buttons
> - 4 curtain/cover controls
> - 4 LED controls
> - Renaming support for all the controls above
> - Date & time sync.
> 
>  We still need e.g. Home Assistant Automation to control other devices. Let's complete it later. 
> 
> Here is my converter code, let me know if it works for you :)
> 
> ```js
> // Tuya TS0601_F3-Pro external converter
> 
> import {
>   access as ea,
>   presets as e,
> } from "zigbee-herdsman-converters/lib/exposes";
> import * as tuya from "zigbee-herdsman-converters/lib/tuya";
> 
> /** Local converters */
> const convLocal = {
>   name: {
>     // String -> hex (UTF-16BE)
>     to: (v) => Buffer.from(String(v), "utf16le").swap16().toString("hex"),
> 
>     // hex (UTF-16BE) -> String
>     from: (hex = "") => {
>       if ((hex.length & 1) !== 0) return ""; // must be even-length hex
>       return Buffer.from(hex, "hex").swap16().toString("utf16le");
>     },
>   },
> };
> 
> /** @type {import('zigbee-herdsman-converters/lib/types').DefinitionWithExtend} */
> export default {
>   fingerprint: [{ modelID: "TS0601", manufacturerName: "_TZE284_7zazvlyn" }],
>   zigbeeModel: ["TS0601_F3-Pro"],
>   model: "TS0601",
>   vendor: "TuYa",
>   description:
>     "Tuya F3-pro smart panel. 4-gang switch with 8 scenes, 4 dimmers and 4 curtains.",
> 
>   // Modern extend handles fz/tz, MagicPacket, time-sync, and initial queries
>   extend: [
>     tuya.modernExtend.tuyaBase({
>       dp: true,
>       forceTimeUpdates: true,
>       queryOnConfigure: true,
>     }),
>   ],
> 
>   exposes: [
>     // Enable screen saver or turn off screen.
>     e
>       .binary("backlight_switch", ea.STATE_SET, "ON", "OFF")
>       .withDescription("Panel screen on/off"),
> 
>     // Relays.
>     tuya.exposes.switch().withEndpoint("l1"),
>     tuya.exposes.switch().withEndpoint("l2"),
>     tuya.exposes.switch().withEndpoint("l3"),
>     tuya.exposes.switch().withEndpoint("l4"),
> 
>     // Switch names.
>     e
>       .text("switch_name", ea.STATE_SET)
>       .withEndpoint("l1")
>       .withDescription("Name for Switch 1"),
[... 273 lines cut ...]
>       [145, "scene_name_5", convLocal.name],
>       [146, "scene_name_6", convLocal.name],
>       [147, "scene_name_7", convLocal.name],
>       [148, "scene_name_8", convLocal.name],
> 
>       // Screen backlight switch.
>       [149, "backlight_switch", tuya.valueConverter.onOff],
> 
>       // Seems not implemented by the panel.
>       [
>         150,
>         "mode_change",
>         tuya.valueConverterBasic.lookup({
>           switch: tuya.enum(0),
>           dimmer: tuya.enum(1),
>           curtain: tuya.enum(2),
>         }),
>       ],
>     ],
>   },
> };
> 
> ```

this already works, you can try it

GabrielSimoes1983, 2025-11-03:
Two questionsS6E TuyaAsking for help on 2 questions:
1 - does the presence radar work?
2 - as a newbie on zigbee2mqtt how can I map the entities from this to home assistant so I can see changes reflected on both sides?