Koenkk/zigbee2mqtt #24625: Startup not completing?

corporategoth, 2024-11-03:
### What happened?

I migrated from TrueNAS 24.04 -> 24.10, which meant my container went from being Kubernetes to Docker (which actually gives me more control).
I also moved from Mosquitto to EMQX for an MQTT broker.

Since doing so, Z2M is not completing it's startup.
Specifically, the Ember driver is coming up, and detects all my zigbee devices.
I also see Z2M logged into the EMQX server, and subscribed to zigbee2mqtt/#

HOWEVER:
1. I don't see any PUBLISH log lines to MQTT for the devices.
2. The Z2M front end does not start up (it doesn't even open a socket).
3. I am using HA, so I also don't see any auto-configuration going on (it does not subscribe to the HA topic either).

If I turn on debugging, I see communications with the Zigbee network, but nothing else.

### What did you expect to happen?

I expected the Z2M UI to show up.
I expected it to publish the state of my devices.

### How to reproduce it (minimal and precise)

Just start up the docker container.  Here is my container config:

```
services:
    zigbee2mqtt:
        container_name: zigbee2mqtt
        image: koenkk/zigbee2mqtt:latest
        volumes:
          - /mnt/bigfast/app/zigbee2mqtt:/app/data
          - /run/udev:/run/udev:ro
        networks:
          bound_bond0:
          mqtt:
          web:
        ports:
          - 10103:80
        environment:
          - TZ=America/New_York
        devices:
          - /dev/ttyACM0:/dev/ttyACM0
        restart: always
```

Any my configuration.yaml:

```
experimental:
  new_api: true
frontend:
  port: 8080
permit_join: false
mqtt:
  server: mqtt://emqx
  base_topic: zigbee2mqtt
  user: zigbee2mqtt
  password: [[REDACTED]]
  keepalive: 60
  reject_unauthorized: true
  version: 5
serial:
  port: /dev/ttyACM0
  adapter: ember
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_syslog:
[... 611 lines cut ...]
      - 0x04cd15fffe79bbe5/1
homeassistant:
  legacy_triggers: false
  status_topic: homeassistant/status
```

### Zigbee2MQTT version

latest (1.41.0)

### Adapter firmware version

8.0.2 [GA]

### Adapter

Sonoff Dongle-E

### Setup

Docker container in TrueNAS Scale

### Debug log

[log.log.gz](https://github.com/user-attachments/files/17611835/log.log.gz)

corporategoth, 2024-11-03:
I thought this might be related to https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/624 - however my Z2M is not 'functioning normally' (publishing to the MQTT server), and I don't have availability: true enabled.

corporategoth, 2024-11-14:
@Koenkk @Nerivec any ideas on this?

Nerivec, 2024-11-14:
I know several users had issues when moving to that broker and quickly reverted (i.e. never got any answers...).
And looks like MQTT is having troubles:
```
[2024-11-03 14:45:15] error: 	z2m: MQTT error: write ECONNRESET
```

Koenkk, 2024-11-14:
> I also moved from Mosquitto to EMQX for an MQTT broker.

Can you try moving back to mosquitto?

corporategoth, 2024-11-15:
@Nerivec I noticed that it does get the MQTT error, disconnects, then connects the second time (I can verify from EMQX that zigbee2mqtt was connected).  So there was definitely something weird about the connection then disconnection.  HOWEVER, if it could not connect, I would expect it would shut down Z2M (which it does not).  That is what happens if I have an incorrect passwrod or the MQTT server is not accessible at all.

@Koenkk Changing back to mosquitto fixed it, and I see stuff publishing to the MQTT server again, and I see a UI.  So from a base functionality perspective, I guess it's working.  But from a usability perspective I don't see why EMQX would not work (as it DOES make a connection, and maintains it, otherwise Z2M would die due to no MQTT connection).  I would ideally like to move to EMQX as it has much better management and monitoring capabilities.

Nerivec, 2024-11-15:
It must be something specific to your network that creates such a huge packet. `/bridge/definitions` should be less than 150000 chars, and `/bridge/devices`, with 150 or so "regular" devices should be somewhere around 1-300000 chars (guesstimate, depends on clusters quite a bit).
Can you take a closer look at the logs, see what topic the packet was for? Since it's right after `/bridge/state`, it's one of the retained.

corporategoth, 2024-11-15:
Here is, the 4100 bytes it prints in the log, presumably truncated:
```
1Ø¿␦zigbee2mqtt/bridge/devices[{"disabled":false,"endpoints":{"1":{"bindings":[{"cluster":"genOnOff","target":{"endpoint":1,"ieee_address":"0xe0798dfffea88b0a","type":"endpoint"}}],"clusters":{"input":["genBasic","genIdentify","genOnOff","genTime","genOta","26","lightingColorCtrl"],"output":["genBasic","genIdentify","genGroups","genScenes","genOnOff","genLevelCtrl","genPollCtrl","lightingColorCtrl","msIlluminanceMeasurement","msTemperatureMeasurement","msRelativeHumidity","msOccupancySensing","ssIasZone","haMeterIdentification","haApplianceStatistics","haElectricalMeasurement","seMetering","touchlink","manuSpecificLegrandDevices","manuSpecificSamsungAccelerometer"]},"configured_reportings":[],"scenes":[]},"242":{"bindings":[],"clusters":{"input":[],"output":["greenPower"]},"configured_reportings":[],"scenes":[]}},"friendly_name":"Coordinator","ieee_address":"0xe0798dfffea88b0a","interview_completed":true,"interviewing":false,"network_address":0,"supported":true,"type":"Coordinator"},{"date_code":"20240412","definition":{"description":"2-in-1 switch + dimmer","exposes":[{"features":[{"access":7,"description":"On/off state of this light","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","label":"Brightness","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0}],"type":"light"},{"access":1,"category":"diagnostic","description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Sum of consumed energy","label":"Energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":3,"category":"config","features":[{"access":3,"description":"Animation Effect to use for the LEDs","label":"Effect","name":"effect","property":"effect","type":"enum","values":["off","solid","fast_blink","slow_blink","pulse","chase","open_close","small_to_big","aurora","slow_falling","medium_falling","fast_falling","slow_rising","medium_rising","fast_rising","medium_blink","slow_chase","fast_chase","fast_siren","slow_siren","clear_effect"]},{"access":3,"description":"Calculated by using a hue color circle(value/255*360) If color = 255 display white","label":"Color","name":"color","property":"color","type":"numeric","value_max":255,"value_min":0},{"access":3,"description":"Brightness of the LEDs","label":"Level","name":"level","property":"level","type":"numeric","value_max":100,"value_min":0},{"access":3,"description":"1-60 is in seconds calculated 61-120 is in minutes calculated by(value-60) Example a value of 65 would be 65-60 = 5 minutes - 120-254 Is in hours calculated by(value-120) Example a value of 132 would be 132-120 would be 12 hours. - 255 Indefinitely","label":"Duration","name":"duration","property":"duration","type":"numeric","value_max":255,"value_min":0}],"label":"Led effect","name":"led_effect","property":"led_effect","type":"composite"},{"access":3,"category":"config","features":[{"access":3,"description":"Individual LED to target.","label":"Led","name":"led","property":"led","type":"enum","values":["1","2","3","4","5","6","7"]},{"access":3,"description":"Animation Effect to use for the LED","label":"Effect","name":"effect","property":"effect","type":"enum","values":["off","solid","fast_blink","slow_blink","pulse","chase","falling","rising","aurora","clear_effect"]},{"access":3,"description":"Calculated by using a hue color circle(value/255*360) If color = 255 display white","label":"Color","name":"color","property":"color","type":"numeric","value_max":255,"value_min":0},{"access":3,"description":"Brightness of the LED","label":"Level","name":"level","property":"level","type":"numeric","value_max":100,"value_min":0},{"access":3,"description":"1-60 is in seconds calculated 61-120 is in minutes calculated by(value-60) Example a value of 65 would be 65-60 = 5 minutes - 120-254 Is in hours calculated by(value-120)  Example a value of 132 would be 132-120 would be 12 hours.
```

I have a lot (70? 80?) of Inovelli Blue switches, and I guess the description of them can be verbose once you factor in descriptions.
I don't see this logged in my Z2M log files (though maybe it would @ debug? I don't know).   Either way, as I said, increasing the max packet size worked.

Nerivec, 2024-11-16:
Indeed, lots of features per device (multi-function), larger payload.
The combination of hitting the max, and having `version: 5` triggers a `DISCONNECT` packet, which was not present in previous mqtt spec from what I gathered... 😅

> though maybe it would @ debug?

Yes, from retained on startup.

corporategoth, 2024-11-16:
Presumably the size of that could be calculated before you send the max packet size.
Then you could possibly send a max packet size of <sz>  rounded up to next power of 2.  Or <sz> * 1.2 rounded up to the next power of 2.

corporategoth, 2024-11-25:
@Nerivec Unless I mistake the PR, even with the change, the max packet size would still need to be adjusted on the broker side to allow large packets - the Z2M setting is just the max packet size it will allow FROM the broker.  At least, that's how the docs read.

If that's the case, there should probably be a section in the website docs on 'Broker setup' for what options may need to be changed (more than just documenting the field exists as a notation on the maximum_packet_size setting).

Or at least put it under the [Zigbee2MQTT fails to start](https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#zigbee2mqtt-fails-to-start) section.

closed 2025-01-09 by Nerivec