Koenkk/zigbee2mqtt #24626: [New device support]: Philips Hue Light Strip LCL007

elduca, 2024-11-03:
### Link

https://www.philips-hue.com/de-ch/

### Database entry

I"id":95, "type": "Router"  "ieeeAddr": "0x0017880104c6f33", "nwkAddr" :64409, "manufId" :4107, "manufName": "Signify Netherlands B.V.", "powerSource": "Mains (single phase)", "modelId" : "LCL007"

### Zigbee2MQTT version

1.41.0

### Comments

i paired the lightstrip and all worked.

### External definition

```shell
const {philipsLight} = require ('zigbee-herdsman-converters/lib/philips');
const definition = {
zigbeeModel: ['LCL007'],
model: 'LCL007' , vendor: 'Signify Netherlands B.V.', description: 'Automatically generated definition',
extend: [philipsLight({"colorTemp": {"range": [153,500]}, "color": {"modes": ["xy", "hs"], "enhancedHue" : true}}) ],
meta: {},
module. exports = definition;
```

### What does/doesn't work with the external definition?

Everythng works fine.

Wrangler-93, 2024-12-09:
Ordered the LCL700 because of this post.

I could pair, update firmware, control it from the device page and rename it,everything works (Shows as NOT Supported)
(Does NOT show in dashboard)

Does NOT show in Homebridge (So cannot be added to Homekit)

so NOT supported (Yet) for me :-(

Koenkk, 2024-12-10:
Will be supported in z2m 2.0.0 (next month)