Koenkk/zigbee2mqtt #24627: Ikea Inspelning power no change when socket is off

Alexofhiswalnut, 2024-11-03:
### What happened?

Hello, with my Ikea Inspelning plugs, when my washing machine finishes, the power consumption stays at 0.5W indefinitely, unless I press the refresh button in Z2M, which then brings it down to 0W.

What solution do you have to automatically refresh the plug to avoid showing the false residual power?

Have a good day.

### Zigbee2MQTT version

1.41

### Adapter firmware version

7.4.3 GA

### Adapter

Sonoff v2

### Setup

Pi4

Alexofhiswalnut, 2024-11-04:
Non problem

closed 2024-11-04 by Alexofhiswalnut

MaxRower, 2024-11-05:
You should fine tune the report intervals to fit your needs. I set them all to min 10s, max 10s, just like my tasmota plugs.

nicola-spreafico, 2024-11-08:
Same problem here,

This morning (more than 10h ago), I switched off the printer using the plug switch, as you can see here:
[image]

Despite being off, the power consumption is reading 3.2W which is not possibile considering the device is not powered.

I went to Z2M interface and found this:
[image]
Only after I pressed the refresh button I got the actual 0.0 value
[image]