Koenkk/zigbee2mqtt #24630: IKEA INSPELNING detected as TRETAKT - support all the models

Misiu, 2024-11-04:
### What happened?

I recently bought IKEA INSPELNING, after adding it to Z2M it is shown as TRETAKT. On/Off works, but there is no power monitoring.
[image]

Other users report that they have a different model on that plug: https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2454074804

According to https://csa-iot.org/csa_product/inspelning-smart-plug-3/ the same plug has different models: E2206, E2220, E2221, E2222, E2223, E2224, E2225, E2226

I'm using https://github.com/ficueu/ESPHome-IoT-modules/tree/main/ESP-POE-MM-v1.0#esp-poe-mm-v10 as my zigbee coordinator (Ebyte E72-2G4M20S1E (CC2652p with U.FL connector) as ZigBee coordinator (on external board))

### What did you expect to happen?

I expected the plug to be correctly detected and the power monitor functionality should work as expected

### How to reproduce it (minimal and precise)

Go to Polish IKEA, buy INSPELNING plug, plug it in, enable pairing, check Z2M interface.

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

Home Assistant OS

Misiu, 2024-11-04:
I've just ordered IKEA DIRIGERA hub, will check the plug there update the firmware to the latest version, and try again in Z2M.

Koenkk, 2024-11-09:
> Is there a way to reflash the plug? 

AFAIK not, looks like they sold you a broken plug.

closed 2025-01-16 by github-actions[bot]

chpego, 2025-04-21:
Just for information, and so that other people don't feel “alone” :) 

I'm in France, I also made this observation after buying 3 INSPELNING connected plugs, it turned out that 2 of them were identified as the other model (TRETAKT - E22X4); I found this on the 2 plugs with two different Zigbee keys and also tested on ZHA; strange this marketing by IKEA. I returned the 2 plugs, made an exchange, and surprise on the 2 recovered in exchange, one of them was marked again (TRETAKT - E22X4).

So the problem doesn't come from Z2M.

nekomanta, 2025-05-04:
@chpego I just hung up with IKEA support, it seems that the problem is localised to the strasbourg store, lucky us ! :)
Support asked me to go back to the store to exchange them (3 out of 4 were seen as TRATAKT), I'll ask them to test before leaving as you mentionned they can give me buggy ones :p

chpego, 2025-05-04:
> [@chpego](https://github.com/chpego) I just hung up with IKEA support, it seems that the problem is localised to the strasbourg store, lucky us ! :) Support asked me to go back to the store to [...]

[Not only Strasbourg ](https://forum.hacf.fr/t/ikea-inspelning-detectee-comme-tretakt/61533/2)

nekomanta, 2025-05-04:
Damn ! :0

alju67, 2025-05-05:
> [Not only Strasbourg ](https://forum.hacf.fr/t/ikea-inspelning-detectee-comme-tretakt/61533/2)

Unfortunately yes. I also bought them at Strasbourg store !!!