Koenkk/zigbee2mqtt #24632: DIYRuZ_AirSense - controller:device: Device '0x00124b001ec6c625' is only compliant to revision 'pre-21' of the ZigBee specification (current revision: 23)

Creo2005, 2024-11-04:
### What happened?

DIYRuZ_AirSense stopped transmitting the CO2. 

### Zigbee2MQTT version

 1.41.0

### Adapter firmware version

20240710

### Adapter

SMARTLIGHT CC2652P (SLZB-02)

### Setup

add-on HA Supervised on Orange PI 3B 4Gb/32GB nvme

### Debug log

info 2024-11-04 15:36:47z2m: Interviewing 'Co2'
info 2024-11-04 15:36:47zh:controller:device: Device '0x00124b001ec6c625' is only compliant to revision 'pre-21' of the ZigBee specification (current revision: 23).
info 2024-11-04 15:36:48z2m: Successfully interviewed 'Co2'
info 2024-11-04 15:36:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{"id":"Co2"},"status":"ok","transaction":"rktj7-1"}'
info 2024-11-04 15:36:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Co2', payload '{"co2":null,"enable_abc":"ON","humidity":null,"humidity_offset":0,"led_feedback":"ON","linkquality":105,"pressure":null,"pressure_offset":0,"temperature":31.25,"temperature_offset":0,"threshold1":1000,"threshold2":2000}'

andye2801, 2024-11-18:
I have the same issue with a Phillips Hue dimmer, any fix or workaround for this?

Kiano950, 2024-11-25:
I have the same error  with a Tuya TH02Z Sensor, but so far without any problems.

amp88759, 2024-12-28:
I have the same with 3x Philips Hue Motion Sensors: model number 9290012607

Creo2005, 2024-12-29:
> ### What happened?
> DIYRuZ_AirSense stopped transmitting the CO2.
>
> ### What did you expect to happen?
> _No response_
>
> ### How to reproduce it (minimal and precise)
> _No response_
>
> ### Zigbee2MQTT [...]

[image]

for some reason, the converter version is not updated. Is very old - zigbee-herdsman-converters version
20.58.0, on github 21.10.0 (2024-12-28)

AddeoPL, 2025-01-16:
The same on my end with humidity RTX zigbee tuya humidity controler zh:controller:device: Device '0xa4c138a6d77dc08a' is only compliant to revision '21' of the ZigBee specification (current revision: 23).

egabor, 2025-01-17:
I have the same issue with a Moes thermostate. Device '0xa4c138ca28322f96' is only compliant to revision '21' of the ZigBee specification (current revision: 23). And also I can't access any of the thermostate settings in the Exposes tab.

LC44fun, 2025-01-22:
Same Issue with Tuya Garage door opener TS0603 (https://www.zigbee2mqtt.io/devices/TS0603.html#tuya-ts0603)

> zh:controller:device: Device '0xa4c1384a227c8815' is only compliant to revision '21' of the ZigBee specification (current revision: 23).

olgru, 2025-01-28:
Have the same with 3 Hue Outdoor sensors... Pairing reported as successful but no data, no binding - log shows the same line:
Device '0x00178801064772c5' is only compliant to revision 'pre-21' of the ZigBee specification (current revision: 23).

Creo2005, 2025-02-12:
With latest Zigbee2MQTT Edge.
CO2 value is 0. 
Recent changes to fix CO2 reflection not working properly on DIYRuZ_AirSense.
[image]
[image]

LeoS3, 2025-02-27:
Anyone with any updates on the pre-21 error?
I'm seeing those too, with Tuya SH-SC07 buttons.

Creo2005, 2025-02-27:
> Anyone with any updates on the pre-21 error? I'm seeing those too, with Tuya SH-SC07 buttons.

NO. 
Some sensors work normally even with this error. They wrote that they fixed the display of CO2 in z2m, but the sensor 
DIYRuZ_AirSense does not show CO2.

closed 2025-05-07 by github-actions[bot]