Koenkk/zigbee2mqtt #24636: Curtain/blind switch Tuya TS130F don't work after last 3 updates (lastest1.41.0 update )

JaCoLJcL, 2024-11-04:
### What happened?

3 devices TS130F (LoraTap SC400): https://www.zigbee2mqtt.io/devices/TS130F.html don't open/closed in HA after last 2 udpates. 
Working good current version is 1.40.1-1 

### What did you expect to happen?

Open/Closed for zigbee Curtain/blind switch

### Zigbee2MQTT version

1.40.1-1 

### Adapter firmware version

1.40.2

### Adapter

Sonoff Usb Zigbee Cc2531

### Setup

Rasperry pi 4, HAOS

pamkkl, 2024-11-09:
Hi,
not excactly the same problem, but probably related.
I use  Moes MS-108ZR / TS130F for shutter/blinds, since z2m 1.41.0 several issues with these devices appeared: Automations stopped working, wrong status is set (closed while open or vice versa), pressing close/open button does not work, but manually setting the the position works as expected.

Downgrade to 1.40.2-1 currently works.
I could provide any logs if necessary.
Cheers.

JaCoLJcL, 2024-11-10:
Hello, Yes this is the same problem that I have!

Koenkk, 2024-11-10:
Can you check if it is fixed in the  [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)?

pamkkl, 2024-11-10:
Thanks for the feedback. I guess this fixes the problem for me.
Open/Close button works with dev branch. Only after closing the blinds, the status remains "open" until I click on it, then it states "closed" immediately. But that might have been the case previously and is not a big deal for me.
So I will probably just wait for the next release and skip the current (or stay on dev until then).
Thanks for your great work, I really appreciate it!

JaCoLJcL, 2024-11-10:
> Open/Close button works with dev branch. Only after closing the blinds, the status remains &quot;open&quot; until I click on it, then it states &quot;closed&quot; immediately. But that might have been the case previously and is not a big deal for me.

Hmm but If you condition automation on closing the blind, this is already a problem 

If it can also be improved, I'm all for it, because I use such automation when turning on the light in the room if the blind is closed

Koenkk, 2024-11-10:
> Open/Close button works with dev branch. Only after closing the blinds, the status remains "open" until I click on it, then it states "closed" immediately.

Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

pamkkl, 2024-11-12:
Actually it seems, that I had this problem before switching to dev. I was trying to reproduce it now and could not. 
So I guess everyting is working as expected now. I will stay on dev until 1.42. is released.
Thanks again and sorry for wasting your time.

domome, 2024-11-15:
> Hi, not excactly the same problem, but probably related. I use Moes MS-108ZR / TS130F for shutter/blinds, since z2m 1.41.0 several issues with these devices appeared: Automations stopped working, [...]

Indeed, with the latest versions this is the behaviour. In my case, some automations do not work correctly because the status shown on the device is not correct, I have disabled everything while waiting for a correction.
Regards

pamkkl, 2024-11-15:
Both downgrading and dev branch work well for me.

Koenkk, 2024-11-17:
Closing as it seems to be fixed in the dev branch

closed 2024-11-17 by Koenkk

ErikSteiner, 2025-03-06:
The curtain switch does not work for several releases of Zigbee2MQTT. Latest version is 2.1.3-1.
The curtain can be controlled if the position is set to 0 or 100 % or if the close or open button is pressed. However, every position 0><100 will not move the cover. Interestingly the position is set on the device itself, if I look into the Zigbee2MQTT device. Any help is appropriated.

Koenkk, 2025-09-13:
This does not include the debug logging.

> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.