Koenkk/zigbee2mqtt #24637: Avatto wall thermostat - auto and manual modes inverted

Danieldiazi, 2024-11-04:
### What happened?

I've Avatto wall thermostat , TS0601 model, and  manufacturer "_TZE200_viy9ihs7"

When I choose "auto" mode into zigbee2mqtt interface, the device changes to "manual" and if I change to "manual" the device changes to "auto"

As far as I'm concerned, I think modes are inverted.

[image]

### What did you expect to happen?

zigbee2mqtt mode and devices mode must to be the same.

### How to reproduce it (minimal and precise)

Changing mode into web interface:
[image]

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240510

### Adapter

SLZB-06M

### Setup

Docker container

domome, 2024-11-12:
Same here

**Zigbee Model:** TS0601
**Zigbee Manufacturer:** _TZE204_lzriup1j

**Model:** ZWT198/ZWT100-BH 

**Zigbee2mqtt version:** 1.41.0

[image]

domome, 2024-12-04:
Hello 
with the new version 1.42.0 the problem has not been solved on the two devices I have.

Danieldiazi, 2025-01-06:
After this Pull Request : https://github.com/Koenkk/zigbee-herdsman-converters/pull/8446 it works correctly for my device

Danieldiazi, 2025-01-06:
> Same problem here since a few days ago.

Which is your manufacturer?, mine is "_TZE200_viy9ihs7" and it's resolved

Timmy93, 2025-01-09:
Guys, sorry to jump into this issue but I updated yesterday zigbee2mqtt to the latest version.
My thermostat (TS0601 - _TZE204_xnbkhhdr - Avatto wall thermostat)

From yesterday, when I set it to _manual_ it follows my _schedule_weekday_, instead when I set it to _auto_ it remains on until a certain temperature.

Is this behavior correct or it is an error?

dawogid, 2025-01-26:
I can confirm that this is still an issue for _TZE204_xnbkhhdr (Avatto).

Danieldiazi, 2025-01-27:
> I can confirm that this is still an issue for _TZE204_xnbkhhdr (Avatto).

I don't have that device (_TZE204_xnbkhhdr) but I think you can solve the problem modifying tuya.ts like I've done with my device.
If you can test, you can modify that file ( /app/node_modules/zigbee-herdsman-converters/devices) :

At line 7018 (in my case) you must replace

```
                        if (device.manufacturerName === '_TZE204_lzriup1j') {
                            return {auto: tuya.enum(1), manual: tuya.enum(0), temporary_manual: tuya.enum(2)};
                        } else {
                            return {auto: tuya.enum(0), manual: tuya.enum(1), temporary_manual: tuya.enum(2)};
                        }
```

to

```
                        if (device.manufacturerName === '_TZE204_lzriup1j' || device.manufacturerName === '_TZE204_xnbkhhdr') {
                            return {auto: tuya.enum(1), manual: tuya.enum(0), temporary_manual: tuya.enum(2)};
                        } else {
                            return {auto: tuya.enum(0), manual: tuya.enum(1), temporary_manual: tuya.enum(2)};
                        }
```

Create a backup before this change.

If it works, we can create a pull request.

Timmy93, 2025-01-27:
> > I can confirm that this is still an issue for _TZE204_xnbkhhdr (Avatto).
> 
> [...]
> Create a backup before this change.
> 
> If it works, we can create a pull request.

A user has already patched this problem [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8565/files) and the change has been merged into main branch.

So, watch for upcoming updates, keep in mind that your thermostat may stop working as it did in recent weeks.

terradefrikis, 2025-03-29:
It works now

> A user has already patched this problem [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8565/files) and the change has been merged into main branch.
>
> So, watch for upcoming updates, [...]

closed 2025-03-29 by Danieldiazi