Koenkk/zigbee2mqtt #24638: [New device support]: New manufactureName for Tuya ZDMS16 from Avatto

DmitryMyadzelets, 2024-11-04:
### Link

https://avatto-smart.com/

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0xa4c138d0e6e12a28","nwkAddr":22668,"manufId":4417,"manufName":"_TZE204_nqqylykc","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1730743371532}

### Zigbee2MQTT version

    1.41.0 commit: cc31cea

### Comments

The device is a dimmer that looks identical to [Avatto ZDMS16-1](https://www.zigbee2mqtt.io/devices/ZDMS16-1.html). The package's label says it's a model number DMS16/ZDMS16 "Smart Dimmer Module". Yet another sticker from the seller says it's "Avatto Tuya Zigbee Smart Dimmer 1 gang, model ZDMS16-Y1".

The dimmer's manufacture fingerprint differs from the already supported one. It may be enough to add the new fingerprint to the existing one, I believe.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_nqqylykc',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

The device is "Unsupported". Nothing works.

DmitryMyadzelets, 2024-11-05:
It's supported now. I've forgotten to reload the service after updating it:
```bash
systemcl restart zigbee2mqtt
```

closed 2024-11-05 by DmitryMyadzelets

EisDaveLO, 2024-11-26:
For me it remain to be unsupported?

1.41.0-dev commit: 267d800
herdsman converter version: 20.54.0