Koenkk/zigbee2mqtt #24640: After updating the firmware of the SLZB-06M and switching to ember, no devices are accessible.

Kruemelino, 2024-11-04:
### What happened?

I updated the firmware of the SLZB-06M to v2.5.8 and installed Zigbee Coordinator 20240510. The goal was to migrate from ezsp to ember. So far so good. However, all devices are no longer accessible. Error messages are displayed:
```warning:  z2m: Failed to ping 'Esstisch' (attempt 1/1, ZCL command 0xa4c1386a43020808/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '18688'.)) ```

There is also no way back. I tried to install the old firmware. No chance. The devices don't respond.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

v2.5.8 (7.4.1 [GA])

### Adapter

SLZB-06M

### Setup

Docker Container

### Debug log

I can see one error in debug log```
[2024-11-04 19:31:16] debug:    zh:controller:endpoint: Error: ZCL command 0x001788010929bde2/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '26079'.)
    at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31)
    at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:469:42)
    at Ezsp.emit (node:events:517:28)
    at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5032:14)
    at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:735:26)
    at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:395:22)
    at UartAsh.emit (node:events:517:28)
    at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45)
    at processImmediate (node:internal/timers:476:21)
[2024-11-04 19:31:16] warning:  z2m: Failed to ping 'Schlafzimmer 2' (attempt 1/1, ZCL command 0x001788010929bde2/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '26079'.))
```
[log (1).log](https://github.com/user-attachments/files/17622506/log.1.log)

nickdesi, 2024-11-05:
Yep same issue for me since yesterday with , I get timeout and table full errors error
 2024-11-05 19:17:16z2m: Failed to read state of 'Ampoule' after reconnect (ZCL command 0x8c6fb9fffeaab47e/1 genLevelCtrl.read(["currentLevel"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '42267'.))

So many errors and now my devices are not responding anymore (coordinator Sonoff  ZigBee 3 dongle-E latest version firm. 7.4.4.0 z2m version 1.41)

jonwilliams84, 2024-11-06:
I also updated to 2.58 yesterday, albeit with mDNS disabled. It worked fine on ezsp driver for a few hours but keep getting this in my logs:

```
error 2024-11-06 11:01:41zh:ezsp:uart: --> Error: Error: {"sequence":1} after 4000ms
error 2024-11-06 11:01:41zh:ezsp:uart: -!- break waiting (1)
error 2024-11-06 11:01:41zh:ezsp:uart: Can't send DATA frame (0,1,0): 580001340000007304010600010100010000292a0301dd01
error 2024-11-06 11:08:52zh:ezsp:uart: --> Error: Error: {"sequence":4} after 4000ms
error 2024-11-06 11:08:52zh:ezsp:uart: -!- break waiting (4)
error 2024-11-06 11:08:52zh:ezsp:uart: Can't send DATA frame (3,7,0): 9b00010500
```

Although the network still works, it is noticeably slower sometimes and some of my devices aren't responding occasionally.

I am going to flash back but stick with ezsp. On Ember on 2.56 I was getting the same errors as @nickdesi and @Kruemelino

Kruemelino, 2024-12-04:
After reseting and reconnecting every device: the problem is solved