Koenkk/zigbee2mqtt #24646: SPLZB-131 Incorrect reporting values

sdalu, 2024-11-04:
### What happened?

The interview seems to succeed, but reporting value is either empty or a short list with max rep interval being 65000

After initial pairing
[image]

After explicitly asking for an interview (the list is missing some cluster, max rep interval is 65000)
[image]

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Z2M on RPI

### Debug log

~~~
[2024-11-04 22:58:57] error:    z2m: Failed to configure '0x0015bc002f011682', attempt 1 (Error: ZCL command 0x0015b
c002f011682/1 genOnOff.configReport([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"
attribute":"onOff"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true
,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 27299 - 1 - 65 - 6 - 7 after 10000ms)
    at Object.start (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/
src/adapter/z-stack/adapter/zStackAdapter.ts:640:47)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/
src/adapter/z-stack/adapter/zStackAdapter.ts:672:28)
    at /home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:4
22:20
    at Queue.execute (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/adap
ter/z-stack/adapter/zStackAdapter.ts:420:16)
    at Request.send (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.t
s:82:20)
    at Endpoint.sendRequest (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/end
point.ts:325:20)
    at Endpoint.zclCommand (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endp
oint.ts:951:28)
    at Endpoint.configureReporting (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/mo
del/endpoint.ts:688:9))
~~~

torandreroland, 2024-11-14:
I'm seeing the same thing. This is because the device has two endpoints with cluster 0x0006, endpoint 0x01 with application profile Id 0xC0C9 (Develco Products private profile) and endpoint 0x02 with application profile Id 0x0104 (Home Automation).

@Koenkk: Any way of making for example [this function](https://github.com/Koenkk/zigbee-herdsman-converters/blob/e20dac00520d4ef74c8a1388e4abbd1f212abb8a/src/lib/modernExtend.ts#L54) profileId aware so that it only tries to configure reporting for the correct endpoint?

sdalu, 2024-12-04:
Any hotfix that could be applied? Problem persists in 1.42.0

closed 2025-02-11 by github-actions[bot]