Koenkk/zigbee2mqtt #24650: [New device support]:

h0m3d0m0, 2024-11-05:
### Link

https://www.aliexpress.com/item/1005004627144844.html?spm=a2g0o.order_detail.order_detail_item.5.5d24f19cqmliCQ

### Database entry

{"id":31,"type":"Router","ieeeAddr":"0xe498bbe3615e3b9a","nwkAddr":20152,"manufId":4660,"manufName":"_TZE200_ojtqawav","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,61184],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":2,"hwVersion":3,"dateCode":"20210625","swBuildId":"600+TZZUT11HH102","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1730813505203}

### Zigbee2MQTT version

1.40.2

### Comments

Device not supported.
Try external convertor but no luck

[image]

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_ojtqawav',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

device paired but nothing works

ehwwong, 2024-11-13:
Unfortunately i bought the same switch as well and now having the same issue. Did you manage to resolve it?

h0m3d0m0, 2024-11-16:
no try everything but no solution for the momenent

closed 2025-01-23 by github-actions[bot]