Koenkk/zigbee2mqtt #24651: Fails to connect to ConbeeII

johnsprogs, 2024-11-05:
### What happened?

On start-up z2m gives error: 
`Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'getDevice')
    at Bridge.publishInfo (/opt/zigbee2mqtt/lib/extension/bridge.ts:746:69)...`
Tried serial/by-device and dev/tty..., makes no difference.
Stick worked with Deconz on previous Pi, but now making a fresh install from scratch.
My settings in configuration.yaml:
`permit_join: true
homeassistant: true
frontend:
  port: 8090
mqtt:
  broker: core-mosquitto
  port: 1883
  server: mqtt://localhost
  discovery: true
  base_topic: zigbee2mqtt
  discovery_prefix: Zigbee2MQTT
birth_message:
    topic: 'hass/status'
    payload: 'online'
will_message:
    topic: 'hass/status'
    payload: 'offline'
server: mqtt://localhost
serial:
  adapter: deconz
  port: >-
   /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2228982-if00
advanced:
  log_level: debug
  network_key:
    - 197
    - 208
    - 70
    - 175
    - 155
    - 204
    - 176
    - 141
    - 99
    - 152
    - 227
    - 244
    - 168
    - 247
    - 166
    - 252
  pan_id: 34867
  ext_pan_id:
    - 3
    - 207
    - 62
    - 224
    - 189
    - 59
    - 129
    - 105
`

### What did you expect to happen?

It would connect properly. Earlier in the process the Conbee II seems to be recognised:
`Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":120,"product":0,"revision":"0x26780700","transportrev":0},"type":"ConBee2/RaspBee2"}'`

### How to reproduce it (minimal and precise)

Just starting it up with npm start

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

26780700

### Adapter

Conbee II

### Setup

Running on Pi 4B, At the moment plain, will later connect to Domoticz

### Debug log

[log.log](https://github.com/user-attachments/files/17634004/log.log)

medabida, 2024-11-08:
i am having the same problem as well
i tried 2 different conbee || firmwares (`0x26720700` and `0x26780700`) without any luck.

`this.zigbee.firstCoordinatorEndpoint()` seems to be undefined..
https://github.com/Koenkk/zigbee2mqtt/blame/cc31ceabd1c29bc1f266640cc6e6bea95d9d6db3/lib/extension/bridge.ts#L746

medabida, 2024-11-09:
Deleting database.db prompted Zigbee2MQTT to generate a new one, which resolved the issue.

This is my initial database.db 
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00212effff093c77","nwkAddr":0,"manufId":4405,"epList":[],"endpoints":{},"interviewCompleted":true,"meta":{}}
```

and this is the newly generated database.db

```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00212effff093c77","nwkAddr":0,"manufId":4405,"epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[0,1536,10,25,1281],"outClusterList":[1,32,1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":2080,"inClusterList":[0],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{}}
```

Since this is my first time using Zigbee2MQTT, I’m not entirely sure why this change fixed the issue or what each property represents. My guess, however, is that when Zigbee2MQTT is initially installed and the coordinator isn’t properly configured, it creates a partial configuration and saves it to database.db to reuse in the future, which then causes issues. Regenerating database.db after the device is correctly set up seems to resolve this, as the database then reflects the proper configuration.

If someone with more Zigbee2MQTT experience could explain this further, I’d be happy to provide additional details and investigate more.

closed 2024-12-07 by johnsprogs

michaelbrusegard, 2024-12-23:
> Deleting database.db prompted Zigbee2MQTT to generate a new one, which resolved the issue.
>
> This is my initial database.db
>
> ``` [...]

I am having this exact same issue, deleting the db did not fix it

JoeArny, 2025-01-03:
> I am having this exact same issue, deleting the db did not fix it

Same for me...

michaelbrusegard, 2025-01-04:
It did work for me after restarting it and deleting the database many times. I dont know why it did not work every time