Koenkk/zigbee2mqtt #24652: eWeLink SQ510A leak sensor detector as Sonoff motion sensor

Claudybellic, 2024-11-05:
### What happened?

Just paired an eWeLink SQ510A leak sensor and Z2MQTT detected it as a sonoff SNZB-03. No values can be read from the device.

### Zigbee2MQTT version

    1.41.0-dev commit: 1558a5d

### Adapter firmware version

7.3.1.0 build 0

### Adapter

EZSP ember

### Setup

add-on on HAOS on a VM (Truenas)

Claudybellic, 2024-11-05:
device external definition:

const {battery, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SNZB-03'],
    model: 'SNZB-03',
    vendor: 'eWeLink',
    description: 'Automatically generated definition',
    extend: [battery(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};

module.exports = definition;

LuxMcDiver, 2024-11-06:
Hi,
I just paired an Aliexpress Motion sensor labelled ZM-35ZH-Q, and Z2MQTT detected it as an Ewelink SQ510A water leak sensor.

I'm using koenkk/zigbee2mqtt:latest docker-compose image.

It looks like we have a generic issue with the Z2MQTT Ewelink SQ510A water leak sensor seeting in the database.

🤔

Fietspomp86, 2024-11-07:
> Hi, I just paired an Aliexpress Motion sensor labelled ZM-35ZH-Q, and Z2MQTT detected it as an Ewelink SQ510A water leak sensor.
>
> I'm using koenkk/zigbee2mqtt:latest docker-compose image.
>
> It looks [...]

I'm having the same issue.
Had one occupancy sensor, since a few days it's suddenly changed into a water leak sensor.

`const definition = {
    zigbeeModel: ['SNZB-03'],
    model: 'SNZB-03',
    vendor: 'eWeLink',
    description: 'Automatically generated definition',
    extend: [battery(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},`

vikrilow, 2024-11-24:
I've just try to add motion sensor from Aliexpress and I have this problem, Zigbee2mqtt detect it as water leak..
Any solution for that?

closed 2025-02-01 by github-actions[bot]