Koenkk/zigbee2mqtt #24661: [New device support]: Zemismart USB IR Blaster

MiAutomations, 2024-11-06:
### Link

https://www.zemismart.com/products/zm-18-usb

### Database entry

{"id":189,"type":"Router","ieeeAddr":"0x6cfd22fffecaadf2","nwkAddr":33178,"manufId":4098,"manufName":"_TZ3290_uc8lwbi2","powerSource":"Mains (single phase)","modelId":"TS1201","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":61440,"inClusterList":[0,4,5,3,60672,57348,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS1201","manufacturerName":"_TZ3290_uc8lwbi2","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1730904878149}

### Zigbee2MQTT version

 1.41.0

### Comments

Hello I try to add this device but the device appears as not supported 

Can you please add the support for this device? 

Thank you 

### External definition

```shell
const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS1201'],
    model: 'TS1201',
    vendor: '_TZ3290_uc8lwbi2',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

[image]

fabianosan, 2024-12-20:
Here is a converter done by @19fabiom84 on this [thread](https://github.com/Koenkk/zigbee2mqtt/issues/24160#issuecomment-2479061216).

The sensors and controls show up in Z2M and Home Assistant, but I can't know how to use it, because it not shows as a remote controle in action or development console to learn IR code.

[TS1201.js.txt](https://github.com/user-attachments/files/18211850/TS1201.js.txt)
*** rename to *.js and add as external converter in Z2M.

19fabiom84, 2024-12-20:
Hello, I think you can send MQTT command to activate learn and other MQTT command to send IR command. You don't need it appear as remote control in action.

closed 2025-02-26 by github-actions[bot]