Koenkk/zigbee2mqtt #24663: Unable to update firmware

ncortines, 2024-11-06:
### What happened?

Hi,
I have got a [Philips 9290012573A](https://www.zigbee2mqtt.io/devices/9290012573A.html#philips-9290012573a)
The current firmware version 1.19.0_r19755 with build date 20161206 (as reported by zigbee2mqtt).
I would like to update to a newer firmware version so I can update the `hue_power_on_behavior`
However, zigbee2mqtt reports that there is no upgrade available (please see screenshot)

[image]

Anything I can do about this?

Zigbee2MQTT version 1.40.2 commit: e06848db
Coordinator type zStack3x0
Coordinator revision 20220219
Frontend version 0.7.4
zigbee-herdsman-converters version 20.21.0
zigbee-herdsman version 2.1.3

Thanks

### What did you expect to happen?

I expected that zigbee2mqtt would allow me to upgrade the firmware version

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20220219

### Adapter

Sonoff ZigBee 3.0 USB CC2652P

### Setup

self-hosted, docker

ncortines, 2024-11-07:
weird, the model number printed on the bulb is `9290012575A` (z2m recognises it as `9290012573A` instead)

ncortines, 2024-11-07:
ok, I think it's incorrectly recognised by z2m. 

The model is LCT015/9290012575A, but it is recognised as LCT016/9290012573A

FullHD23, 2024-12-25:
Some problem for me but with different Firmware Version. Coming to 99,39% progress and then abort.

  Hersteller | Modell | Firmware-Datum | Firmware-Version |  
  Philips | 9290012573A | 20221006 | 67.101.2

```
[2024-12-25 12:53:36] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Deckenfluter', payload '{"brightness":254,"color":{"x":0.5267,"y":0.4133},"color_mode":"color_temp","color_temp":500,"linkquality":192,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":1124099330,"latest_version":1124103171,"state":"available"},"update_available":null}'
[2024-12-25 12:53:36] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Deckenfluter"},"error":"Update of 'Deckenfluter' failed (Update failed with reason: 'ABORT')","status":"error","transaction":"q8df0-1"}'
[2024-12-25 12:53:36] [31merror[39m: 	z2m: Update of 'Deckenfluter' failed (Update failed with reason: 'ABORT')
[2024-12-25 12:53:36] [34mdebug[39m: 	z2m: Error: Update failed with reason: 'ABORT'
    at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:751:15)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at runNextTicks (node:internal/process/task_queues:64:3)
    at processImmediate (node:internal/timers:449:9)
    at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/zigbeeOTA.ts:126:12)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:282:41)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17)
```

Attached the log with 500 lines if it may helps

[45df7312_zigbee2mqtt_2024-12-25T11-56-56.892Z.log](https://github.com/user-attachments/files/18245801/45df7312_zigbee2mqtt_2024-12-25T11-56-56.892Z.log)

closed 2025-03-06 by github-actions[bot]