Koenkk/zigbee2mqtt #24668: ONENUO Tuya Zigbee Contact Sensor reports false state after recent fix #24305

pushpinderbal, 2024-11-06:
### What happened?

The ONENUO zigbee sensor(_TZ3000_n2egfsli) was previously being reported as a SONOFF SNZB-04, which was deemed a cosmetic issue and fixed as reported on https://github.com/Koenkk/zigbee2mqtt/issues/24305
Related commit: https://github.com/Koenkk/zigbee-herdsman-converters/commit/a551de6956b5e12f35c5183f236505a20808c023

After the upgrade to 1.41.0 the device now reports incorrect state. For example, if the door is left open for too long, it changes to closed after anytime between 1min-1hours. At first I thought this was device related, I changed batteries and reset the device multiple times, but then started to notice the behaviour with multiple devices of the same model.

### What did you expect to happen?

The door contact sensor matches the original state.

### How to reproduce it (minimal and precise)

Leve the contact sensor in an open state, it will close after some time on it's own.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

SONOFF ZBDongle-E

### Setup

HomeAssistant on RockPro64 in docker

### Debug log

relevant log message is at `2024-11-06 18:10:49` from device 'Back Door' when it reported as closed even though it was in an open state physically

[debug.log](https://github.com/user-attachments/files/17654197/debug.log)

pushpinderbal, 2024-11-08:
I wonder if this is related: https://github.com/Koenkk/zigbee-herdsman-converters/commit/7e8a807a334c4f21a59ef8770ceee935a9fa4c3a

pushpinderbal, 2024-11-10:
It's caused by the use of modern extends in the latest z2m version. I've reverted to an external definition from the previous version which seems to be working fine.

```js
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const definition = {
    fingerprint: [
        // ModelID is from the temperature/humidity sensor (SNZB-02) but this is SNZB-04, wrong modelID in firmware?
        // https://github.com/Koenkk/zigbee-herdsman-converters/issues/1449
        {
            type: 'EndDevice',
            manufacturerName: 'eWeLink',
            modelID: 'TH01',
            endpoints: [{ID: 1, profileID: 260, deviceID: 1026, inputClusters: [0, 3, 1280, 1], outputClusters: [3]}],
        },
    ],
    zigbeeModel: ['DS01', 'SNZB-04'],
    model: 'SNZB-04',
    vendor: 'SONOFF',
    whiteLabel: [{vendor: 'eWeLink', model: 'RHK06'}, tuya.whitelabel('Tuya', 'WL-19DWZ', 'Contact sensor', ['_TZ3000_n2egfsli'])],
    description: 'Contact sensor',
    exposes: [e.contact(), e.battery_low(), e.battery(), e.battery_voltage()],
    fromZigbee: [fz.ias_contact_alarm_1, fz.battery],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryVoltage(endpoint, {min: 3600, max: 7200});
        await reporting.batteryPercentageRemaining(endpoint, {min: 3600, max: 7200});
    },
};

module.exports = definition;
```

A wild guess based on what I have been looking at is that the device sends an attribute report with the payload as -> `"payload":[{"attrId":61441,"dataType":32,"attrData":1}]` which is not parsed correctly by `iasZoneAlarm` setting the `zoneStatus` to undefined here, `alarm1Payload` is then computed to an incorrect value.
https://github.com/Koenkk/zigbee-herdsman-converters/blob/15835f6fa2e9658e6dbb1f1599c5ab88b7051622/src/lib/modernExtend.ts#L1519

Koenkk, 2024-11-10:
Thanks for investigating, pushed a fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-11-10 by Koenkk