Koenkk/zigbee2mqtt #24669: Duplicate reporting attributes

nplawes, 2024-11-06:
### What happened?

This feels the same issue that I had before, that was previously fixed.

When editing a reporting attribute from the UI or via a mosquitto_post, I end up with a duplicate entry for the cluster/attribute.
Looking at the db file I can see that all of the reporting nodes on the device have a null manufacturerCode entry, and the duplicated updated attribute does not have this entry.

For example, when I changed cluster 1026 repChange value from 10 to 11, this is the result:
```
{
  "cluster": 1030,
  "attrId": 0,
  "minRepIntval": 0,
  "maxRepIntval": 3600,
  "repChange": 0,
  "manufacturerCode": null
}
{
  "cluster": 1,
  "attrId": 33,
  "minRepIntval": 900,
  "maxRepIntval": 3600,
  "repChange": 0,
  "manufacturerCode": null
}
{
  "cluster": 1024,
  "attrId": 0,
  "minRepIntval": 60,
  "maxRepIntval": 3600,
  "repChange": 200,
  "manufacturerCode": null
}
{
  "cluster": 1026,
  "attrId": 0,
  "minRepIntval": 300,
  "maxRepIntval": 3600,
  "repChange": 10,
  "manufacturerCode": null
}
{
  "cluster": 1026,
  "attrId": 0,
  "minRepIntval": 300,
  "maxRepIntval": 3600,
  "repChange": 11
}
```

Both of the cluster 1026 appear in the UI list.

10 of my 13 devices of this type (Hue motion sensors) seem to be in this state and have the same issues. The remaining 3 do not have the manufacturerCode in them, and behave as expected.

Using the UI to disable the bad attribute with value 10, simply removed the good attribute with the value 11.

This sounds like the attributes with the manufacturerCode are not being processed.

### What did you expect to happen?

I expected the original cluster/attribute to be replaced by the new one

### How to reproduce it (minimal and precise)

Change any attribute on an affected device gets a duplicate entry. Deleting either of those entries will remove the new entry, leaving the original one.

### Zigbee2MQTT version

1.41.0 commit: cc31cea

### Adapter firmware version

20230507

### Adapter

Electrolama zzh

### Setup

Z2M, mqtt, home assistant all docker on the same Linux mini PC running Rocky 8.10, docker-ce. All software at the latest version

Great-Chart, 2024-11-07:
I've noted similar and equally have recollections of this being fixed previously.
There is a lot of information covered in https://github.com/Koenkk/zigbee2mqtt/issues/23933 too (still open and thus pending a further fix it appears).

Unclear (to me) if it's an issue in front end or Z2M itself - refer https://github.com/nurikk/zigbee2mqtt-frontend/issues/2125

My example for completeness (power monitoring plug):-
[image]

nplawes, 2024-11-07:
I also get the issue when updating directly via mosquitto_post, so I assume it's not the UI itself. If the UI uses mosquitto_post or an equivalent library call, then it could be in the mqtt processing somewhere.

I guess the question is what added the manufacturerCode, and can it be ignored when processing the attributes...

alexh3o, 2025-01-07:
Still an issue:
[image]

I have this in my database, issue being on cluster 1 attrId 33
`{"id":108,"type":"EndDevice","ieeeAddr":"0x3c2ef5fffe42181f","nwkAddr":29756,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"PARASOLL Door/Window Sensor","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":259,"inClusterList":[0,1,3,32,2821,4096,64636,64641],"outClusterList":[3,4,6,25,4096],"clusters":{"genPollCtrl":{"attributes":{"checkinInterval":172800}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":194}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c82c76","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c82c76","endpointID":1},
{"cluster":1280,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c82c76","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0,"manufacturerCode":null},{"cluster":1,"attrId":33,"minRepIntval":"3600","maxRepIntval":"65000","repChange":"10"}],"meta":{}},"2":{"profId":260,"epId":2,"devId":1026,"inClusterList":[0,3,1280],"outClusterList":[3],"clusters":{"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024c82c76","zoneState":1,"zoneStatus":0}}},"binds":[{"cluster":1280,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c82c76","endpointID":1}],"configuredReportings":[{"cluster":1280,"attrId":2,"minRepIntval":"0","maxRepIntval":"7200","repChange":"0"}],"meta":{}}},"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20230516","swBuildId":"1.0.19","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1736288164273,"checkinInterval":43200}`

nplawes, 2025-01-08:
There is a `"manufacturerCode": null` in there which was causing my grief... I ended up manually fixing the database, and the problem has not (yet) returned.

However I have had this issue in the past (and there was an update which fixed it) but it did eventually return recently. I still have no idea what caused it either time, so it may well return...

closed 2025-03-17 by github-actions[bot]