Koenkk/zigbee2mqtt #24671: Since version 1.40.2 and after few hours, unable to change state of lights

thomaschampagne, 2024-11-07:
### What happened?

Since version 1.40.2, and after few hours (maybe 6h-8h), turning on/off lights from wall switches don't work anymore.

It affects also version 1.41.0

No issue with 1.40.1 : I rollback to version 1.40.1 on which turning on/off light keep working after few hours

### What did you expect to happen?

Turning on/off lights from wall switches still working after few hours or a night

### How to reproduce it (minimal and precise)

This is a challenging issue to replicate due to its time-dependent nature. After a week of investigation, I managed to isolate the problem :

The issue pattern:

- Initial setup: Deploying either zigbee2mqtt version 1.40.2 or 1.41.0
- Evening operation: Light control (on/off) works normally
- Next morning: Light control functionality breaks down

The debug logs provided below should help understand the problem better I think

I made clean install of home-assistant, mosquitto. Tried mixed combinations of various versions of HA, mosquito, Z2M until I found 1.40.2 was the issue.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20211115

### Adapter

Texas Instrument CC2531

### Setup

Home-Assistant 2024.10.4, MQTT eclipse-mosquitto 2.0.20, Kubernetes 1.29.9 (k3s), RockyLinux 9.4 (Intel Core i5)

### Debug log

You will see 2 debug logs streams here : v1.40.1 OK & 

Logs are focusing on a single click of `wc_wallsw` (Wallswitch) entity which must turn on the `wc_spot` (Light).

- v1.40.1 : OK 

```text
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     zh:controller: Received payload: clusterID=18, address=32491, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=81, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":187,"commandIdentifier":10},"payload":[{"attrId":85,"dataType":33,"attrData":1}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     zh:zstack:unpi:parser: --- parseNext ]
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     z2m: Received Zigbee message from 'wc_wallsw', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/wc_wallsw', payload '{"action":"single","battery":100,"click":"single","device_temperature":26,"linkquality":81,"power_outage_count":75,"voltage":3015}'
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/wc_wallsw', payload '{"action":"","battery":100,"device_temperature":26,"linkquality":81,"power_outage_count":75,"voltage":3015}'
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/wc_wallsw', payload '{"action":null,"battery":100,"click":"","device_temperature":26,"linkquality":81,"power_outage_count":75,"voltage":3015}'
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/wc_wallsw/action', payload 'single'
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/wc_wallsw/click', payload 'single'
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     z2m:mqtt: Received MQTT message on 'zigbee2mqtt/wc_spot/set' with data '{"state":"OFF"}'
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     z2m: Publishing 'set' 'state' to 'wc_spot'
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     zh:controller:endpoint: ZCL command 0x000d6ffffe5ba0a0/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     zh:zstack: sendZclFrameToEndpointInternal 0x000d6ffffe5ba0a0:24257/1 (0,0,1)
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":24257,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":241,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,121,0]}}
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     zh:zstack:unpi:writer: --> frame [254,13,36,1,193,94,1,1,6,0,241,0,30,3,1,121,0,37]
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
home-zigbee2mqtt-dm-t2rm8 zigbee2mqtt [2024-11-07 11:05:02] debug:     zh:zstack:unpi:parser: --- parseNext ]
[... 41 lines cut ...]
[2024-10-31 13:37:00] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 196 - [247,123,0] - 14
[2024-10-31 13:37:00] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd - {"dstaddr":31735,"relaycount":0,"relaylist":[]}
[2024-10-31 13:37:00] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-31 13:37:00] debug: 	zh:zstack:unpi:parser: <-- [254,32,68,129,0,0,25,0,247,123,1,13,0,73,0,194,54,219,0,0,12,1,187,1,0,126,17,0,1,1,0,0,0,247,123,29,83]
[2024-10-31 13:37:00] debug: 	zh:zstack:unpi:parser: --- parseNext [254,32,68,129,0,0,25,0,247,123,1,13,0,73,0,194,54,219,0,0,12,1,187,1,0,126,17,0,1,1,0,0,0,247,123,29,83]
[2024-10-31 13:37:00] debug: 	zh:zstack:unpi:parser: --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,247,123,1,13,0,73,0,194,54,219,0,0,12,1,187,1,0,126,17,0,1,1,0,0,0,247,123,29] - 83
[2024-10-31 13:37:00] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":31735,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":73,"securityuse":0,"timestamp":14366402,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,187,1,0,126,17,0,1,1,0,0,0]}}
[2024-10-31 13:37:00] debug: 	zh:controller: Received payload: clusterID=25, address=31735, groupID=0, endpoint=1, destinationEndpoint=13, wasBroadcast=false, linkQuality=73, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":187,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":4478,"imageType":256,"fileVersion":1},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}
[2024-10-31 13:37:00] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-31 13:37:00] debug: 	z2m: Received Zigbee message from 'kit_spots_e', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1,"imageType":256,"manufacturerCode":4478}' from endpoint 1 with groupID 0
[2024-10-31 13:37:00] debug: 	z2m: Device 'kit_spots_e' requested OTA
[2024-10-31 13:37:00] debug: 	zh:controller:endpoint: CommandResponse 0x00158d00047683cf/1 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false})
[2024-10-31 13:37:01] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,196,247,123,0,14]
[2024-10-31 13:37:01] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,196,247,123,0,14]
[2024-10-31 13:37:01] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 196 - [247,123,0] - 14
[2024-10-31 13:37:01] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd - {"dstaddr":31735,"relaycount":0,"relaylist":[]}
[2024-10-31 13:37:01] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-31 13:37:01] debug: 	zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,6,0,247,123,1,1,0,76,0,251,67,219,0,0,7,8,188,10,0,0,16,0,247,123,29,67]
[2024-10-31 13:37:01] debug: 	zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,6,0,247,123,1,1,0,76,0,251,67,219,0,0,7,8,188,10,0,0,16,0,247,123,29,67]
[... cut ...] ort","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-10-31 13:37:01] debug: 	zh:controller:endpoint: ZCL command 0x00158d00047683cf/1 genOnOff.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":188,"writeUndiv":false})
[2024-10-31 13:37:01] debug: 	zh:zstack:unpi:parser: --- parseNext []
```

lux73, 2024-11-09:
please do NOT use CC2531 anymore - it's really weak in 2024 and not worth to deal with

do consider Upgrade to any [CC2652x Coordinator](https://www.zigbee2mqtt.io/guide/adapters/zstack.html#recommended-cc2652-cc1352) - these are more stable & powerful

CC2531 works as Router flawless

thomaschampagne, 2024-11-09:
> please do NOT use [CC2531 anymore](https://www.zigbee2mqtt.io/guide/adapters/zstack.html#not-recommended-cc2530-cc2531-cc2538) - it's really weak in 2024 and not worth to deal with
>
> do consider [...]

Ok but this is not the solution to this problem. Many people are probably still using the CC2531 and it should be also fixed for project consistency no?

lux73, 2024-11-09:
for sure this is the Solution - i doubt anyone is willing to try fix Problems for the outdatet CC2531

s-y-o-t-o-s, 2024-11-09:
> for sure this is the Solution - i doubt anyone is willing to try fix Problems for the outdatet CC2531

I have been using zigbee2mqtt for over 6 years (started on version 0.2.0) with a CC2531 without much issues. Back then it was basically the only supported option. I understand that there are now more (and better options) and before this issue happened already ordered a new coordinator based on CC2652.

Phasing out support for CC2531 is something that could happen but would need a proper plan and communication to not alienate loyal zigbee2mqtt users who have been using the project for years with a device that was recommended when they started out.

Just not fixing issues because it is old is certainly not the way to go.

lux73, 2024-11-09:
thats true but for now CC2531 is not recommended anymore (and definitely for more than a Year)  -> https://www.zigbee2mqtt.io/guide/adapters/zstack.html#not-recommended-cc2530-cc2531-cc2538

I can certainly understand if some People find this stupid, but that's the way it is - Sometimes it's time to shed old braids

s-y-o-t-o-s, 2024-11-09:
I have no problem with deprecating old hardware, it makes sense from a maintenance perspective. My only point is that is should be done thoughtfully with a good (communication) plan. I did see that the CC2531 was "not recommended" anymore, but to me that indicated something completely different than deprecated/no longer supported.

thomaschampagne, 2024-11-09:
@lux73 I’ve decided to upgrade my hardware after you pointed out that the CC2531 is deprecated and not recommended.

I found this **CC2652P** dongle: https://www.aliexpress.us/item/3256805998875331.html

There's 2 chips variant : CP2102N or CH9102X. 

Do you know which chip I should choose? How compatible are  CP2102N/CH9102X with Zigbee2MQTT?

Thanks for your feedback

lux73, 2024-11-09:
Q1: i wish i could - i only have Experience with [CC2652RB](https://slae.sh/projects/cc2652/) - for me this Stick works very reliable & fast with latest FW 20240710 with actually 59 Devices

btw i have started with CC2531 few Years ago - but from the beginning I kept having inexplicable problems with it

Q2: any USB Extension Cable should work (and really recommended even with CC2652x based Coordinator) - no matter if 2.0 or 3.x Cable

thomaschampagne, 2024-11-09:
> Q1: i wish i could - i only have Experience with [CC2652RB](https://slae.sh/projects/cc2652/) - for me this Stick works very reliable & fast with latest FW 20240710 with actually 59 Devices
>
> btw i [...]

Thanks for your help. I will look after CC2652RB too

sector101010, 2024-11-17:
Same problem here, but with adapter **slzb-06m**:

ESP32 firmware ver: 	0.9.9 (Sep 5 2023)
ESP32 version: 	ESP32-D0WDQ5

Bruceforce, 2024-11-17:
> for sure this is the Solution - i doubt anyone is willing to try fix Problems for the outdatet CC2531

I'm experiencing the same issue with the CC2531. I totally agree with @thomaschampagne that this is clearly a bug and buying new hardware is in my opinion *not* the solution for this issue, but a workaround.

> I have no problem with deprecating old hardware, it makes sense from a maintenance perspective. My only point is that is should be done thoughtfully with a good (communication) plan. I did see that [...]

This. If support for CC2531 is really dropped, it should be marked as unsupported in the docs and zigbee2mqtt should at least log a warning saying "CC2531 is not supported anymore since version X. If you are experiencing malfunction please refer to the wiki and buy a new adapter".

Just to clarify: In the meantime I also bought another adapter, but I would still suggest to officially drop support for the CC2531 so that all people with the same issues are aware and don't waste hours in debugging. But it's on the maintainer(s) to decide if it's really unsupported or they want to fix the issue.

thomaschampagne, 2024-11-17:
> I'm experiencing the same issue with the CC2531. I totally agree with @thomaschampagne that this is clearly a bug and buying new hardware is in my opinion _not_ the solution for this issue, but a [...]

Exactly. Bug should be fixed on principle or CC2531 marked unsupported since 1.40.2 version in documentation.

By the way I bought this new **CC2652P** adapter (CP2102N Chip) for 15 bucks on ali => https://aliexpress.com/item/1005006185190083.html

purche42, 2024-11-18:
This seems not to be a problem caused by the old stick ... I use a CC2652RB stick with brand new firmware and see similiar issues after upgrade to 1.40.2. 

Problem occurs after several - unknown how many - hours. Restart fix it temporarily . Only downgrade to 1.40.1 solved problem but is not a permanent fix.

See also https://github.com/Koenkk/zigbee2mqtt/issues/24293

frankivo, 2024-11-19:
Same on ConBee II

Bruceforce, 2024-11-19:
Just for the records: I replaced the CC2531 with a Sonoff ZBdongle-E and this issue is gone. At least it's working since three days now while the CC2531 had this issue every 12 hours or so.

Koenkk, 2024-11-20:
Support for CC2531 was definitely not dropped, even though its weak and known to easily crash, an update of z2m alone should not make it suddenly crash. It seems there is some regression since 1.40.2. My gut feeling is that 1.40.2 puts more load on the coordinator which makes it crash.

To start debugging this, in the location where z2m is installed or in the container run: `npm install zigbee-herdsman@0.57.3`. This is the zigbee-herdsman version that came with 1.40.1. Now restart the container and in the frontend -> settings -> about tab verify that the zigbee-herdsman version is 0.57.3.

Let me know if this fixes the issue, then we can debug more (basically bumping the zigbee-herdsman versions until it fails again)

thomaschampagne, 2024-11-20:
> Support for CC2531 was definitely not dropped, even though its weak and known to easily crash, an update of z2m alone should not make it suddenly crash. It seems there is some regression since [...]

Thanks for your reply. 

- I changed my image to 1.40.2 in my k8s cluster. 
- I disabled rootless mode to install npm via apk (Only node is shipped inside image).
- I applied `npm install zigbee-herdsman@0.57`
- cat ./node_modules/zigbee-herdsman/package.json display version 0.57.3

But restarting service inside Tools tab seems not update the zigbee-herdsman version in about section.

Thanks again for your work and support :)

thomaschampagne, 2024-11-21:
@Koenkk I forced the command with

[image]

- Versions are now properly set

[image]

- Lights are working now

- I tell you tomorrow morning If no more lights

thomaschampagne, 2024-11-22:
@Koenkk Update 32 hours later: Lights switches are still working with `zigbee-herdsman@0.57.3`  inside `1.40.2` !

thomaschampagne, 2024-11-24:
> Tried doing the same and can also confirm that it is now stable. Been running for a few hours now without any hiccups. Previously it wouldn't even last 30 minutes before it started getting unstable.
> 
> ```
> Zigbee2MQTT version: 1.40.1
> Coordinator type: EmberZNet
> Coordinator revision: 7.4.3 [GA]
> Frontend version: 0.7.4
> zigbee-herdsman-converters version: 20.12.1
> zigbee-herdsman version: 0.57.3
> ```

Do you also have CC2531 like me ?

Koenkk, 2024-11-24:
> @Koenkk Update 32 hours later: Lights switches are still working with `zigbee-herdsman@0.57.3`  inside `1.40.2` !
> 
> 

Can you now increase the zigbee herdsman version and see when the issue occurs again? You can find a list of all versions here: https://github.com/Koenkk/zigbee-herdsman/tags

thomaschampagne, 2024-11-24:
> > @Koenkk Update 32 hours later: Lights switches are still working with `zigbee-herdsman@0.57.3`  inside `1.40.2` !
> 
> Can you now increase the zigbee herdsman version and see when the issue occurs again? You can find a list of all versions here: https://github.com/Koenkk/zigbee-herdsman/tags

👍

thomaschampagne, 2024-11-25:
@Koenkk zigbee-herdsman@1.0.1 passed the last 24 hours successfully.

Trying 2.0.0 for the upcoming 24 hours

thomaschampagne, 2024-11-26:
@Koenkk Lights failed to work this morning with `1.40.2` + `zigbee-herdsman@2.0.0`.

So issue is between `zigbee-herdsman@1.0.1` & `zigbee-herdsman@2.0.0`

Hope this helps

Koenkk, 2024-11-27:
So it seems [this commit](https://github.com/Koenkk/zigbee-herdsman/commit/9bce063767ec3c63f287a44dcdef60338c51b3d9) causes the regression, unfortunately a quite big one. Could you provide the debug log of running 24 hours with z2m 1.40.2 with zh `1.0.1` and from start till crash of `2.0.0`?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

thomaschampagne, 2024-11-28:
> So it seems [this commit](https://github.com/Koenkk/zigbee-herdsman/commit/9bce063767ec3c63f287a44dcdef60338c51b3d9) causes the regression, unfortunately a quite big one. Could you provide the debug log of running z2m 1.40.2 with zh 2.0.0 from start till lights fail to control? Please indicate the name of the light which becomes uncontrollable.
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Test started now.
Where can you transmit you the logs? Not sure I want transmit everything here in public. Thanks

Koenkk, 2024-11-29:
@thomaschampagne thanks, I need some more logging for the issue for which I created a new zigbee-herdsman branch. From the z2m dir, can you execute:

```
cd node_modules
rm -rf zigbee-herdsman
git clone https://github.com/Koenkk/zigbee-herdsman.git -b fix/z2m-24671
cd zigbee-herdsman
npm ci
npm run build
```

Now start z2m and provide the log again, you should see the extra logging I added in https://github.com/Koenkk/zigbee-herdsman/commit/643a52e0de864a69d87d58507a879212c833faae

thomaschampagne, 2024-11-30:
> @thomaschampagne thanks, I need some more logging for the issue for which I created a new zigbee-herdsman branch. From the z2m dir, can you execute:
> 
> ```
> cd node_modules
> rm -rf zigbee-herdsman
> git clone https://github.com/Koenkk/zigbee-herdsman.git -b fix/z2m-24671
> cd zigbee-herdsman
> npm ci
> npm run build
> ```
> 
> Now start z2m and provide the log again, you should see the extra logging I added in [Koenkk/zigbee-herdsman@643a52e](https://github.com/Koenkk/zigbee-herdsman/commit/643a52e0de864a69d87d58507a879212c833faae)

I'm not able to get this working as is. I tried to edit script according your package.json (root + lib) without positive result in an acceptable time.

Is unable to to find `tsc` or version is not the one expected when I force `npm install` or use `npx -y tsc` within lib folder

Here is my kube container definition:

```yaml
    containers:
      - name: zigbee2mqtt
        image: ghcr.io/thomaschampagne/oci-zigbee2mqtt:2024.10.18 # matches z2m 1.40.2
        command:
          - sh
          - -c
          - |
            echo "Installing npm and git..."  && \
            apk add npm git && \
            echo "Navigating to node_modules directory..."  && \
            cd node_modules && \
            echo "Removing existing zigbee-herdsman directory..."  && \
            rm -rf zigbee-herdsman && \
            echo "Cloning zigbee-herdsman repository from GitHub..."  && \
            git clone https://github.com/Koenkk/zigbee-herdsman.git -b fix/z2m-24671 && \
            echo "Navigating to zigbee-herdsman directory..."  && \
            cd zigbee-herdsman && \
            echo "Building zigbee-herdsman..."  && \
            npm i && npm run build && \
            echo "Navigating back to the root directory..."  && \
            cd ../.. && pwd && \
            echo "Starting the application with tini..."  && \
            /sbin/tini -- node index.js
        imagePullPolicy: Always
        ports:
          - name: http
            containerPort: 8080
            protocol: TCP
        startupProbe:
          httpGet:
            path: /
            port: 8080
          failureThreshold: 20
          periodSeconds: 10
          timeoutSeconds: 5
        readinessProbe:
          tcpSocket:
            port: 8080
          failureThreshold: 5
          periodSeconds: 30
          timeoutSeconds: 5
        livenessProbe:
          httpGet:
            path: /
            port: 8080
          failureThreshold: 5
          periodSeconds: 30
          timeoutSeconds: 5
        securityContext:
          privileged: true # Required to read /dev/ttyACM0
          runAsUser: 0
          runAsGroup: 0 # Overridden in overlays ('dialout' operating system group id to access /dev/ttyACM0 path)
          runAsNonRoot: false
        resources: {}
        volumeMounts:
          - name: localtime-volume
            mountPath: /etc/localtime
            readOnly: true
          - name: apps-data-volume
            mountPath: /app/data/
            subPath: ./home/zigbee2mqtt/
          - name: zigbee-usb-stick
            mountPath: /dev/ttyACM0
            readOnly: true
          - name: run-udev
            mountPath: /run/udev # Required by zigbee2mqtt
            readOnly: true
```

Koenkk, 2024-11-30:
@thomaschampagne could you try with `npm ci --production=false`?

thomaschampagne, 2024-11-30:
@Koenkk Same issue. Script is 

```bash
              echo "Installing npm and git..."  && \
              apk add npm git && \
              echo "Navigating to node_modules directory..."  && \
              cd node_modules && \
              echo "Removing existing zigbee-herdsman directory..."  && \
              rm -rf zigbee-herdsman && \
              echo "Cloning zigbee-herdsman repository from GitHub..."  && \
              git clone https://github.com/Koenkk/zigbee-herdsman.git -b fix/z2m-24671 && \
              echo "Navigating to zigbee-herdsman directory..."  && \
              cd zigbee-herdsman && \
              echo "Building zigbee-herdsman..."  && \
              npm ci && npm run build && \
              echo "Navigating back to the root directory..."  && \
              cd ../.. && pwd && \
              echo "Starting the application with tini..."  && \
              /sbin/tini -- node index.js
```

Error log

```
zigbee2mqtt OK: 85 MiB in 40 packages
zigbee2mqtt Navigating to node_modules directory...
zigbee2mqtt Removing existing zigbee-herdsman directory...
zigbee2mqtt Cloning zigbee-herdsman repository from GitHub...
zigbee2mqtt Cloning into 'zigbee-herdsman'...
zigbee2mqtt Navigating to zigbee-herdsman directory...
zigbee2mqtt Building zigbee-herdsman...
zigbee2mqtt npm WARN EBADENGINE Unsupported engine {
zigbee2mqtt npm WARN EBADENGINE   package: 'jackspeak@4.0.1',
zigbee2mqtt npm WARN EBADENGINE   required: { node: '20 || >=22' },
zigbee2mqtt npm WARN EBADENGINE   current: { node: 'v18.20.1', npm: '9.6.6' }
zigbee2mqtt npm WARN EBADENGINE }
zigbee2mqtt npm WARN EBADENGINE Unsupported engine {
zigbee2mqtt npm WARN EBADENGINE   package: 'path-scurry@2.0.0',
zigbee2mqtt npm WARN EBADENGINE   required: { node: '20 || >=22' },
zigbee2mqtt npm WARN EBADENGINE   current: { node: 'v18.20.1', npm: '9.6.6' }
zigbee2mqtt npm WARN EBADENGINE }
zigbee2mqtt npm WARN EBADENGINE Unsupported engine {
zigbee2mqtt npm WARN EBADENGINE   package: 'lru-cache@11.0.1',
zigbee2mqtt npm WARN EBADENGINE   required: { node: '20 || >=22' },
zigbee2mqtt npm WARN EBADENGINE   current: { node: 'v18.20.1', npm: '9.6.6' }
zigbee2mqtt npm WARN EBADENGINE }
zigbee2mqtt npm WARN EBADENGINE Unsupported engine {
zigbee2mqtt npm WARN EBADENGINE   package: 'rimraf@6.0.1',
zigbee2mqtt npm WARN EBADENGINE   required: { node: '20 || >=22' },
zigbee2mqtt npm WARN EBADENGINE   current: { node: 'v18.20.1', npm: '9.6.6' }
zigbee2mqtt npm WARN EBADENGINE }
zigbee2mqtt npm WARN EBADENGINE Unsupported engine {
zigbee2mqtt npm WARN EBADENGINE   package: 'glob@11.0.0',
zigbee2mqtt npm WARN EBADENGINE   required: { node: '20 || >=22' },
zigbee2mqtt npm WARN EBADENGINE   current: { node: 'v18.20.1', npm: '9.6.6' }
zigbee2mqtt npm WARN EBADENGINE }
zigbee2mqtt npm WARN EBADENGINE Unsupported engine {
zigbee2mqtt npm WARN EBADENGINE   package: 'minimatch@10.0.1',
zigbee2mqtt npm WARN EBADENGINE   required: { node: '20 || >=22' },
zigbee2mqtt npm WARN EBADENGINE   current: { node: 'v18.20.1', npm: '9.6.6' }
zigbee2mqtt npm WARN EBADENGINE }
stream closed EOF for focale/home-zigbee2mqtt-dm-fxq45 (preload-config-files)
zigbee2mqtt 
zigbee2mqtt added 21 packages, and audited 22 packages in 3s
zigbee2mqtt 
zigbee2mqtt 6 packages are looking for funding
zigbee2mqtt   run `npm fund` for details
zigbee2mqtt 
zigbee2mqtt found 0 vulnerabilities
zigbee2mqtt 
zigbee2mqtt > zigbee-herdsman@2.0.0 build
zigbee2mqtt > tsc
zigbee2mqtt 
zigbee2mqtt sh: tsc: not found
```

BoronFinder, 2024-12-01:
I'm joining this thread from https://github.com/Koenkk/zigbee2mqtt/issues/24293 which is (now obviously) about the same error.
There I've isolated this error to occur already with the zigbee-herdsman 1.1.0, not 2.0.0 (see commit https://github.com/Koenkk/zigbee2mqtt/commit/d289194a71f67dd821335d467532f74b3d47f03a ). Tested this yesterday and the error occurs reproducibly. If I can assist in any way with isolating it further, just tell me.

(Note: I'm also using a Texas Instrument CC2531 dongle.)

thomaschampagne, 2024-12-01:
> This seems not to be a problem caused by the old stick ... I use a CC2652RB stick with brand new firmware and see similiar issues after upgrade to 1.40.2.
>
> Problem occurs after several - unknown how [...]

@purche42 Could you also join the "testing program" to figure out if bug analysis is same and upcoming fix (we hope) is working on your CC2652RB  ? Thanks

thomaschampagne, 2024-12-01:
@Koenkk Indeed I have been able to get the error in version `1.1.0` of ` zigbee-herdsman` too with `CC2531` like @BoronFinder. I sent you log file by email.

@purche42 Do you also reproduce error on `CC2652RB` with that version?

Koenkk, 2024-12-01:
@thomaschampagne `npm ci && npm run build && \` should be `npm ci --production=false && npm run build && \` or did you already try that? The logs you send me, are these with the `fix/z2m-24671` branch?

Koenkk, 2024-12-03:
@thomaschampagne I did not check these logs, they will not be different than the 2.0.0 logs. I need the ones with the added logging (`fix/z2m-24671` branch). Let me know once you have those.

thomaschampagne, 2024-12-03:
> @thomaschampagne I did not check these logs, they will not be different than the 2.0.0 logs. I need the ones with the added logging (`fix/z2m-24671` branch). Let me know once you have those.

I sent you the logs you are asking (branch `fix/z2m-24671`) yesterday in my last email (archive `log_1.40.2_with_2.0.0.zip`).

Koenkk, 2024-12-03:
I pushed a change to `fix/z2m-24671`, could you check if the issue is gone now? (this is not a proper fix yet)

thomaschampagne, 2024-12-04:
@Koenkk I confirm it's working with your last patch. Lights are still working.

Koenkk, 2024-12-08:
@thomaschampagne I've implemented a possible fix, can you check again with the latest `fix/z2m-24671`?

thomaschampagne, 2024-12-09:
@Koenkk Thanks for the patch.

I will not be able to provide the final testing of this

@BoronFinder & @purche42 could you please take my relay for testing and validate the patch on `CC2531` and `CC2652RB` respectively?

Command script to override on 1.40.2 version:

```
echo "Installing npm and git..."  && \
              apk add npm git && \
              echo "Navigating to node_modules directory..."  && \
              cd node_modules && \
              echo "Removing existing zigbee-herdsman directory..."  && \
              rm -rf zigbee-herdsman && \
              echo "Cloning zigbee-herdsman repository from GitHub..."  && \
              git clone https://github.com/Koenkk/zigbee-herdsman.git -b fix/z2m-24671 && \
              echo "Navigating to zigbee-herdsman directory..."  && \
              cd zigbee-herdsman && \
              echo "Building zigbee-herdsman..."  && \
              npm ci --production=false && npm run build && \
              echo "Navigating back to the root directory..."  && \
              cd ../.. && pwd && \
              echo "Starting the application with tini..."  && \
              /sbin/tini -- node index.js
```
 
Thanks

purche42, 2024-12-09:
@thomaschampagne 
> @BoronFinder & @purche42 could you please take my relay for testing and validate the patch on `CC2531` and `CC2652RB` respectively?

I'm running z2m on docker. I would test on on CC2652 with fix with an image which includes the fix. I have also another test installation on HAOS with Z2M Addon - but only with old CC2531 stick.

thomaschampagne, 2024-12-09:
> @thomaschampagne
>
> I'm running z2m on docker. I would test on on CC2652 with fix with an image which includes the fix. I have also another test installation on HAOS with Z2M Addon - but only with [...]

You just need to override the docker command like I do above using kubernetes ;)

r0bb10, 2024-12-10:
waiting to test it as homeassistant addon, will it be in edge version?

purche42, 2024-12-11:
> > @thomaschampagne
... 
> You just need to override the docker command like I do above using kubernetes ;)
Thanks for the tip ... that works fine

Test runs with CC2652RB (Slaesh's stick)

Zigbee2MQTT-Version [1.40.2]
Coordinator-Typ zStack3x0
Coordinator-Version 20240710
Frontend-Version 0.7.4
Zigbee Herdsman Konverter Version 20.21.0
Zigbee Herdsman Version 2.0.0

I will provice feedback, if light ON/OFF problem is gone (hopefully) with that fix.

thank you also for all your work - Koen - Thomas !

Koenkk, 2024-12-14:
Integrated the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2024-12-14 by Koenkk