Koenkk/zigbee2mqtt #24678: Unknown battery status for several devices

merlinpimpim, 2024-11-07:
### What happened?

Battery status is unknown for these devices::
- https://www.zigbee2mqtt.io/devices/SNZB-02D.html
- https://www.zigbee2mqtt.io/devices/NAS-AB06B2.html#neo-nas-ab06b2 (Yes, it's been connected for over 24 hours, and the battery level hasn't risen for 6 months).

[image]
(As a control device, the Xiaomi temperature device correctly shows its battery level)

[image]

### What did you expect to happen?

Knowing the battery charge level of these devices

### How to reproduce it (minimal and precise)

n/a

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240315

### Adapter

SLZB-06

### Setup

HAOS (Green)

DaN66O, 2024-11-07:
The same problem
https://www.zigbee2mqtt.io/devices/SNZB-01P.html#sonoff-snzb-01p
[image]

chris-1243, 2024-11-07:
Have you tried to follow the [troubleshoot](https://www.zigbee2mqtt.io/devices/SNZB-02D.html#sonoff-snzb-02d) ?

Mine are perfectly reporting battery value

merlinpimpim, 2024-11-08:
> Have you tried to follow the [troubleshoot](https://www.zigbee2mqtt.io/devices/SNZB-02D.html#sonoff-snzb-02d) ?
> 
> Mine are perfectly reporting battery value.
> 
> Also check if the `PowerCfg` cluster is bound to the coordinator and if the reporting is correctly set

Yes, (it's the link I posted when I opened the ticket :) )

On the other hand, I forgot to specify that the information does go up in the device details for SNZB-02D:
[image]
And the PowerCfg cluster seems to be bound correctly.
[image]

But I think this is another problem with the Neo NAS-AB06B2, because in the Exposes tab, the battery field is “Null”.
[image]

chris-1243, 2024-11-08:
For your SNZB-02P, it looks more an issue in the frontend. May you try to shirtly press the button on the device to wake it up and press the refresh button close to the battery in the expose tab ?

> But I think this is another problem with the Neo NAS-AB06B2, because in the Exposes tab, the battery field is “Null”.
image.png (view on web)

I had this kind of issue with a Ikea remote. Try to remove the battery, wait some seconds and put it back. Then refresh the value...

merlinpimpim, 2024-11-08:
> For your SNZB-02P, it looks more an issue in the frontend. May you try to shirtly press the button on the device to wake it up and press the refresh button close to the battery in the expose tab ?
> 
Quickly pressing the device wakes it up (last seen= just now in Z2M), the values are refreshed in Exposes but the question mark remains in the dashboard.
So, yes, probably a fontend problem. Is this the right place to report this type of problem?

> I had this kind of issue with a Ikea remote. Try to remove the battery, wait some seconds and put it back. Then refresh the value...

For the Neo NAS-AB06B2 I tested your proposal (but I'd done it before) without success.
Note that now in Exposes tab the value is N/A instead of Null:
[image]

But I think the problem is deeper because in the payload I don't see anything about the battery. Maybe a problem with the converter?
[image]

chris-1243, 2024-11-08:
Yes, I would bet for the converter and it is a Tuya based sensor... (always tricky).

It would be good to create an external converter and try...I had a quick check and nothing wrong.

Before making a converter, what is it configured in the reporting tab ?

merlinpimpim, 2024-11-08:
> Before making a converter, what is it configured in the reporting tab ?

Nothing...
[image]

chris-1243, 2024-11-08:
Would you try to add this ?

[image]

Before clicking on `apply`, awake your sensor. A quick press on the pairing button should do the trick

chris-1243, 2024-11-08:
> Cache cleared and HA Green rebooted: no change

Frontend problem I guess. I can not help more on that topic.

What clusters do you have on this device ?

merlinpimpim, 2024-11-08:
> What clusters do you have on this device ?

[image]

apydo, 2024-11-08:
This is issue is related to #24453.

merlinpimpim, 2024-11-08:
> This is issue is related to #24453.

Thank you @apydo .
It does indeed look similar for the SNZB-02D, however **I think the problem is different for the Neo NAS-AB06B2** (which has never displayed a battery level in the 1 year I've had it...)

> Would you provide the cluster tab?

[image]

> Mine are perfectly reporting battery value.

How is this possible @chris-1243  given the https://github.com/Koenkk/zigbee2mqtt/issues/24453 and https://github.com/Koenkk/zigbee2mqtt/issues/24025 tickets?

chris-1243, 2024-11-09:
> How is this possible @chris-1243 given the #24453 and #24025 tickets?

Firmware version ? I have 4 of them bought at different time and everything is fine (except a randomly polling error which do not have any impact on the reporting of the values)

nchevsky, 2025-01-10:
🏓

merlinpimpim, 2025-01-12:
For me it's solved with ZM 2.0.0

Sabjul33, 2025-02-01:
News for me with the 2.1.0 update !!!  have the battery percentage for the NAS-AB06B2
https://github.com/Koenkk/zigbee2mqtt/issues/19827#issuecomment-2629130678

closed 2025-04-10 by github-actions[bot]