Koenkk/zigbee2mqtt #24681: Moes MS-108ZR reverse

romain1206, 2024-11-07:
### What happened?

hello,

since the update to 1.41.0
my shutters have the reversed state

when the shutter is physically closed zeegbee2mqtt sees it open
when it is open it sees it closed.

i have this problem on my 5 shutters since the update 1.41.0 before no problem

many thanks

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

6.5.0.0 build 188

### Adapter

lidl gateway

### Setup

docker addon home assistant

Great-Chart, 2024-11-07:
This has a fix already in progress - also see https://github.com/Koenkk/zigbee2mqtt/issues/24597 where others have corrected in the short term

Koenkk, 2024-11-09:
Already fixed indeed! 

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.