Koenkk/zigbee2mqtt #24682: Regression on 1.41.0 when running behind nginx reverse proxy

lkomurcu, 2024-11-07:
### What happened?

I have been running zigbee2mqtt behind nginx reverse proxy with ssl under https://z2m.mydomain.com. All of my Home automation stack is running inside containers. After updating to 1.41.0 I was not able to access to zigbee2mqtt frontend via reverse proxy but I was able access it via ip/port. latest-dev tag (dist/.hash [b5d4a25](https://github.com/Koenkk/zigbee2mqtt/tree/b5d4a253b4ef5b4930f8d84e878e2343cc0b38df))  still has the issue. Chrome devtools console shows

index-6f55efcd.js:61 WebSocket connection to 'wss://z2m.mydomain.com/api' failed: 
(anonymous)	@	index-6f55efcd.js:61 
I was able to bisect the latest changes commit where regression happens is coming from Koenkk/zigbee2mqtt#24244 and commit Koenkk/zigbee2mqtt@cbe08c3

I reverted the line where a websocket server is created to not pass the new path parameter than regression seems to be fixed but the new feature is obviously would be broken.
this.wss = new WebSocket.Server({noServer: true, path: path.join(this.frontendBaseUrl, 'api')});

I am not that good that typescript to be able to solve this issue. Hope we can restore this functionality back.

_This issue was originally opened on frontend (https://github.com/nurikk/zigbee2mqtt-frontend/issues/2223) but i think this repo is more appropriate since the code in question is in this repo._

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

NA

### Adapter

NA

### Setup

X64 Docker Container

lux73, 2024-11-09:
https://github.com/Koenkk/zigbee2mqtt/discussions/24570#discussioncomment-11122454

this should work 😉

lkomurcu, 2024-11-09:
Even tough I don't have the same setup ( sub folder vs subdomain root ) removing the secondary location setting for /api on proxy configuration seems to have fixed it for me. Thank you.

closed 2024-11-09 by lkomurcu