Koenkk/zigbee2mqtt #24683: Z2M lost device paired after a short little

titou4307, 2024-11-07:
### What happened?

Since many weeks I am looking for my problem : 

I have many devices who disappear of the frontend after a time under 1 hour.

I have paired all my devices since many many month and suddently there are 3 or 4 devices who appears offline. I have repair these, so they appear into frontend but after a while (30 minutes, 10 minutes....) they are offline.....

I am not sure that is done after the update to the last version of Z2M

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

Sonoff Dongle P

### Setup

Intel Nuc with OS Debian, 

### Debug log

[log.log](https://github.com/user-attachments/files/17670443/log.log)

fsedarkalex, 2024-11-11:
maybe related? #24089

titou4307, 2024-11-11:
Hello @fsedarkalex 

I have follow your link and indication on #24089 

I have delete my device Legrand of Z2M and restart all the Home Assistant
After I have repair the device Legrand, name it

And after I have adjust  the parameter Adapter Delay to 200

So I will observe if this solve the problem....

Thanks

fsedarkalex, 2024-11-11:
@titou4307 as you have a SONOFF coordinator I think the delay parameter is not used at all.

I am just trying to find similarities under the most recent device dropping issues to probably find a common cause.
But I am not deep enough involved in Z2M's code to understand whats happening actually.

titou4307, 2024-11-11:
@fsedarkalex 

Don't worry, i will try (or I have try....)
But I confirm you this option of solution not function, my device Legrand will be out another time (it is online one more time...)

I will erase the delay parameter to return at origin config

I have all the device who have LQI more low, I think I will change my coordinator for to test....

daconas, 2024-11-13:
Hello i lost also devices.
sonoff dongle-e amber
Version: 1.41.0-1
by devices are 39 and by database 40 why is here a different

closed 2025-01-20 by github-actions[bot]

titou4307, 2025-01-20:
Hi
Problem not solved for me.....