Koenkk/zigbee2mqtt #24684: OTA update failed with ABORT for Aqara QBKG11LM wall switch

mrin, 2024-11-08:
### What happened?

Any attempts to update firmware ended up with fail message "ABORT"
I have the same several wall switches with updated firmware that updated 2 years ago successfully.

**Device**
Zigbee Model: lumi.ctrl_ln1
IEEE Address: 0x00158d00014a654b
Firmware build date: 10-13-2016
Model: QBKG11LM

**State**
```
{
    "consumption": 0.07831941545009613,
    "device_temperature": 30,
    "energy": 0.08,
    "linkquality": 72,
    "power": 0,
    "power_outage_count": 9,
    "state": "OFF",
    "update": {
        "installed_version": 7,
        "latest_version": 32,
        "state": "available"
    },
    "update_available": true,
    "voltage": 221.10000000000002,
    "operation_mode": null
}
```

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

ZigStar LAN CC2652P2

### Setup

Home Assistant addon

### Debug log

info 2024-11-08 11:40:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'hl_light_switch' failed (Update failed with reason: 'ABORT')","meta":{"device":"hl_light_switch","status":"update_failed"},"type":"ota_update"}'
info 2024-11-08 11:40:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"hl_light_switch"},"error":"Update of 'hl_light_switch' failed (Update failed with reason: 'ABORT')","status":"error","transaction":"n78s2-1"}'
error 2024-11-08 11:40:53z2m: Update of 'hl_light_switch' failed (Update failed with reason: 'ABORT')

hanszant, 2024-12-07:
Same Here. Is there any solition?

mrin, 2025-02-06:
The issue still persists with v2.1.0

closed 2025-04-15 by github-actions[bot]