Koenkk/zigbee2mqtt #24688: Ikea INSPELNING providing power > 0W even if switch is turned off

nicola-spreafico, 2024-11-08:
### What happened?

This morning (more than 10h ago), I switched off the printer using the plug switch, as you can see here:
[image]

Despite being off, the power consumption is reading 3.2W which is not possibile considering the device is not powered.

I went to Z2M interface and found this:
[image]
Only after I pressed the refresh button I got the actual 0.0 value
[image]

### What did you expect to happen?

If the switch is off, the read power should be 0.0

### How to reproduce it (minimal and precise)

Connect some appliance to smart plug, switch off the plug, see that power is not going to 0

### Zigbee2MQTT version

1.41.0 commit: cc31ceab

### Adapter firmware version

7.4.1 [GA]

### Adapter

smlight slzb-06m

### Setup

Dedicated LXC on Proxmox

MaxRower, 2024-11-17:
You need to fine tune the reporting settings, so changes get reported with an acceptable delay.

nicola-spreafico, 2024-11-18:
Hi @MaxRower ,
I understand you are referring to this view:
[image]

As far I can see the default values are up to 60 (seconds I guess), but in my case after many hours the value was not updated, so I don't think this is the root cause.

MaxRower, 2024-11-18:
I think those values are seconds, so it would be 65000 seconds, nearly a day?
Do you have the latest firmware on your plugs?

MaxRower, 2024-11-18:
Additionally, you are missing the entry for the acPowerDivisor, so you will have wrong voltage values, probably?
Take a close look at that thread, there is a lot of discussion about that plug.
https://github.com/Koenkk/zigbee2mqtt/issues/23961

nicola-spreafico, 2024-11-18:
Yes, I'm having a wrong voltage value, but I did not give much interest as I'm using this plug only for switch and power, voltage value are ignored in any case. I was my understanding that a fix is waited for the next release, but now from your message I understand I need to fix it by myself with the link you provided? In any case is not a priority for me.

I never touched the values inside the Reporting tab, how it works? All those values what they represent?

nicola-spreafico, 2025-01-19:
Problem still there
[image]

Switch is off (so device is entirely without current), but the plug notify 1W consumption (and it is like that since several hours).
Clicking the refresh button solve the issue, but it requires a manual action

rpadafo, 2025-03-20:
Hello all, I have the same issue.

When I turn off the plug the power is the last power when the plug was on.
The only way to refresh the data is pushing the "Power refresh" button on zigbee2mqtt.

Any solution??

rpadafo, 2025-03-20:
I think I have a solution... In "Reporting" section. Modify the "activepower" line, "max rep interval".
From 65000 to 600 (5 minutes for example).

[image]