Koenkk/zigbee2mqtt #24689: Crash when trying to pair zzh as Router

trashmaster9000, 2024-11-08:
### What happened?

Z2M crashes after pairing a zzh stick flashed with router Firmware, the router is not listed in Devices list despite getting a "pairing successfull" message before the crash

### What did you expect to happen?

successfull pairing and no crash

### How to reproduce it (minimal and precise)

Enable Pairing, press button on zzh, wait

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

Sonoff Zigbee 3.0

### Setup

Homeassistant OS, addon, x86_64

### Debug log

```
[2024-11-08 19:02:12] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Coordinator"},"type":"device_announced"}'
[2024-11-08 19:02:12] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Coordinator","ieee_address":"0x00124b0021b952ca"},"type":"device_announce"}'
[2024-11-08 19:02:13] warning: 	z2m: Device '0x00124b0021b952ca' left the network
[2024-11-08 19:02:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x00124b0021b952ca"},"type":"device_removed"}'
[2024-11-08 19:02:15] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0021b952ca","ieee_address":"0x00124b0021b952ca"},"type":"device_leave"}'
[2024-11-08 19:02:15] error: 	z2m: EventBus error 'Bridge/permitJoinChanged': Cannot read properties of undefined (reading 'endpoints')
[2024-11-08 19:02:16] error: 	z2m: EventBus error 'Bridge/permitJoinChanged': Cannot read properties of undefined (reading 'endpoints')
[2024-11-08 19:02:17] error: 	z2m: EventBus error 'Bridge/permitJoinChanged': Cannot read properties of undefined (reading 'endpoints')
[2024-11-08 19:02:18] error: 	z2m: EventBus error 'Bridge/permitJoinChanged': Cannot read properties of undefined (reading 'endpoints')
[2024-11-08 19:02:19] error: 	z2m: EventBus error 'Bridge/permitJoinChanged': Cannot read properties of undefined (reading 'endpoints')
[2024-11-08 19:02:20] error: 	z2m: EventBus error 'Bridge/permitJoinChanged': Cannot read properties of undefined (reading 'endpoints')
[2024-11-08 19:02:20] info: 	zh:controller: Interview for '0x00124b0021b952ca' started
[2024-11-08 19:02:20] info: 	z2m: Device 'Coordinator' joined
[2024-11-08 19:02:22] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"Coordinator"},"type":"device_connected"}'
[2024-11-08 19:02:22] info: 	z2m: Starting interview of 'Coordinator'
[2024-11-08 19:02:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"Coordinator"},"type":"pairing"}'
[2024-11-08 19:02:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Coordinator","ieee_address":"0x00124b0021b952ca"},"type":"device_joined"}'
[2024-11-08 19:02:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Coordinator","ieee_address":"0x00124b0021b952ca","status":"started"},"type":"device_interview"}'
[2024-11-08 19:02:47] info: 	zh:controller: Succesfully interviewed '0x00124b0021b952ca'
[2024-11-08 19:02:47] info: 	z2m: Successfully interviewed 'Coordinator', device has successfully been paired
TypeError: Cannot destructure property 'vendor' of 'data.device.definition' as it is null.
    at Zigbee.logDeviceInterview (/app/lib/zigbee.ts:166:24)
    at Controller.<anonymous> (/app/lib/zigbee.ts:101:18)
```

lux73, 2024-11-09:
`z2m: Successfully interviewed 'Coordinator', device has successfully been paired`

you sure the Stick is flashed with a Router Firmware? if i read your Log correctly you try to add another Coordinator ^^

trashmaster9000, 2024-11-09:
yes, i've flashed it three times already since i thought maybe it didnt work because of the name...
and it looks like it flashed successfully, see below.

the zzh adaptor was the coordinator until yesterday, i've replaced it with the sonos adaptor since i had trouble with the usb connection.
does z2m still think this is also the coordinator or do i have to flash the FW differently?

```
sudo python3 cc2538-bsl-master/cc2538-bsl.py -p /dev/ttyUSB0 -evw CC2652R_router_20221102.hex 
[sudo] Passwort für poe: 
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC2652R_router_20221102.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:21:B9:52:CA
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0x630abbc5)

```

duceduc, 2024-11-09:
What program did you use to change the ieee for the zzh device? I am wanting to do the same as I have upgraded a new coordinator (uzg-01) and now wanting to use the zzh as a router.

duceduc, 2024-11-14:
Hey. Thanks for the reply. That is what I did, but was hoping to use the programmer 2 for windows to change it since the router version is 2022xxx which should still work. Unfortunately, it did not want to write to the router.

I have a rpi4 running as a test unit and tried to installed the cc2538-bsl on it and that was asking the python should be run in a env environment which I did not know how to do.

Fortunately, I did not wipe clean my old windows 10 pc which had python3 installed with cc2538-bsl.

closed 2025-01-21 by github-actions[bot]