Koenkk/zigbee2mqtt #24690: Centralite 3157100 - Incorrect Celcius-Farenheit Conversion - local_temperature_calibration

martyg77, 2024-11-08:
### What happened?

I have a Centralite 3157100-E on my network.

I observe the following message in my HA logs about every 3 minutes:
2023-10-25 11:23:50.683 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.furnace_local_temperature_calibration: 0 (range 27.5 - 36.5)

MQTT has these limits correctly populated as -2.5..+2.5 (Temperature offset)
HA treats the MQTT data as absolute temperatures, converting min = -2.5C = 27.5F; max = +2.5C = 36.5F

HA code should not be blindly converting C->F for these offsets
Correct conversion in this case should be -4.5..+4.5F

From the HA web interface,
Settings -> Devices and Services -> Devices -> Furnace -> MQTT Info
I see in the Payload section:

```
availability:
  - topic: zigbee2mqtt/bridge/state
    value_template: '{{ value_json.state }}'
command_topic: zigbee2mqtt/Furnace/set/local_temperature_calibration
device:
  identifiers:
    - zigbee2mqtt_0x000d6f00055b73d4
  manufacturer: Centralite
  model: 3-Series pearl touch thermostat (3157100)
  name: Furnace
device_class: temperature
entity_category: config
icon: mdi:math-compass
max: 2.5
min: -2.5
name: Local temperature calibration
object_id: furnace_local_temperature_calibration
origin:
  name: Zigbee2MQTT
  sw_version: 1.33.2
  support_url: https://www.zigbee2mqtt.io
state_topic: zigbee2mqtt/Furnace
step: 0.1
unique_id: 0x000d6f00055b73d4_local_temperature_calibration_zigbee2mqtt
unit_of_measurement: °C
value_template: '{{ value_json.local_temperature_calibration }}'
platform: mqtt
```

However, from the file generated by
Settings -> Devices and Services -> Devices -> Furnace -> Device Info -> Download Diagnostics (pulldown)
I see this excerpt:

```
        {
          "device_class": null,
          "disabled_by": null,
          "disabled": false,
          "entity_category": "config",
          "entity_id": "number.furnace_local_temperature_calibration",
          "icon": null,
          "original_device_class": "temperature",
          "original_icon": "mdi:math-compass",
          "state": {
            "entity_id": "number.furnace_local_temperature_calibration",
            "state": "unknown",
            "attributes": {
              "min": 27.5,
              "max": 36.5,
              "step": 0.1,
              "mode": "auto",
              "unit_of_measurement": "\u00b0F",
              "device_class": "temperature",
              "icon": "mdi:math-compass",
              "friendly_name": "Furnace Local temperature calibration"
            },
            "last_changed": "2023-11-03T20:45:39.175541+00:00",
            "last_updated": "2023-11-03T20:45:39.175541+00:00"
          },
          "unit_of_measurement": "\u00b0F"
        }
```

I originally filed this PR with the HA people, and I was told to come here.  ref. https://github.com/home-assistant/core/issues/129341

If you could point me to the code where the __local_temperature_calibration__ entity gets set up in HA, I would be happy to dig some more on my own.  Right now, I am stumped.  Thnaks in advance for your help.

### What did you expect to happen?

The code setting up the HA entity entity __number.local_temperature_calibration__ to be properly populated with -4.5..+4.5F instead of 27.5F..36.5F

### How to reproduce it (minimal and precise)

Any system with a Centralite thermostat set as Farenheit should see this error in their HA logs every time the device reports.

### Zigbee2MQTT version

1.40.2-1

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee USB CC2652P

### Setup

HAOS add-on

rpatel3001, 2024-11-23:
either z2m needs to convert locally and send as °F or home assistant needs a new device_class for temperature_difference that applies a different conversion. I don't know if z2m can figure out what temperature unit home assistant is using and automatically convert, so it probably needs to be a manual setting here.

rpatel3001, 2024-11-23:
actually there is a thermostat unit setting for the device which controls the display units on the actual device. it's not necessarily the case that the display unit will match the unit used in home assistant but it's probably a pretty good guess for now.

rpatel3001, 2024-11-23:
It's an easy fix: set the unit for number.xxxxxxxx_local_temperature_calibration to degrees C in the homeassistant GUI

https://github.com/home-assistant/core/issues/129341

martyg77, 2024-11-24:
A fix has been documented for this issue in https://github.com/home-assistant/core/issues/129341

Closing this PR.

closed 2024-11-24 by martyg77