Koenkk/zigbee2mqtt #24695: Ikea Styrbar fails to work

rajil, 2024-11-09:
### What happened?

Hello,
I upgraded Ikea Styrbar to firmware 20240708 and now it doesnt work.

```
zigbee2mqtt  | [2024-11-09 16:25:24] info:      z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x90ab96fffe286b66","ieee_address":"0x90ab96fffe286b66"},"type":"device_leave"}'
zigbee2mqtt  | [2024-11-09 16:25:30] info:      zh:controller: Interview for '0x90ab96fffe286b66' started
zigbee2mqtt  | [2024-11-09 16:25:30] info:      z2m: Device '0x90ab96fffe286b66' joined
zigbee2mqtt  | [2024-11-09 16:25:31] info:      z2m: Starting interview of '0x90ab96fffe286b66'
zigbee2mqtt  | [2024-11-09 16:25:31] info:      z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x90ab96fffe286b66","ieee_address":"0x90ab96fffe286b66"},"type":"device_joined"}'
zigbee2mqtt  | [2024-11-09 16:25:31] info:      z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x90ab96fffe286b66","ieee_address":"0x90ab96fffe286b66","status":"started"},"type":"device_interview"}'
zigbee2mqtt  | [2024-11-09 16:25:31] info:      z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x90ab96fffe286b66","ieee_address":"0x90ab96fffe286b66"},"type":"device_announce"}'
zigbee2mqtt  | [2024-11-09 16:25:32] info:      zh:controller: Succesfully interviewed '0x90ab96fffe286b66'
zigbee2mqtt  | [2024-11-09 16:25:32] info:      z2m: Successfully interviewed '0x90ab96fffe286b66', device has successfully been paired
zigbee2mqtt  | [2024-11-09 16:25:32] info:      z2m: Device '0x90ab96fffe286b66' is supported, identified as: IKEA STYRBAR remote control (E2001/E2002)
zigbee2mqtt  | [2024-11-09 16:25:32] info:      z2m: Configuring '0x90ab96fffe286b66'
zigbee2mqtt  | [2024-11-09 16:25:32] info:      z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"STYRBAR remote control","exposes":[{"access":2,"category":"config","description":"Initiate device identification. This device is asleep by default.You may need to wake it up first before sending the identify command.","label":"Identify","name":"identify","property":"identify","type":"enum","values":["identify"]},{"access":5,"category":"diagnostic","description":"Remaining battery in %","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"category":"diagnostic","description":"Triggered action (e.g. a button click)","label":"Action","name":"action","property":"action","type":"enum","values":["on","off","brightness_move_up","brightness_move_down","brightness_stop","arrow_left_click","arrow_left_hold","arrow_left_release","arrow_right_click","arrow_right_hold","arrow_right_release"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E2001/E2002","options":[{"access":2,"description":"Sets the duration of the identification procedure in seconds (i.e., how long the device would flash).The value ranges from 1 to 30 seconds (default: 3).","label":"Identify timeout","name":"identify_timeout","property":"identify_timeout","type":"numeric","value_max":30,"value_min":1},{"access":2,"description":"Simulate a brightness value. If this device provides a brightness_move_up or brightness_move_down action it is possible to specify the update interval and delta. The action_brightness_delta indicates the delta for each interval. ","features":[{"access":2,"description":"Delta per interval, 20 by default","label":"Delta","name":"delta","property":"delta","type":"numeric","value_min":0},{"access":2,"description":"Interval duration","label":"Interval","name":"interval","property":"interval","type":"numeric","unit":"ms","value_min":0}],"label":"Simulated brightness","name":"simulated_brightness","property":"simulated_brightness","type":"composite"},{"access":2,"description":"Set to false to disable the legacy integration (highly recommended), will change structure of the published payload (default true).","label":"Legacy","name":"legacy","property":"legacy","type":"binary","value_off":false,"value_on":true}],"supports_ota":true,"vendor":"IKEA"},"friendly_name":"0x90ab96fffe286b66","ieee_address":"0x90ab96fffe286b66","status":"successful","supported":true},"type":"device_interview"}'
zigbee2mqtt  | [2024-11-09 16:25:47] error:     z2m: Failed to configure '0x90ab96fffe286b66', attempt 1 (Error: Bind 0x90ab96fffe286b66/1 genPowerCfg from '0x00124b00258ce0a2/1' failed (AREQ - ZDO - bindRsp after 10000ms)
zigbee2mqtt  |     at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
zigbee2mqtt  |     at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:403:47
zigbee2mqtt  |     at processTicksAndRejections (node:internal/process/task_queues:95:5)
zigbee2mqtt  |     at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
zigbee2mqtt  |     at ZStackAdapter.sendZdo (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:311:16)
zigbee2mqtt  |     at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:563:30)
zigbee2mqtt  |     at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:160:13)
zigbee2mqtt  |     at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:211:17)
zigbee2mqtt  |     at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:205:21)
zigbee2mqtt  |     at Configure.configure (/app/lib/extension/configure.ts:130:13))

```

### Zigbee2MQTT version

1.41.0 commit: cc31cea

### Adapter firmware version

20221226

### Adapter

smlight slzb-06

### Setup

Plain docker

chris-1243, 2024-11-09:
I have two perfectly working on the latest firmware and no errors at all. (2.4.16 - 20240708)

While pairing, I would suggest to shortly press a button to keep the device awake

liselotte111, 2024-12-31:
For me removing the device and repairing after the update fixed the problem

closed 2025-03-10 by github-actions[bot]