Koenkk/zigbee2mqtt #24698: [Feature request]: Add OTA support to EcoDim ED-100xx

ThorPrez, 2024-11-09:
### Is your feature request related to a problem? Please describe

In manufacturer manual is written down: The switch supports firmware updates via OTA and will automatically search for updates every 10 minutes via the ZigBee controller.

### Describe the solution you'd like

Be able to load a firmware via OTA

### Describe alternatives you've considered

nothing

### Additional context

ED-10012 is always showing a battery status of 50%, ED-10014 is alway showing 200%.

Koenkk, 2024-11-10:
To support these we need to have the OTA files from the manufacturer. Once we have them https://github.com/koenkk/zigbee-ota?tab=readme-ov-file#example-using-github can be followed to add them.

ThorPrez, 2024-11-10:
Hi.
Thanks for response.
I´ve sent an email to their support.

ThorPrez, 2024-11-11:
@Koenkk 
What´s about the problem with the battery status ?
Ecodim told me on Homey it´s working correctly and they cannot test it on Sonoff.

Koenkk, 2024-11-12:
Fixed the battery % part (after updating wait until it reports a new value, takes max 25 hours)

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-11-12 by Koenkk

ThorPrez, 2024-11-23:
Problem is not fixed. On ED-10014 I´m still having always 200%, on ED-10012 it´s always 43,5%.