Koenkk/zigbee2mqtt #24699: Aqara RTCGQ11LM (lumi.sensor_motion.aq2) reports Illuminance in Lux twice

RubenKelevra, 2024-11-09:
### What happened?

I've noticed that the Aqara RTCGQ11LM (self reporting as `vendor: LUMI`, `product: lumi.sensor_motion.aq2`) is reporting Lux twice.

I think this stems from their other product Xiaomi GZCGQ01LM (self reporting as `vendor: LUMI`, `product: lumi.sen_ill.mgl01`) which is exposing an internal measurment under `illuminance`, which is there correctly identified as "Raw measured illuminance" via the description.

The Aqara RTCGQ11LM however shows the same lux measurement twice. So I think we can skip one of them?

[image]
[image]
[image]
[image]

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20221226

### Adapter

Slaesh's CC2652 RB stick

### Setup

Add-On on Home Assistant OS on RasPi 4B

Koenkk, 2024-11-10:
`illuminance_lux` is deprecated indeed, I will remove it with Zigbee2MQTT 2.0.0 (https://github.com/Koenkk/zigbee2mqtt/discussions/24198)

RubenKelevra, 2024-11-10:
@Koenkk thanks. Please note that illuminance_lux is however the useful value for devices like previously mentioned Xiaomi GZCGQ01LM:

[image]
[image]

So I guess your plan is to move `illuminance_lux` to `illuminance` then for those devices, and add a field like `illuminance_raw` or something for the raw value?

Koenkk, 2024-11-10:
> So I guess your plan is to move illuminance_lux to illuminance then for those devices, and add a field like illuminance_raw or something for the raw value?

Just illuminance (which is the value in lux), no raw value (that's legacy)

RubenKelevra, 2024-11-10:
Ah, alright. Yeah, most people won't ever use it anyway.

RubenKelevra, 2025-01-10:
Fixed in 2.0

closed 2025-01-10 by RubenKelevra