Koenkk/zigbee2mqtt #24701: wrong local_temperature_calibration value is not a number got string https://www.zigbee2mqtt.io/devices/TRVZB.html#sonoff-trvzb

JacekKac, 2024-11-09:
### What happened?

2024-11-09 23:58:20z2m: Publish 'set' 'local_temperature_calibration' to 'Thermostat_1' failed: 'Error: Value is not a number, got string (2)'

https://www.zigbee2mqtt.io/devices/TRVZB.html#sonoff-trvzb

### What did you expect to happen?

accept the value

### How to reproduce it (minimal and precise)

just post a command to this field

### Zigbee2MQTT version

 1.41.0 commit: cc31ceab

### Adapter firmware version

8.0.2 [GA]

### Adapter

ember sonoff E dongle

### Setup

plain zigbee2mqtt

### Debug log

2024-11-09 23:58:20z2m: Publish 'set' 'local_temperature_calibration' to 'Thermostat_1' failed: 'Error: Value is not a number, got string (2)'

lux73, 2024-11-10:
check your devices.yaml or configuration.yaml - there is a incorrect calibration entry that contains weird chars

it must be an Number

JacekKac, 2024-11-10:
no there is not. 
configuration.yaml
 '0x0cae5ffffecaa53f':
    friendly_name: Thermostat_1
    description: Salon - pod oknem
    thermostat_unit: celsius
i am not using devices.yaml

JacekKac, 2024-11-10:
ok got it it was a problem in mqtt message was a string instead of number

JacekKac, 2024-11-12:
solved

closed 2024-11-12 by JacekKac