Koenkk/zigbee2mqtt #24705: Startup fails with "z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'zh')"

aquila, 2024-11-10:
### What happened?

I have a brand new SLZB-06. I used a fresh Zigbee2MQTT installation, first struggling with the PAN ID issue but after generating a new one that works. However, now startup fails with a TypeError on zh. 

### How to reproduce it (minimal and precise)

Get new SLZB-06
Configuration of Z2M:
`permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mqtt
serial:
  port: tcp://192.168.1.120:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
frontend:
  port: 8080
advanced:
  pan_id: 36990
  ext_pan_id:
    - 1
    - 2
    - 3
    - 4
    - 5
    - 6
    - 7
    - 8
  network_key:
    - 1
    - 1
    - 2
    - 7
    - 1
    - 7
    - 1
    - 1
    - 2
    - 7
    - 6
    - 2
    - 1
    - 7
    - 1
    - 3
  transmit_power: 20
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false`
(Network key has been shortened for security purposes)

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20221226

### Adapter

SLZB-06

### Setup

Z2M in Docker container on x86, SLZB-06 connected over Ethernet

### Debug log

Starting Zigbee2MQTT without watchdog.
[2024-11-10 11:32:59] info: 	z2m: Logging to console, file (filename: log.log)
[2024-11-10 11:32:59] info: 	z2m: Starting Zigbee2MQTT version 1.41.0 (commit #cc31cea)
[2024-11-10 11:32:59] info: 	z2m: Starting zigbee-herdsman (2.1.7)
[2024-11-10 11:33:00] info: 	zh:zstack:znp: Opening TCP socket with 192.168.88.120:6638
[2024-11-10 11:33:00] info: 	zh:zstack:znp: Socket connected
[2024-11-10 11:33:00] info: 	zh:zstack:znp: Socket ready
[2024-11-10 11:33:00] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-11-10 11:33:01] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-11-10 11:33:27] info: 	zh:controller: Wrote coordinator backup to '/app/data/coordinator_backup.json'
[2024-11-10 11:33:27] info: 	z2m: zigbee-herdsman started (reset)
[2024-11-10 11:33:27] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20221226,"transportrev":2},"type":"zStack3x0"}'
[2024-11-10 11:33:27] info: 	z2m: Currently 0 devices are joined.
[2024-11-10 11:33:27] warning: 	z2m: `permit_join` set to  `true` in configuration.yaml.
[2024-11-10 11:33:27] warning: 	z2m: Allowing new devices to join.
[2024-11-10 11:33:27] warning: 	z2m: Set `permit_join` to `false` once you joined all devices.
[2024-11-10 11:33:27] info: 	z2m: Zigbee: allowing new devices to join.
[2024-11-10 11:33:27] info: 	z2m: Connecting to MQTT server at mqtt://mqtt
[2024-11-10 11:33:27] info: 	z2m: Connected to MQTT server
[2024-11-10 11:33:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2024-11-10 11:33:27] error: 	z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'zh')
    at Bridge.publishDevices (/app/lib/extension/bridge.ts:781:43)
    at Bridge.start (/app/lib/extension/bridge.ts:175:20)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Controller.callExtensions (/app/lib/controller.ts:399:17)
    at Controller.start (/app/lib/controller.ts:218:9)
    at start (/app/index.js:154:5))
[2024-11-10 11:33:27] info: 	z2m: Started frontend on port 8080
TypeError: Cannot read properties of undefined (reading 'ID')
    at State.exists (/app/lib/state.ts:92:34)
    at Controller.start (/app/lib/controller.ts:223:32)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at start (/app/index.js:154:5)

ingheuer, 2024-11-23:
Same problem to me... any suggestions?

Koenkk, 2024-11-24:
Could you provide me your data/database.db entry?

m90att, 2024-11-25:
Hi, same for me...

`Nov 25 19:30:35 zigbee2mqtt npm[318]: Starting Zigbee2MQTT without watchdog.
Nov 25 19:30:37 zigbee2mqtt npm[318]: [2024-11-25 19:30:37] info:         z2m: Logging to console, file (filename: log.log)
Nov 25 19:30:37 zigbee2mqtt npm[318]: [2024-11-25 19:30:37] info:         z2m: Starting Zigbee2MQTT version 1.41.0 (commit #69f7282)
Nov 25 19:30:37 zigbee2mqtt npm[318]: [2024-11-25 19:30:37] info:         z2m: Starting zigbee-herdsman (2.1.7)
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] info:         z2m: zigbee-herdsman started (resumed)
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] info:         z2m: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] info:         z2m: Currently 0 devices are joined.
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] warning:         z2m: `permit_join` set to  `true` in configuration.yaml.
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] warning:         z2m: Allowing new devices to join.
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] warning:         z2m: Set `permit_join` to `false` once you joined all devices.
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] info:         z2m: Zigbee: allowing new devices to join.
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] info:         z2m: Connecting to MQTT server at mqtt://192.168.50.110:1883
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] info:         z2m: Connected to MQTT server
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] error:         z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'getDevice')
Nov 25 19:30:38 zigbee2mqtt npm[318]:     at Bridge.publishInfo (/opt/zigbee2mqtt/lib/extension/bridge.ts:746:69)
Nov 25 19:30:38 zigbee2mqtt npm[318]:     at Bridge.start (/opt/zigbee2mqtt/lib/extension/bridge.ts:174:20)
Nov 25 19:30:38 zigbee2mqtt npm[318]:     at processTicksAndRejections (node:internal/process/task_queues:95:5)
Nov 25 19:30:38 zigbee2mqtt npm[318]:     at Controller.callExtensions (/opt/zigbee2mqtt/lib/controller.ts:399:17)
Nov 25 19:30:38 zigbee2mqtt npm[318]:     at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:218:9)
Nov 25 19:30:38 zigbee2mqtt npm[318]:     at start (/opt/zigbee2mqtt/index.js:154:5))
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] info:         z2m: Started frontend on port 9442
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"69f7282","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"info","network":{>
Nov 25 19:30:38 zigbee2mqtt npm[318]: [2024-11-25 19:30:38] info:         z2m: Zigbee2MQTT started!`

m90att, 2024-11-26:
I have tried this in debian and ubuntu, with a full config and with the minimum config and I'm still seeing the same issue.  I'm considering installing an earlier version temporarily (but need to find out how to do that)!

Koenkk, 2024-11-27:
Can you try to remove your `data/database.db`, start z2m and see if it work now? If not repeat (stop z2m, remove database.db, start z2m) and provide me your debug log of this.

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

m90att, 2024-11-27:
Hi, thank you for the response.  I've tried deleting the db but it still shows the same issue.  I've done it again and attached the log.
[log.txt](https://github.com/user-attachments/files/17939554/log.txt)

m90att, 2024-11-28:
I have reproduced this error with a different dongle to eliminate that as the cause.

Koenkk, 2024-11-29:
@m90att thanks, could you try the following (in the logging you provided, the database.db file was already there before start)

1. Stop z2m
2. remove `database.db` file
3. Start z2m <-- provide me the debug log of this

Koenkk, 2024-12-01:
This only seems to have logging up to the `info` level, please provide the `debug` one.

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

m90att, 2024-12-01:
Ah, sorry @Koenkk .  Now attached with debug.  
[log_debug.txt](https://github.com/user-attachments/files/17970302/log_debug.txt)

m90att, 2024-12-02:
It looks like the issue has been resolved with version 1.42.0 commit: 861cba6, it's started up ok so I'll see how it goes

pxFIN, 2025-01-06:
I have similar challenge like @aquila and @ingheuer using zigbee2mqtt 1.4x and now 2.x in HA (no experience before 1.42 due to fact i just started experimenting with below mentioned adapter)

Adapter is ZBOSS (experimental firmware / nRF52840 dongle with latest firmware as mentioned from the page)

https://www.zigbee2mqtt.io/guide/adapters/zboss.html

Scenarios

1) If i setup everything from scratch all work.. just one aqara temp+humidity+pressure sensor
2) If i have to restart zigbee2mqtt for whatever reason -> see log
3) If after restart i delete database.db zigbee2mqtt restarts but i loose all sensors i have previously added

Log after restart

[zigbee2mqtt_2025-01-06T11-17-59.326Z.log](https://github.com/user-attachments/files/18318790/zigbee2mqtt_2025-01-06T11-17-59.326Z.log)

danielfrg, 2025-01-12:
I am having the same issue after upgrading to 2.0.0. Logs attached:

[logs.txt](https://github.com/user-attachments/files/18391271/logs.txt)

My dongle is the HA skyconnect with the latest firmware.

danielfrg, 2025-01-12:
More info: If I delete the z2m db file and start it the first time it works fine, but if i restart the container I get the error

pxFIN, 2025-01-13:
I have been testing latest dev version (z2m: Starting Zigbee2MQTT version 2.0.0-dev (commit #7392557)) and **good news** is latest ZBOSS Herdsman fixes helps with restarting issue but i still have the issue after restart sensor is lost and not discovered (weirdly yes it is found but cleaned as outdated?  -> z2m: Clearing outdated Home Assistant config)

Log of one sensor added and z2m restart

[zigbee2mqtt_edge_2025-01-13T09-20-29.579Z.log](https://github.com/user-attachments/files/18395419/zigbee2mqtt_edge_2025-01-13T09-20-29.579Z.log)

kirovilya, 2025-01-13:
@pxFIN I looked at your logs and I see that the next time you restart z2m, you have a network reset again... it's very strange. I'll take another look.

kirovilya, 2025-01-20:
@pxFIN Can you check how zboss works on the current dev/edge version?

pxFIN, 2025-01-20:
@kirovilya I can confirm issues seems to be fixed after installing the latest dev version which is using latest herdsman v3.2.3

1) No new network is created
2) Sensors are persistent and joined correctly after z2m restarts

closed 2025-03-31 by github-actions[bot]