Koenkk/zigbee2mqtt #24706: [New device support]: Tuya _TZE204_eekpf0ft TS0601 TR-M3Z ZigBee Radiator valve with thermostat

RainoPikkarainen, 2024-11-10:
### Link

https://www.aliexpress.com/item/1005007412184036.html

### Database entry

{"id":40,"type":"EndDevice","ieeeAddr":"0xa4c138e61f406c8e","nwkAddr":27112,"manufId":4417,"manufName":"_TZE204_eekpf0ft","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"�z�.i","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":74,"modelId":"TS0601","manufacturerName":"_TZE204_eekpf0ft","powerSource":3,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1731239207933}

### Zigbee2MQTT version

1.41.0

### Comments

Ordered from AliExress, didn't work out of the box, so tried converters from https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts.

External definition based on "fingerprint: tuya.fingerprint('TS0601', ['_TZE204_pcdmj88b']), model: 'TS0601_thermostat_4'"

- Changed the presets.
- Used the fixedValueConverter from https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs/tree/main, discussed in PR https://github.com/Koenkk/zigbee-herdsman-converters/pull/7406

### External definition

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fixedValueConverter = {
    thermostatScheduleDayMultiDP: (numberOfSchedules = 4) => {
        return {
            from: (v) => {
                const schedule = [];
                for (let index = 1; index < 4 * numberOfSchedules + 1; index = index + 4) {
                    schedule.push(
                        String(parseInt(v[index+0])).padStart(2, '0') + ':' +
                        String(parseInt(v[index+1])).padStart(2, '0') + '/' +
                        // @ts-ignore
                        (parseFloat((v[index+2] << 8) + v[index+3]) / 10.0).toFixed(1),
                    );
                }
                return schedule.join(' ');
            },
            to: (v) => {
                const payload = [0];
                const transitions = v.split(' ');
                if (transitions.length != numberOfSchedules) {
                    throw new Error(`Invalid schedule: there should be ${numberOfSchedules} transitions`);
                }
                for (const transition of transitions) {
                    const timeTemp = transition.split('/');
                    if (timeTemp.length != 2) {
                        throw new Error('Invalid schedule: wrong transition format: ' + transition);
                    }
                    const hourMin = timeTemp[0].split(':');
                    const hour = parseInt(hourMin[0]);
                    const min = parseInt(hourMin[1]);
                    const temperature = Math.floor(parseFloat(timeTemp[1]) * 10);
                    if (hour < 0 || hour > 24 || min < 0 || min > 60 || temperature < 50 || temperature > 300) {
                        throw new Error('Invalid hour, minute or temperature of: ' + transition);
                    }
                    payload.push(
                        hour,
                        min,
                        (temperature & 0xff00) >> 8,
                        temperature & 0xff,
                    );
                }
                return payload;
[... 91 lines cut ...]
        ],
    },
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Works:
- child lock
- battery
- current heating setpoint
- local temperature
- preset ('manual', 'schedule', 'eco', 'comfort', 'frost_protection' ,'holiday')
- schedules (6 schedule points per day)
- preset holiday temperature
- preset confort temperature
- preset eco temperature

Untested:
- boost heating
- frost protection
- scale protection
- local temperature calibration

RainoPikkarainen, 2024-11-10:
[image]

boboc5, 2024-11-18:
Hello Raino ! did you manage to make it to fully work ? I also bought the same thermostat valve 2 weeks ago and I tried it in ZHA but is not working correctly. With an external quirk (ME167) I managed to make it to show me the current temperature and the target temperature, and the child lock was working, but nothing more. I ordered another zigbee coordinator to be able to use it with zigbee2mqtt and I'm waiting to see if it will be ok.

RainoPikkarainen, 2024-11-19:
I believe most features available in this device are working. Certainly current_heating_setpoint and local_temperature are working, as are the scheduled times/temperatures and other presets.

- The boost heating is not working
  - or maybe that feature is just not there in the device, no mention of it in the manual
  - I think this feature would just turn the valve open regardless of any target or ambient temperatures
  - for use in automations, I would just set the current_heating_setpoint to 35 when "on" and 5 (or preset "off") when "off"
- running_state in the state seems to always be "idle"
- system_mode can be set to "heat"/"off" via a climate control card
  - but it has no effect, as far as I know
  - eg. if in preset "manual", changing current_heating_setpoint will trigger the valve even when system_mode is "off"

Once it gets a bit colder outside, I could test the "window detection", and if I could find an extra radiator valve, I could try connecting it to the device and test the "frost protection" (I think the device wants to calibrate with the valve before functioning, so it cannot just be tossed in a freezer).

RainoPikkarainen, 2024-11-19:
[image]

RainoPikkarainen, 2024-12-08:
- fixed running_state
- fixed system_mode
- tested window mode
- tested frost protection

boboc5, 2024-12-09:
Mine is not working at all, after the last update... :(

butuz1994, 2024-12-09:
Hello! Temperature calibration does not work. The sensor with the thermostat shows a high temperature, I change the settings, the temperature does not change. What could be the matter?

RainoPikkarainen, 2024-12-09:
> Mine is not working at all, after the last update... :(

The system_mode is now in use in my suggested external converter, so you could try to cycle it to 'off' / 'heat' with e.g. a climate UI card, so the valve would be on (and in one of the preset modes). Just in case the valve is in some kind of limbo state after different kinds of converter settings.

RainoPikkarainen, 2024-12-09:
> Hello! Temperature calibration does not work. The sensor with the thermostat shows a high temperature, I change the settings, the temperature does not change. What could be the matter?

Yes. Going from full -9.5 to 9.5 seems to only changes the local temperature by one degree and the state goes
`"local_temperature_calibration": -10` -> `"local_temperature_calibration": 2304`

Changing to localTempCalibration3 in `[47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration3],` seemed to work, so maybe no custom convertions are needed here.

butuz1994, 2024-12-09:
> tuya.valueConverter.localTempCalibration3

Unfortunately, this did not help, everything is the same, the temperature is regulated in a range of about 1 degree.

butuz1994, 2024-12-09:
> > tuya.valueConverter.localTempCalibration3
> 
> Unfortunately, this did not help, everything is the same, the temperature is regulated in a range of about 1 degree.

Can only be calibrated by manually entering values ​​larger than the range in zigbee2mqtt. when determining -30, I got the calibration I needed

RainoPikkarainen, 2024-12-09:
After changing to localTempCalibration3 (previously localTempCalibration2) and rebooting zigbee2mqtt, I get the expected 19 degree range in local temperature, in my four devices (in the clip going from 9.5 -> -9.5 -> 9.5).

https://github.com/user-attachments/assets/4741e632-1b64-4912-ab88-2974a730cb10

I wonder if there could be some revision differences in the devices? Doesn't have OTA, though.

RainoPikkarainen, 2024-12-10:
The new `tuya.thermostatScheduleDayMultiDPWithDayNumber` with transitionCount was merged recently to zigbee-herdsman-converters: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8414
(in release https://github.com/Koenkk/zigbee-herdsman-converters/releases/tag/v21.1.0)
Once that release version makes it to zigbee2mqtt (https://github.com/Koenkk/zigbee2mqtt/blob/1.42.0/package.json), it should be simpler to make this into a pull request.

RainoPikkarainen, 2024-12-27:
fixed 'window_open' to use values 0 & 1.

Z0ldberg, 2024-12-29:
Did you manage to connect this thermal head to zigbee2mqtt somehow?

RainoPikkarainen, 2024-12-30:
> Did you manage to connect this thermal head to zigbee2mqtt somehow?

Yes, I was able to connect the valve using the external definition procedure: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html.

Just submitted the PRs to merge the change into Zigbee2MQTT addon, so sometime in future the valve would be supported outof-the-box.

closed 2024-12-30 by Koenkk

Z0ldberg, 2024-12-30:
> Yes, I was able to connect the valve using the external definition procedure: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html.
>
> Just submitted the PRs to merge [...]

Tell me how I can add it to zigbee2mqtt now, I don’t understand it very well.

RainoPikkarainen, 2025-01-04:
> Tell me how I can add it to zigbee2mqtt now, I don’t understand it very well.

TR-M3Z valve is now supported out-of-the-box by Zigbee2MQTT v2.0.0 (https://github.com/zigbee2mqtt/hassio-zigbee2mqtt). So after updating your Z2M add-on to v2.0.0 you should be able to use this valve in HA.