Koenkk/zigbee2mqtt #24710: Philips Hue Bloom lamps stop responding after packets are routed through other devices

LukeT92, 2024-11-10:
### What happened?

Philips Hue "Bloom" lamps stop responding once Zigbee packets are routed through other devices, rather than directly from the coordinator. Zigbee2MQTT lists the lamp as [Philips 7299760PH](https://www.zigbee2mqtt.io/devices/7299760PH.html#philips-7299760ph)

I have three of these lamps, and run Zigbee2MQTT in docker. They work briefly after restarting the Zigbee2MQTT container, but given enough time, stop responding to commands, and are eventually listed as unavailable in the frontend / Home Assistant.

I'll focus on one of these lamps (ID: 0x8c22). I have sniffed the Zigbee traffic when the lamp stops responding, and I see a pattern... Initially, packets are addressed directly to the lamp from the coordinator (wpan.src16 == 0x0000, wpan.dst16 == 0x8c22). When packets are sent directly, the lamp is responsive, and replies from the lamp are sent directly to the coordinator. Everything works as intended.

Later, Zigbee routing determines that a stronger route is available, via another Philips Hue bulb (different model, ID: 0x34c7). From this point onwards, the coordinator always sends command packets to the lamp via this second bulb, i.e. 0x0000 -> 0x34c7 -> 0x8c22, but it seems the lamp is ignoring them. It looks like these lamps drop command packets that arrive via a route it wasn't expecting. "ZCL Level Control" and "ZCL Read Attribute" packets are ignored and the lamp cannot be controlled.

The lamp continues to broadcast Link Status packets every 15 seconds, and responds to Route Request packets from the coordinator and other devices. The Route Reply packets from the lamp are addressed directly to the coordinator, which I guess suggests the lamp wants to route directly to the coordinator, rather than via the second device. Despite the directly addressed Route Reply packets, Zigbee2MQTT continues to send commands via the two-hop route through 0x34c7, and the lamp is uncontrollable.

Restarting the Zigbee2MQTT container seemingly clears the route, and the lamp can be controlled again, up until the two-hop route is re-established and the problem reappears.

I've attached a Wireshark capture of the Zigbee packets between the devices involved (the coordinator at 0x0000, the problematic lamp at 0x8c22, and the intermediate routing bulb at 0x34c7). Packet numbers 90 and 91 are the first time the coordinator sends a packet to the Bloom lamp via the two-hop route, and from this point onwards, the lamp is unresponsive.

I have found that loading the network map can coerce the coordinator into using the two-hop route to repro the problem rather than waiting for it to happen naturally, so this is what the Wireshark capture starts with.

I should note that I had no issues with these lamps when they were connected to the official Philips Hue bridge.

[Zigbee_Routing_PhilipsHueBloom_RoutingFail.zip](https://github.com/user-attachments/files/17691835/Zigbee_Routing_PhilipsHueBloom_RoutingFail.zip)

### Zigbee2MQTT version

1.41.0-dev commit: a353acc

### Adapter firmware version

8.0.2 [GA]

### Adapter

SONOFF Dongle-E

### Setup

Docker Container, x86-64, on Ubuntu Server 20.04.6 LTS

mattnl, 2024-11-14:
I also have the same problem with hue light bulbs. As soon as they have no direct connection to the coordinator the light bulbs do not respond and cannot be controlled.
Z2M version: 1.41.0 commit: cc31cea
Adapter: SLZB-06M
Tested with Firmware: 8.0.2 [GA] and 7.4.4, 7.4.2, Revision: 20240510

LukeT92, 2024-11-14:
Hi mattnl. It looks like your SLZB-06M is a SiliconLabs / EmberZNet based device, which I think is the source of the problems...

Since I opened this issue, I've replaced the SONOFF ZBDongle-E I was using as the coordinator with a SONOFF ZBDongle-P (Texas Instruments chipset with the zStack driver), and the difference is stark. My Philips Hue Bloom lights are solid now, they haven't stopped responding once. Looking at Wireshark, I can see the lamps establish an asymmetric route to/from the coordinator, via the other nearby routers / bulbs. The repeaters all appear to be working fine, and all of my devices are listed as Online in the frontend and Home Assistant.

I'm new to Zigbee development, so I spent some time looking through the code in zigbee2mqtt / zigbee-herdsman to figure out where the routing decisions are made. I can't see anything obvious in the stack on Github, so I can only assume that it's the firmware on the dongle that manages this, and that zigbee-herdsman is just handing command payloads to the dongle via the firmware's serial API. It's then up to the dongle's firmware to build the Zigbee route table and manage sending those command payloads out.

If that's the case, then the ZBDongle-E (and possibly all EmberZNet / Silicon Labs based chipsets) have some serious routing issues. It seems to ignore Route Reply packets from endpoint devices, or doesn't ask for a new route via Route Request packets when it should. It stubbornly continues to use routes that are clearly dead, rather than trying to re-establish new ones.

I'm curious to know if others using EmberZNet / Silicon Labs based dongles see the same issue with Philips bulbs. Note, the problem only manifests when there is no strong radio link directly between the bulb and the coordinator, meaning a route through an intermediate repeater device is needed.

mattnl, 2024-11-15:
Hi Luke,
thanks for your reply. The coordinator (SLZB-06M) is indeed a EmberZNet based device.
I do not think that this coordinator has a general problem with routing. I have a some router-nodes and quite a few endpoint-nodes which have no direct connection to the coordinator but work perfectly fine. The only problematic devices that won't work without a direct connection seem to be the hue light bulbs.

Nerivec, 2024-11-17:
@LukeT92 I've tried to reproduce the behavior you described with the Dongle-E and some crappy LQI devices (with network map, and a few variations), haven't seen any issue, it goes through other routers fine. If it was a general issue though, we'd have no doubt seen it before with some of the large networks using EmberZNet and Z2M (and I know of a couple composed mainly of Hue & Inovelli devices).
Seems to be something about that device, or the router it is going through.

Can you provide the Z2M log that goes with that packet capture, the unfiltered packet capture (too much stuff is missing, like source routing), and the network key to decrypt the packet capture (in DM on Discord or via private git repo if you prefer)?

_Routing is pretty much only network-driven (Z2M configures a few options at best)._

https://community.silabs.com/s/article/how-does-source-routing-work-with-ezsp-x?language=en_US

closed 2025-01-24 by github-actions[bot]