Koenkk/zigbee2mqtt #24712: Broadcasts failing on ember after migration #22453

IgorDancik, 2024-11-10:
### What happened?

Problem was closed, but not fixed. Version 8.0.2 has the same problem as issue #22453 
IMHO #22453  is still pending

### What did you expect to happen?

Linking IKEA PARASOLL Door/window sensor

### How to reproduce it (minimal and precise)

Failed broadcast/ interview

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

8.0.2

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

RPi4 add-on

### Debug log

error 2024-11-10 17:09:16zh:ember: Delivery of BROADCAST failed for '65532'.
error 2024-11-10 17:09:31zh:ember: Delivery of BROADCAST failed for '65533'.
error 2024-11-10 17:12:14zh:controller: Interview failed for '0xf84477fffedcc1d5 with error 'Error: DatabaseEntry with ID '21' does not exist'

Ricc68, 2024-11-19:
You should double check the position of the adapter, at least 1m from any other radio device, use USB2 port or USB2 extension cable because usb3 generates interferences, set your ZigBee channel far from WiFi channels in use in your area, make sure the quality of the signal is good for all your devices and eventually add repeaters. Done that myself and broadcast failed messages disappeared for me.

sstarcher, 2024-11-21:
I also had this issue go completely away after removing every device from my network and readding 50 devices 😢

Ricc68, 2024-12-02:
> Same issue since a long time. Currently i try to add a IKEA GU10 bulb https://www.zigbee2mqtt.io/devices/LED1923R5.html and getting those errors
> 
> ```
> error 2024-12-02 19:10:27zh:controller: Interview failed for '0x8c65a3fffec8a8f3 with error 'Error: Interview failed because can not get active endpoints ('0x8c65a3fffec8a8f3')'
> error 2024-12-02 19:10:27z2m: Failed to interview '0x8c65a3fffec8a8f3', device has not successfully been paired
> error 2024-12-02 19:10:28z2m: Failed to read state of '0x8c65a3fffec8a8f3' after reconnect (ZCL command 0x8c65a3fffec8a8f3/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '49284'.))
> error 2024-12-02 19:10:28z2m: Failed to read state of '0x8c65a3fffec8a8f3' after reconnect (ZCL command 0x8c65a3fffec8a8f3/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '57358'.))
> error 2024-12-02 19:10:30zh:ember: Delivery of BROADCAST failed for '65533'.
> error 2024-12-02 19:10:30z2m: Failed to read state of '0x8c65a3fffec8a8f3' after reconnect (ZCL command 0x8c65a3fffec8a8f3/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '46200'.))
> error 2024-12-02 19:10:31z2m: Failed to read state of '0x8c65a3fffec8a8f3' after reconnect (ZCL command 0x8c65a3fffec8a8f3/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '26546'.))
> error 2024-12-02 19:10:31z2m: Failed to read state of '0x8c65a3fffec8a8f3' after reconnect (Cannot read properties of undefined (reading 'manufacturerID'))
> error 2024-12-02 19:10:31z2m: Failed to read state of '0x8c65a3fffec8a8f3' after reconnect (ZCL command 0x8c65a3fffec8a8f3/1 genOnOff.read(["onOff"], {"timeout":10000,"disableRe```
> 
> Sonoff Dongle E
> Firmware: 7.4.5
> ```

This seems a problem with network stability.
Fix the IKEA bulb issues with failed interview and read state then it's very likely that the broadcast issues will go away.

smartmatic, 2024-12-02:
> Fix the IKEA bulb issues with failed interview and read state then it's very likely that the broadcast issues will go away.

Ok, but how to to fix? I tried a lot like go close to the coordinator, place the coordinator to a different place etc.

Ricc68, 2024-12-02:
> Ok, but how to to fix? I tried a lot like go close to the coordinator, place the coordinator to a different place etc.

Have you tried changing ZigBee channel to 25, far from WiFi channels?
In addition to moving the dongle far from router, use an USB2 port or extension cable, etc

smartmatic, 2024-12-02:
> > Ok, but how to to fix? I tried a lot like go close to the coordinator, place the coordinator to a different place etc.
> 
> Have you tried changing ZigBee channel to 25, far from WiFi channels? In addition to moving the dongle far from router, use an USB2 port or extension cable, etc

My Zigbee channel is 15 and my Wifi Channel is 6

smartmatic, 2024-12-03:
> > My Zigbee channel is 15 and my Wifi Channel is 6
> 
> Definitely not good as they overlap. Read this and you'll see that channel 25 is the way to go:
> 
> https://www.metageek.com/training/resources/zigbee-wifi-coexistence/

No question, it make sense to switch the channel but it means really pain. I have 80 Zigbee devices which needs to be repaired than. Yes it is worth to improve Zigbee network performance but how to start best beacause i still have a Hue Bridge communicating on via channel 25?

Ricc68, 2024-12-03:
> No question, it make sense to switch the channel but it means really pain. I have 80 Zigbee devices which needs to be repaired than. Yes it is worth to improve Zigbee network performance but how to [...]

Then maybe check if hue supports channel 26 and eventually start by freeing up channel 25.
Then migrate to 25 the rest.

smartmatic, 2024-12-06:
> Then maybe check if hue supports channel 26 and eventually start by freeing up channel 25. Then migrate to 25 the rest.
Done, re-paired 100 Devices, changed my Zigbee Channel to 25 and changed as well my Zigbee coordinator to a PoE device. 

But maybe it should be considered to change the default zigbee2mqtt channel 11 to 25? Otherwise their is a high chance that new users will run into trouble.

Ricc68, 2024-12-07:
@Nerivec this seems not a bad idea, would it be reasonable to set by default channel 25 so new users don't fall into the same WiFi coexistence issues?
Channel 25 seems a safe channel for ZigBee.

@IgorDancik can the issue be closed as it seems it's not an issue with z2m anymore?

Nerivec, 2024-12-07:
[image]

Unfortunately, which channel is most likely to conflict is entirely dependent on the environment (yours, your neighbor's, your country's...).

@smartmatic How come you had to re-pair devices, was it because you reset the network when you migrated coordinator? Otherwise Z2M has support for [channel change](https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#changing-the-zigbee-channel) (tested for zstack & ember).
PS: You can also migrate coordinator by using a tool like [Ember ZLI](https://github.com/Nerivec/ember-zli/wiki) to backup/restore NVM3 tokens (should allow to [restore your network without loss](https://github.com/Nerivec/ember-zli/issues/5)).

I'm closing this since the issue is not something that can be fixed from Z2M (already optimized to avoid channel interference as much as possible a while back), but feel free to reply here for this.

For anyone that still has issues with transmit failures (after reducing external interferences as much as possible), you can try [customizing the `CCA_MODE` for the adapter](https://www.zigbee2mqtt.io/guide/adapters/emberznet.html#expert-customizing-stack-configuration), but make sure you [understand the implications](https://www.mathworks.com/help/comm/ug/aloha-and-csma-ca-packetized-wireless-networks.html).

closed 2024-12-07 by Nerivec