Koenkk/zigbee2mqtt #24713: Lidl/Livarno Lux customized power on behavior missing a field

RubenKelevra, 2024-11-10:
### What happened?

I've noticed that my Livrano Lux lights (vendor in z2m is 'Lidl') has an option to set the "Color power on behavior" to customized. But there's no way to set what this customization is supposed to be?

Changing the color/color temperature setting of a light, while it is turned off while `customized` is selected, does not change what the light is turning on to.

So is there a field missing where I can set the customized color or is this function non-existing and the option `customized` does not exist?

[image]

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20221226

### Adapter

Slaesh's CC2652 RB stick

### Setup

Add-On on Home Assistant OS on RasPi 4B

RubenKelevra, 2025-01-13:
@bramstroker the bot is kinda annoying. Is it possible to just mark things stale if there's an open question tagged with "question". So the bot is not constantly threatening to close valid tickets and instead only closes ones where the authors did not provide necessary information after being asked for completion?

bramstroker, 2025-01-13:
> @bramstroker the bot is kinda annoying. Is it possible to just mark things stale if there's an open question tagged with "question". So the bot is not constantly threatening to close valid tickets [...]

@RubenKelevra Are you tagging me by mistake? This is not my repository ;-)

RubenKelevra, 2025-01-13:
@bramstroker yeah, sorry. Normally Github suggest the right username as first entry, but this time it seems it was the wrong one 🥲

Koenkk, 2025-01-13:
TBH I have no clue what this option does, @robvanoostenrijk, since you made https://github.com/Koenkk/zigbee-herdsman-converters/pull/7673, do you have any clue?

RubenKelevra, 2025-01-13:
@Koenkk die idea is that you can set a customized color temperature or RGB color while the light is off. So if you turn it on, it will immediately show this color.

While you can send a color/color temperature with the "turn_on" command, that's not possible for turning the light on by using a switch to cut/connect it to power.

RubenKelevra, 2025-03-15:
Not fixed

closed 2025-05-25 by github-actions[bot]