Koenkk/zigbee2mqtt #24721: [Wrong device]: GIRIER JR-ZPM03 smart plug

sehraf, 2024-11-11:
### Link

https://a.aliexpress.com/_Eur8WjX

### Model

JR-ZPM03

### Description

smart plug with power metering

### Vendor

GIRIER

### Picture (link)

https://ae-pic-a1.aliexpress-media.com/kf/S16d4fe526eca4d31b50253f356aed674I.jpg_640x640.jpg_.webp

### Database entry

{"id":31,"type":"Router","ieeeAddr":"0x70b3d52b600fca15","nwkAddr":1491,"manufId":4660,"manufName":"_TZ3000_okaz9tjs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,6,3,4,5,57345,2820,1794],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_okaz9tjs","powerSource":1,"zclVersion":2,"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZSKT31BS102"}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":232,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":0}},"genOnOff":{"attributes":{"onOff":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0014bde8e8","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0}],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZSKT31BS102","zclVersion":2,"interviewCompleted":true,"meta":{"configured":332242049,"reporting":1},"lastSeen":1731315378807}

### Notes

The device is detected as [BSD29_1](https://www.zigbee2mqtt.io/devices/BSD29_1.html#tuya-bsd29_1) and seem to have a troublesome history:

- https://github.com/Koenkk/zigbee2mqtt/issues/23818
- https://github.com/Koenkk/zigbee2mqtt/issues/18642
- https://github.com/Koenkk/zigbee2mqtt/issues/18151
- https://github.com/Koenkk/zigbee2mqtt/issues/17319

While the device does work as [BSD29_1](https://www.zigbee2mqtt.io/devices/BSD29_1.html#tuya-bsd29_1), I think it is still wrong:

- The are no options for calibration (which BSD29_1 should have). 
- [minor] The image is very different.
- [TS011F_plug_3](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html) was mentioned as the correct mode, but this one is also supposed to have calibration options.
- The device is listed under OTA, but does not respond to any requests (see https://github.com/Koenkk/zigbee2mqtt/issues/18151), firmware version on my two plugs is `500+TZSKT31BS102`
- The model `JR-ZPM03` does not appear under [supported devices](https://www.zigbee2mqtt.io/supported-devices/) 
- The vendor `GIRIER` does not have any smart plug under [supported devices](https://www.zigbee2mqtt.io/supported-devices/)
- The packages calles the model `ZMP03` vs `ZPM03` what is printed on the device, lol

Since the device works, I'd at least see the model listed as a supported device :)

Here are some images:
[image]
[image]
[image]
[image]

neovinter, 2024-11-17:
I have same issue with that socket

Oggerschummer, 2024-12-12:
Here as well, 8 plugs, all the same.

CrazYoshi, 2025-02-28:
I have the same issue.
Furthermore, I have more than one plug. sometimes they stop communicating power value and I need to pair them again to see this value again.
Anyone is experiencing the same?

CrazYoshi, 2025-03-11:
How can I help to correctly integrate this device in Z2M?

closed 2025-05-19 by github-actions[bot]