Koenkk/zigbee2mqtt #24722: [New device support]: TuYa TS0105 Smart Switch Zigbee

calbeeb-grape, 2024-11-11:
### Link

https://pl.aliexpress.com/item/4000369677580.html?

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0xa4c138e3bb0469b9","nwkAddr":2968,"manufId":4107,"manufName":"_TZE600_ibfoajdt","powerSource":"Mains (single phase)","modelId":"TS0105","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,61184,10],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0105","manufacturerName":"_TZE600_ibfoajdt","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1703928654862,"defaultSendRequestWhen":"immediate"}

### Zigbee2MQTT version

1.41.0-1

### Comments

The below external definition from #18346 was working perfectly under Zigbee2MQTT version 1.36.0. However, it is no longer working in the latest version. 

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
// Since a lot of TuYa devices use the same modelID, but use different datapoints
// it's necessary to provide a fingerprint instead of a zigbeeModel
fingerprint: [
{
// The model ID from: Device with modelID 'TS0601' is not supported
// You may need to add \u0000 at the end of the name in some cases
modelID: 'TS0105',
// The manufacturer name from: Device with modelID 'TS0601' is not supported.
manufacturerName: '_TZE600_iypcp4py',
},
],
model: 'TS0105_new',
vendor: 'TuYa',
description: 'Fill in a description of the device here',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
configure: tuya.configureMagicPacket,
exposes: [
e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET),
],
meta: {
multiEndpoint: true,
tuyaDatapoints: [
[1, 'state_l1', tuya.valueConverter.onOff],
[2, 'state_l2', tuya.valueConverter.onOff],
[3, 'state_l3', tuya.valueConverter.onOff],
],
},
endpoint: (device) => {
return {'l1': 1, 'l2': 1, 'l3': 1};
},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Not Supported

Koenkk, 2024-11-11:
> However, it is no longer working in the latest version.

What doesn't work?

calbeeb-grape, 2024-11-11:
The "Support status" from the device page simply shows "Not Supported" ,and I am no longer able to switch it on/off

Koenkk, 2024-11-12:
Added the device.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `TS0105.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

calbeeb-grape, 2024-11-13:
Sure, I have just submitted a product image, thanks again for adding the device!

closed 2025-01-20 by github-actions[bot]