Koenkk/zigbee2mqtt #24726: Zigbee2MQTT HOST_FATAL_ERROR with newly flashed  Sonoff Dongle V2

Rainking80, 2024-11-11:
### What happened?

I flashed my Sonoff Dongle E V2 to the latest FW: ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl and now i see this endlessly repeating:

[17:26:50] INFO: Preparing to start...
[17:26:50] INFO: Socat not enabled
[17:26:50] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-11-11 17:26:52] info: 	z2m: Logging to console, file (filename: log.log)
[2024-11-11 17:26:52] info: 	z2m: Starting Zigbee2MQTT version 1.41.0 (commit #unknown)
[2024-11-11 17:26:52] info: 	z2m: Starting zigbee-herdsman (2.1.7)
[2024-11-11 17:26:52] info: 	zh:ember: Using default stack config.
[2024-11-11 17:26:52] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-11-11 17:26:52] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-11-11 17:26:52] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-11 17:26:52] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-11-11 17:26:52] info: 	zh:ember:uart:ash: Serial port opened
[2024-11-11 17:26:52] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-11 17:26:55] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-11 17:26:55] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-11 17:26:57] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-11 17:26:57] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-11 17:27:00] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-11 17:27:00] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-11 17:27:02] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-11 17:27:02] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-11 17:27:05] error: 	z2m: Error while starting zigbee-herdsman
[2024-11-11 17:27:05] error: 	z2m: Failed to start zigbee
[2024-11-11 17:27:05] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-11-11 17:27:05] error: 	z2m: Exiting...
[2024-11-11 17:27:05] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:684:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1598:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)

my zigbee2mqtt config:
serial:
  port: /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_f45097c0c312ef118bf96db8bf9df066-if00-port0
  baudrate: 115200
  adapter: ember
  zigbee_herdsman_debug: true

i tried to fix that since 3 days and i am out of ideas. If you have any suggestion what i can try please help me :)

### What did you expect to happen?

Zigbee2MQTT starting sucessfully

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

ncp-uart-hw-v7.4.2.0-zbdonglee-115200

### Adapter

Sonoff Zigbee 3 Dongle E V2

### Setup

Home Assistant OS on Pi5

Ricc68, 2024-11-11:
Very similar to the issue #23282 I am experiencing.

Rainking80, 2024-11-11:
@Ricc68 thanks for linking the issue! Unfortunately restart wont solve the issue at my system. I even tried ZHA and Z2M and deleted all configs, reinstalled plugins, down graded HA Core.... but still no luck.

Ricc68, 2024-11-11:
> ... used skyconnect and sonoff zigbee 3 E Dongle V2.
> 

Are you having the same issue with both?

Rainking80, 2024-11-11:
yes, seems to be something with HA OS. Started about 2 weeks ago on both of my home assistant machines(pi4 and pi5). I tried downgrading of my HA OS version, but problem still exists.

Rainking80, 2024-11-11:
after a full shutdown and reboot, i see sometimes more progress in the log:

[22:42:39] INFO: Preparing to start...
[22:42:39] INFO: Socat not enabled
[22:42:39] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-11-11 22:42:42] info: 	z2m: Logging to console, file (filename: log.log)
[2024-11-11 22:42:42] info: 	z2m: Starting Zigbee2MQTT version 1.41.0-dev (commit #22e13c5)
[2024-11-11 22:42:42] info: 	z2m: Starting zigbee-herdsman (2.1.7)
[2024-11-11 22:42:42] info: 	zh:ember: Using default stack config.
[2024-11-11 22:42:42] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-11-11 22:42:42] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-11-11 22:42:42] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-11 22:42:42] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-11-11 22:42:42] info: 	zh:ember:uart:ash: Serial port opened
[2024-11-11 22:42:42] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-11 22:42:43] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-11-11 22:42:43] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-11-11 22:42:43] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-11-11 22:42:43] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2024-11-11 22:42:43] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.1 [GA]","build":0,"major":7,"minor":4,"patch":1,"special":0,"type":170}
[2024-11-11 22:42:43] info: 	zh:ember: [STACK STATUS] Network up.
[2024-11-11 22:42:43] info: 	zh:ember: [INIT TC] Adapter network does not match config. Leaving network...
[2024-11-11 22:42:44] info: 	zh:ember: [STACK STATUS] Network down.
[2024-11-11 22:42:44] info: 	zh:ember: [INIT TC] No valid backup found.
[2024-11-11 22:42:44] info: 	zh:ember: [INIT TC] Forming from config.
[2024-11-11 22:42:44] info: 	zh:ember: [INIT FORM] Forming new network with: {"panId":61483,"extendedPanId":[240,167,41,254,166,82,163,232],"radioTxPower":5,"radioChannel":11,"joinMethod":0,"nwkManagerId":0,"nwkUpdateId":0,"channels":134215680}
[2024-11-11 22:42:44] info: 	zh:ember: [STACK STATUS] Network up.
[2024-11-11 22:42:44] info: 	zh:ember: [INIT FORM] New network formed!
[2024-11-11 22:42:44] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2024-11-11 22:42:44] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt/coordinator_backup.json'
[2024-11-11 22:42:59] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=276:"EXPORT_KEY" Seq=43 Len=25].
[2024-11-11 22:42:59] error: 	zh:ember: Adapter fatal error: ASH_ERROR_TIMEOUTS
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   Total frames: RX=48, TX=94
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   DATA frames : RX=47, TX=45
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   DATA bytes  : RX=428, TX=505
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=48
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-11-11 22:42:59] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-11-11 22:42:59] error: 	z2m: Error while starting zigbee-herdsman
[2024-11-11 22:42:59] error: 	z2m: Failed to start zigbee
[2024-11-11 22:42:59] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-11-11 22:42:59] error: 	z2m: Exiting...
[2024-11-11 22:42:59] error: 	z2m: Error: ASH_ERROR_TIMEOUTS
    at Ezsp.ezspSendUnicast (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:4817:19)
    at /app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1917:41
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at EmberAdapter.sendZdo (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1867:16)
    at Device.updateActiveEndpoints (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:1067:26)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:206:13)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)

Nerivec, 2024-11-13:
What was the previous version on your adapter?
Since you appear to have no previous config in Z2M (nothing to backup, so it's easy), I'd suggest you try clearing the NVM3, either with [ember-zli](https://github.com/Nerivec/ember-zli/wiki/Bootloader#clear-nvm3) or by flashing the [nvm3_initfile](https://github.com/xsp1989/zigbeeFirmware/raw/refs/heads/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/nvm3_initfile.gbl) (same result). At the very least, it will eliminate the possibility of a mismatch in the firmware that could be creating troubles (can happen after jumping version like v6 > v7).

Then make sure you flash the latest 7.4 version: https://github.com/darkxst/silabs-firmware-builder/raw/refs/heads/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl
(or you can go to [version 8](https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update2) too)

Rainking80, 2024-11-13:
damn, still doing the thing:

```
[17:55:10] INFO: Preparing to start...
[17:55:10] INFO: Socat not enabled
[17:55:10] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-11-13 17:55:12] info: 	z2m: Logging to console, file (filename: log.log)
[2024-11-13 17:55:12] info: 	z2m: Starting Zigbee2MQTT version 1.41.0 (commit #unknown)
[2024-11-13 17:55:12] info: 	z2m: Starting zigbee-herdsman (2.1.7)
[2024-11-13 17:55:13] info: 	zh:ember: Using default stack config.
[2024-11-13 17:55:13] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-11-13 17:55:13] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-11-13 17:55:13] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-13 17:55:13] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-11-13 17:55:13] info: 	zh:ember:uart:ash: Serial port opened
[2024-11-13 17:55:13] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-13 17:55:15] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-13 17:55:15] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-13 17:55:18] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-13 17:55:18] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-13 17:55:20] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-13 17:55:20] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-13 17:55:23] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-13 17:55:23] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-13 17:55:25] error: 	z2m: Error while starting zigbee-herdsman
[2024-11-13 17:55:25] error: 	z2m: Failed to start zigbee
[2024-11-13 17:55:25] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-11-13 17:55:25] error: 	z2m: Exiting...
[2024-11-13 17:55:25] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:684:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1598:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)
```

Nerivec, 2024-11-13:
That zlib error with ember-zli likely means you are using an older NodeJS version (it requires >=20.15.0).

---

That looping of `ASH Adapter reset` usually means one of the following:
- The path is not right
- The baudrate is not right

Basically, it can't communicate to the adapter properly (or at all).

After flashing the NVM3, did you re-flash one of the firmware files I linked before?
Can you try running with `log_level: debug` in your configuration.yaml so I can get some more details? _You can find me on Discord, DM me after joining zigbee2mqtt server, if you prefer sharing the logs that way._

```yaml
advanced:
  log_level: debug
```

Rainking80, 2024-11-13:
here ist the log_level: debug:

`[19:19:34] INFO: Preparing to start...
[19:19:34] INFO: Socat not enabled
[19:19:34] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-11-13 19:19:36] info: 	z2m: Logging to console, file (filename: log.log)
[2024-11-13 19:19:36] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2024-11-13 19:19:36] info: 	z2m: Starting Zigbee2MQTT version 1.41.0 (commit #unknown)
[2024-11-13 19:19:36] debug: 	z2m: sd-notify loaded
[2024-11-13 19:19:36] info: 	z2m: Starting zigbee-herdsman (2.1.7)
[2024-11-13 19:19:36] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":1321,\"extendedPanID\":[23,58,119,75,209,77,211,4],\"channelList\":[25],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"baudRate\":115200,\"rtscts\":false,\"path\":\"/dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_f45097c0c312ef118bf96db8bf9df066-if00-port0\",\"adapter\":\"ezsp\"},\"adapter\":{\"disableLED\":false}}"'
[2024-11-13 19:19:36] debug: 	zh:ezsp: Adapter concurrent: 8
[2024-11-13 19:19:36] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":1321,"extendedPanID":[23,58,119,75,209,77,211,4],"channelList":[25]},"serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_f45097c0c312ef118bf96db8bf9df066-if00-port0","adapter":"ezsp"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2024-11-13 19:19:36] warning: 	zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2024-11-13 19:19:36] debug: 	zh:ezsp:uart: RTS/CTS config is off, enabling software flow control.
[2024-11-13 19:19:36] debug: 	zh:ezsp:uart: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_f45097c0c312ef118bf96db8bf9df066-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2024-11-13 19:19:36] debug: 	zh:ezsp:uart: Serialport opened
[2024-11-13 19:19:36] debug: 	zh:ezsp:uart: Uart reseting
[2024-11-13 19:19:36] debug: 	zh:ezsp:uart: --> Write reset
[2024-11-13 19:19:36] debug: 	zh:ezsp:uart: --> [1ac038bc7e]
[2024-11-13 19:19:36] debug: 	zh:ezsp:uart: -?- waiting reset
[2024-11-13 19:19:46] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":-1} after 10000ms
[2024-11-13 19:19:46] error: 	zh:ezsp:ezsp: Connection attempt 1 error: Error: Failure to connect
[2024-11-13 19:19:46] debug: 	zh:ezsp:uart: Port closed. Error? null
[2024-11-13 19:19:46] debug: 	zh:ezsp:ezsp: onSerialClose()
[2024-11-13 19:19:46] debug: 	zh:ezsp:driv: onEzspClose()
[2024-11-13 19:19:46] debug: 	zh:ezsp: onDriverClose()
[2024-11-13 19:19:51] debug: 	zh:ezsp:ezsp: Next attempt 2
[2024-11-13 19:19:51] debug: 	zh:ezsp:uart: RTS/CTS config is off, enabling software flow control.
[2024-11-13 19:19:51] debug: 	zh:ezsp:uart: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_f45097c0c312ef118bf96db8bf9df066-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2024-11-13 19:19:51] debug: 	zh:ezsp:uart: Serialport opened
[2024-11-13 19:19:51] debug: 	zh:ezsp:uart: Uart reseting
[2024-11-13 19:19:51] debug: 	zh:ezsp:uart: --> Write reset
[2024-11-13 19:19:51] debug: 	zh:ezsp:uart: --> [1ac038bc7e]
[2024-11-13 19:19:51] debug: 	zh:ezsp:uart: -?- waiting reset
[2024-11-13 19:20:01] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":-1} after 10000ms
[2024-11-13 19:20:01] error: 	zh:ezsp:ezsp: Connection attempt 2 error: Error: Failure to connect
[2024-11-13 19:20:01] debug: 	zh:ezsp:uart: Port closed. Error? null
[2024-11-13 19:20:01] debug: 	zh:ezsp:ezsp: onSerialClose()
[2024-11-13 19:20:01] debug: 	zh:ezsp:driv: onEzspClose()
[2024-11-13 19:20:01] debug: 	zh:ezsp: onDriverClose()
[2024-11-13 19:20:11] debug: 	zh:ezsp:ezsp: Next attempt 3
[2024-11-13 19:20:11] debug: 	zh:ezsp:uart: RTS/CTS config is off, enabling software flow control.
[2024-11-13 19:20:11] debug: 	zh:ezsp:uart: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_f45097c0c312ef118bf96db8bf9df066-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2024-11-13 19:20:11] debug: 	zh:ezsp:uart: Serialport opened
[2024-11-13 19:20:11] debug: 	zh:ezsp:uart: Uart reseting
[2024-11-13 19:20:11] debug: 	zh:ezsp:uart: --> Write reset
[2024-11-13 19:20:11] debug: 	zh:ezsp:uart: --> [1ac038bc7e]
[2024-11-13 19:20:11] debug: 	zh:ezsp:uart: -?- waiting reset
[2024-11-13 19:20:12] debug: 	zh:ezsp:uart: <-- [c1020b0a527e]
[2024-11-13 19:20:12] debug: 	zh:ezsp:uart: <-- RSTACK c1020b0a527e
[2024-11-13 19:20:12] debug: 	zh:ezsp:uart: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e
[2024-11-13 19:20:12] debug: 	zh:ezsp:uart: <-- RSTACK c1020b0a527e
[... cut ...]
[2024-11-13 19:21:04] debug: 	zh:ezsp:uart: --> [5c5721a9512addd07e]
[2024-11-13 19:21:04] debug: 	zh:ezsp:uart: -?- rewaiting (6)
[2024-11-13 19:21:08] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":6} after 4000ms
[2024-11-13 19:21:08] error: 	zh:ezsp:uart: -!- break rewaiting (6)
[2024-11-13 19:21:08] error: 	zh:ezsp:uart: Can't resend DATA frame (5,4,1): 1500010500
[2024-11-13 19:21:08] error: 	zh:ezsp:ezsp: Watchdog heartbeat timeout Error: Failure send nop:{"type":"Buffer","data":[21,0,1,5,0]}`

Rainking80, 2024-11-13:
> That zlib error with ember-zli likely means you are using an older NodeJS version (it requires >=20.15.0).
> 
> _Make sure to always use `Software flow control, rtscts: false` with Dongle-E, it does not support hardware flow control. I saw you set it as true in ember-zli screenshot._
> 
> That looping of `ASH Adapter reset` usually means one of the following:
> 
> * The path is not right
> * The baudrate is not right
> 
> Basically, it can't communicate to the adapter properly (or at all).
> 
> After flashing the NVM3, did you re-flash one of the firmware files I linked before? Can you try running with `log_level: debug` in your configuration.yaml so I can get some more details? _You can find me on Discord, DM me after joining zigbee2mqtt server, if you prefer sharing the logs that way._
> 
> ```yaml
> advanced:
>   log_level: debug
> ```

thanks i will set rtscts to false and re-test

Nerivec, 2024-11-13:
I think something is wrong with your configuration. That debug log is using `ezsp` not `ember` which doesn't match the config you posted before.
Check that there isn't a sync problem between the configuration in the Z2M add-on page, and the yaml file itself.

Though since both drivers and ember-zli appear to be acting the same (adapter not responding most of the time), I'm going to guess that the "end problem" is something about the adapter or the hardware it is plugged into. Make sure to eliminate the usual, power cable/supply, USB cable.
If you get a chance, grab the logs from `dmesg`, to see what the system sees.

Rainking80, 2024-11-13:
> I think something is wrong with your configuration. That debug log is using `ezsp` not `ember` which doesn't match the config you posted before. Check that there isn't a sync problem between the [...]

sorry, i tried both. i read that some people had stability issues with ember. here is the ember log:

[20:28:17] INFO: Preparing to start...
[20:28:18] INFO: Socat not enabled
[20:28:18] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-11-13 20:28:20] info: 	z2m: Logging to console, file (filename: log.log)
[2024-11-13 20:28:20] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2024-11-13 20:28:20] info: 	z2m: Starting Zigbee2MQTT version 1.41.0 (commit #unknown)
[2024-11-13 20:28:20] debug: 	z2m: sd-notify loaded
[2024-11-13 20:28:20] info: 	z2m: Starting zigbee-herdsman (2.1.7)
[2024-11-13 20:28:20] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":1321,\"extendedPanID\":[23,58,119,75,209,77,211,4],\"channelList\":[25],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"baudRate\":115200,\"rtscts\":false,\"path\":\"/dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_f45097c0c312ef118bf96db8bf9df066-if00-port0\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false}}"'
[2024-11-13 20:28:20] info: 	zh:ember: Using default stack config.
[2024-11-13 20:28:20] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":1321,"extendedPanID":[23,58,119,75,209,77,211,4],"channelList":[25]},"serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_f45097c0c312ef118bf96db8bf9df066-if00-port0","adapter":"ember"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2024-11-13 20:28:20] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-11-13 20:28:20] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-11-13 20:28:20] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-13 20:28:20] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-11-13 20:28:20] debug: 	zh:ember:uart:ash: Opening serial port with {"path":"/dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_f45097c0c312ef118bf96db8bf9df066-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2024-11-13 20:28:20] info: 	zh:ember:uart:ash: Serial port opened
[2024-11-13 20:28:20] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-13 20:28:20] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-11-13 20:28:20] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-11-13 20:28:20] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-11-13 20:28:20] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-11-13 20:28:20] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-11-13 20:28:20] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2024-11-13 20:28:20] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2024-11-13 20:28:21] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2024-11-13 20:28:21] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2024-11-13 20:28:21] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2024-11-13 20:28:21] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2024-11-13 20:28:21] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2024-11-13 20:28:21] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1100/2500
[2024-11-13 20:28:21] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2024-11-13 20:28:21] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
[2024-11-13 20:28:21] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
[2024-11-13 20:28:21] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
[2024-11-13 20:28:22] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
[2024-11-13 20:28:22] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
[2024-11-13 20:28:22] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
[2024-11-13 20:28:22] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
[2024-11-13 20:28:22] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
[2024-11-13 20:28:22] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2100/2500
[2024-11-13 20:28:22] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2200/2500
[2024-11-13 20:28:22] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2300/2500
[2024-11-13 20:28:22] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2400/2500
[2024-11-13 20:28:22] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-13 20:28:22] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-13 20:28:22] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-11-13 20:28:22] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-11-13 20:28:23] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-11-13 20:28:23] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[... cut ...]
    at EmberAdapter.emberSetEzspConfigValue (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1463:24)
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:711:9)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1598:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)

Rainking80, 2024-11-13:
as a note (could be related): i installed a full backup from my pi4 to my p5. first i had issues to find the usb devices, so i had to change the group permissions for the usb port to dialout.

Rainking80, 2024-11-13:
this is what i see with dmesg: ( the port 16 messages pop up when zigbee runs into that error)

[12909.104050] kauditd_printk_skb: 42 callbacks suppressed
[12909.104054] audit: type=1334 audit(1731527637.884:1326): prog-id=268 op=UNLOAD
[12909.142294] audit: type=1325 audit(1731527637.920:1327): table=nat:318 family=2 entries=1 op=nft_unregister_rule pid=363811 subj=unconfined comm="iptables"
[12909.142463] audit: type=1300 audit(1731527637.920:1327): arch=c00000b7 syscall=211 success=yes exit=512 a0=3 a1=7fead204b8 a2=0 a3=1 items=0 ppid=643 pid=363811 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
[12909.142471] audit: type=1327 audit(1731527637.920:1327): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4400444F434B4552002D7000746370002D6400302F30002D2D64706F72740038343835002D6A00444E4154002D2D746F2D64657374696E6174696F6E003137322E33302E33332E31303A383438350000002D690068617373696F
[12909.145484] audit: type=1325 audit(1731527637.924:1328): table=nat:319 family=2 entries=1 op=nft_unregister_rule pid=363813 subj=unconfined comm="iptables"
[12909.145490] audit: type=1300 audit(1731527637.924:1328): arch=c00000b7 syscall=211 success=yes exit=604 a0=3 a1=7fd270ebb8 a2=0 a3=1 items=0 ppid=643 pid=363813 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
[12909.145495] audit: type=1327 audit(1731527637.924:1328): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4400504F5354524F5554494E47002D7000746370002D73003137322E33302E33332E3130002D64003137322E33302E33332E3130002D2D64706F72740038343835002D6A004D415351554552414445
[12909.179281] audit: type=1325 audit(1731527637.956:1329): table=filter:320 family=2 entries=1 op=nft_unregister_rule pid=363815 subj=unconfined comm="iptables"
[12909.179291] audit: type=1300 audit(1731527637.956:1329): arch=c00000b7 syscall=211 success=yes exit=652 a0=3 a1=7fde48f148 a2=0 a3=1 items=0 ppid=643 pid=363815 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
[12909.179296] audit: type=1327 audit(1731527637.956:1329): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D740066696C746572002D4400444F434B45520000002D690068617373696F002D6F0068617373696F002D7000746370002D64003137322E33302E33332E3130002D2D64706F72740038343835002D6A00414343455054
[12909.210479] vethc7268ac: renamed from eth0
[12909.231781] hassio: port 16(vetheee67ec) entered disabled state
[12909.257090] hassio: port 16(vetheee67ec) entered disabled state
[12909.258246] vetheee67ec (unregistering): left allmulticast mode
[12909.258254] vetheee67ec (unregistering): left promiscuous mode
[12909.258266] hassio: port 16(vetheee67ec) entered disabled state
[12909.451103] hassio: port 16(veth49ad628) entered blocking state
[12909.451111] hassio: port 16(veth49ad628) entered disabled state
[12909.451128] veth49ad628: entered allmulticast mode
[12909.451249] veth49ad628: entered promiscuous mode
[12909.451403] hassio: port 16(veth49ad628) entered blocking state
[12909.451408] hassio: port 16(veth49ad628) entered forwarding state
[12909.660079] eth0: renamed from veth784acad

Ricc68, 2024-11-16:
From your logs seems that the dongle stopped responding suddenly at set configuration Seq 14.
The veth port 16 is ethernet, not related to communication with the dongle. It seems ethernet initialization.

I see you are using dongle firmware 7.4.4. Currently I am using dongle fw 8.0.2 and in my case, fingers crossed, seems much more stable (but it may be just temporary luck, I will see if remains stable).

Nerivec, 2024-11-20:
The issue was identified by Rainking80 after a lot of work! It was caused by the HA integration https://www.home-assistant.io/integrations/edl21 which apparently was trying to find a USB device that had been disconnected some time ago. That process was interfering with Z2M and causing any adapter it tried to use to regularly fail.
_If you encounter this issue, also make sure you are not unknowingly restoring said integration from a previous HA backup._

closed 2024-11-20 by Nerivec

nz-ngauruhoe, 2025-01-30:
Hi, I have exactly the same symptoms on a RPI4 with Sonoff Dongle.  Worked like a charm on a RPI3 (but was time to upgrade). I then tried ZHA and the dongle works ok too. So, there's nothing wrong with the dongle or connection. Not a fan of ZHA though.......

I'm not running the EDL21 integration, must be another culprit blocking access. Beats me why tha doesn't bother ZHA.

The solution above only mentions EDL21 but not how that was determined. How do I find out who the blocker is?

Nerivec, 2025-01-30:
That issue was highly specific. Chances are, you have a more common one.
Do you have some `debug` logs?
_Make sure to disable ZHA completely before trying to start Z2M (and anything else that could be using the port, like the flasher addon), otherwise you will definitely get into troubles._

nz-ngauruhoe, 2025-01-30:
I was just adding this then read you answer. Thanks for that!

"BTW, I also found hits on ZHA competing for the stick. I disabled/deleted it first, immediately discovered again by HA, next 'ignore'. Some seem to think that 'ignore' is not enough to leave the stick alone. Hope someone can shed some light on this."

I'll go after the debug logs.......

Nerivec, 2025-01-30:
https://community.home-assistant.io/t/how-to-disable-zha-zigbee-home-automation/553061
https://community.home-assistant.io/t/moving-from-zha-to-z2m-how-to-uninstall-zha-correctly/487871/4
There are many other posts on HA community. Usually, you're good as long as you don't see the "starting ZHA" (in UI/logs) after a full reboot of HA.

nz-ngauruhoe, 2025-01-30:
I added "loglevel: debug" to the Z2M config parms but the log doesn't change. Where should I specify the debug level?

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: true
  **log_level: debug**
mqtt: {}
serial:
  port: /dev/ttyUSB0
  adapter: ember

nz-ngauruhoe, 2025-01-30:
Found it......

[21:45:37] INFO: Preparing to start...
[21:45:37] INFO: Socat not enabled
[21:45:37] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-30 21:45:43] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-16.11-21-42'
[2025-01-30 21:45:43] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-30 21:45:43] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-30 21:45:43] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-30 21:45:43] debug: 	z2m: sd-notify loaded
[2025-01-30 21:45:43] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-30 21:45:43] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":56745,\"extendedPanID\":[103,5,67,38,124,227,58,223],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-30 21:45:43] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyUSB0","manufacturer":"Silicon Labs","serialNumber":"0001","pnpId":"usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","vendorId":"10c4","productId":"ea60"}]
[2025-01-30 21:45:43] debug: 	zh:adapter:discovery: Unable to match USB adapter: ember | /dev/ttyUSB0
[2025-01-30 21:45:43] info: 	zh:ember: Using default stack config.
[2025-01-30 21:45:43] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":56745,"extendedPanID":[103,5,67,38,124,227,58,223],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"ember"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-30 21:45:43] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-30 21:45:43] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-30 21:45:43] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-30 21:45:43] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-01-30 21:45:43] debug: 	zh:ember:uart:ash: Opening serial port with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2025-01-30 21:45:43] info: 	zh:ember:uart:ash: Serial port opened
[2025-01-30 21:45:43] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-30 21:45:43] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2025-01-30 21:45:43] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2025-01-30 21:45:43] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2025-01-30 21:45:44] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2025-01-30 21:45:44] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2025-01-30 21:45:44] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2025-01-30 21:45:44] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2025-01-30 21:45:44] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2025-01-30 21:45:44] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2025-01-30 21:45:44] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2025-01-30 21:45:44] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2025-01-30 21:45:44] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2025-01-30 21:45:44] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1100/2500
[2025-01-30 21:45:45] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2025-01-30 21:45:45] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
[2025-01-30 21:45:45] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
[2025-01-30 21:45:45] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
[2025-01-30 21:45:45] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
[2025-01-30 21:45:45] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
[2025-01-30 21:45:45] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
[2025-01-30 21:45:45] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
[2025-01-30 21:45:45] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
[2025-01-30 21:45:45] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2100/2500
[2025-01-30 21:45:46] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2200/2500
[2025-01-30 21:45:46] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2300/2500
[2025-01-30 21:45:46] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2400/2500
[2025-01-30 21:45:46] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-30 21:45:46] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-30 21:45:46] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2025-01-30 21:45:46] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2025-01-30 21:45:46] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2025-01-30 21:45:46] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2025-01-30 21:45:46] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2025-01-30 21:45:46] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[... cut ...]
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

nz-ngauruhoe, 2025-01-30:
Could [https://github.com/Koenkk/zigbee2mqtt/issues/25037](url) be a solution for this stick as well?

Nerivec, 2025-01-30:
This behavior usually presents in a few scenario:
- your adapter is not the EmberZNet version (Dongle-E), but the zStack version (Dongle-P). _Should use `adapter: zstack` instead of `adapter: ember`_
- you have a mismatch in baudrate between what you have configured in Z2M (default being 115200 if not explicitly specified), and the firmware you flashed on the adapter. _Reflash a 115200 firmware in case you are not sure._
- your adapter's firmware is not working properly. _I think we can exclude this, since it worked in ZHA_
- something else on the system is messing with the USB enough to make it fail to communicate properly, but not enough to just prevent it from opening (like the original EDL21 issue here)

nz-ngauruhoe, 2025-01-31:
About the adapter, with the previous RPI3, it was configured as "adapter: ezsp" and working fine for 3+ years. After moving it to the RPI4 it didn't work and the log showed recommendation to use "adapter: ember", so I tried that. 
I just tried "adapter: zstack" but that results in a loop, producing thousands of  debug log messages. See below.

Last bullit, about something else interfering. It really is a minimal HA setup for my camper. Hardly anything apart from the default ones HA adds automatically. For addons the same. The "batmon", the only not-standard one, is not running. The only thing I could think of here is ZHA but now ruled it out as there is nothing in the log for ZHA.

I'm a bit hesitant flashing the stick because my old RPI3 is my (working) backup and there the stick still works fine, If I flash it, I may loose my working backup. 
But also, it works with ZHA. If it is the baudrate wouldn't that be a problem for ZHA too?

Grateful for your help!

[image]

[image]

[19:09:56] INFO: Preparing to start...
[19:09:56] INFO: Socat not enabled
[19:09:57] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-31 19:10:05] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-30.18-29-09'
[2025-01-31 19:10:05] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-31 19:10:05] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-31 19:10:05] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-31 19:10:05] debug: 	z2m: sd-notify loaded
[2025-01-31 19:10:05] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-31 19:10:05] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":56745,\"extendedPanID\":[103,5,67,38,124,227,58,223],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-31 19:10:05] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyUSB0","manufacturer":"Silicon Labs","serialNumber":"0001","pnpId":"usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","vendorId":"10c4","productId":"ea60"}]
[2025-01-31 19:10:05] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"Silicon Labs","serialNumber":"0001","pnpId":"usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: 4
[2025-01-31 19:10:05] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":56745,"extendedPanID":[103,5,67,38,124,227,58,223],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-31 19:10:05] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-01-31 19:10:05] info: 	zh:zstack:znp: Serialport opened
[2025-01-31 19:10:05] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: <-- [254,2,97,1,89,6,61]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,2,97,1,89,6,61]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
[2025-01-31 19:10:05] debug: 	zh:zstack:znp: <-- SRSP: SYS - ping - {"capabilities":1625}
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-31 19:10:05] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: <-- [254,2,97,1,89,6,61]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,2,97,1,89,6,61]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
[2025-01-31 19:10:05] debug: 	zh:zstack:znp: <-- SRSP: SYS - ping - {"capabilities":1625}
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-31 19:10:05] debug: 	zh:zstack:znp: --> SREQ: SYS - version - {}
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,2,35]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: <-- [254,10,97,2,2,1,2,7,1,20,100,52,1,0,43]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,10,97,2,2,1,2,7,1,20,100,52,1,0,43]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: --> parsed 10 - 3 - 1 - 2 - [2,1,2,7,1,20,100,52,1,0] - 43
[2025-01-31 19:10:05] debug: 	zh:zstack:znp: <-- SRSP: SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20210708}
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-31 19:10:05] debug: 	zh:zstack: Adapter concurrent: 16
[2025-01-31 19:10:05] debug: 	zh:zstack: Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20210708})
[2025-01-31 19:10:05] debug: 	zh:adapter:zstack:manager: beginning znp startup
[... cut ...]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:writer: --> frame [254,2,33,19,33,0,17]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: <-- [254,2,97,19,116,0,4]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,2,97,19,116,0,4]
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 19 - [116,0] - 4
[2025-01-31 19:10:05] debug: 	zh:zstack:znp: <-- SRSP: SYS - osalNvLength - {"length":116}
[2025-01-31 19:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext []

===> repeating forever.........

nz-ngauruhoe, 2025-01-31:
I also have a conbee II stick laying around, I used that some time ago on my 'home' HArpi I think with deconz integration. Minimal zigbee setup, didn't like it very much. I find hits that this stick will work with Z2M as well? Same/similar functionality as sonoff? If yes I will give that a try.

Nerivec, 2025-01-31:
Ok, so it's definitely `adapter: ember` if you used `ezsp` before and that works, it means you have the Dongle-E version.
I'm not sure how the setup works for ZHA, so, it might auto-detect the baudrate.
If you don't know what baudrate, nor what firmware you have, you really need to flash it with a "known firmware" so we remove all possibilities of misconfiguration. Flashing any 7.4.x firmware should not be a problem if you ever want to go back to that old setup you had (`ezsp` is compatible with 7.4.x and below, whereas `ember` is compatible with 7.4.x and above).
https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl

nz-ngauruhoe, 2025-01-31:
Ok, I will try flashing. In the mean time I added the Conbee II and paired an Ikea switch to try. Works like a charm.

nz-ngauruhoe, 2025-02-01:
Hi, I realized today that with the ConbeeII stick I changed to a USB2.0 port. So I tried the Sonoff again with a USB2.0 port and it works! Had not flashed the sonoff yet.

Thanks a lot for you help and patience!