Koenkk/zigbee2mqtt #24727: Aqara ZNCLBL01LM - Value 3 not found

and7ey, 2024-11-11:
### What happened?

[Aqara ZNCLBL01LM](https://www.zigbee2mqtt.io/devices/ZNCLBL01LM.html#aqara-znclbl01lm) sometimes returns unknown value 3 when moving and suddenly stops (motor fault?)

### What did you expect to happen?

The error to be handled by converter

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

Sonoff Dongle CC2652P

### Setup

HAOS with z2m addon

### Debug log

info 2024-11-11 20:00:15z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_driver_hall', payload '{"action":null,"battery":100,"charging":false,"device_temperature":28,"hand_open":null,"hooks_lock":"LOCK","hooks_state":"locked","illuminance_lux":0,"linkquality":145,"motor_state":"opening","position":100,"power_outage_count":0,"power_source":"battery","running":true,"state":"CLOSE","target_position":100,"update":{"installed_version":27,"latest_version":6168,"state":"available"},"update_available":null,"voltage":3000}'

error 2024-11-11 20:00:15z2m: Exception while calling fromZigbee converter: Value: '3' not found in: [0, 1, 2]}

info 2024-11-11 20:00:21z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_driver_hall', payload '{"action":null,"battery":100,"charging":false,"device_temperature":28,"hand_open":null,"hooks_lock":"LOCK","hooks_state":"locked","illuminance_lux":0,"linkquality":145,"motor_state":"stopped","position":100,"power_outage_count":0,"power_source":"battery","running":false,"state":"CLOSE","target_position":100,"update":{"installed_version":27,"latest_version":6168,"state":"available"},"update_available":null,"voltage":3000}'

doddyth, 2024-12-31:
is there any update regarding this?

doddyth, 2024-12-31:
Just want to add another info for my case. I've updated the zigbee2mqtt to 1.42.0 and tried to readded the device multiple times but it is still happening.

alicankaralar, 2025-01-25:
I have problems with this device too, hand_open value can't be changed. I tried a custom converter to change it but although i can persist it in the state with a custom external converter (curtainHandOpen on Zigbee, and hand_open in mqtt), that does not do anything on the device. I remember it worked before, somehow it glitched. Also firmware/ota situation seems out of sync too.

yozsever, 2025-04-01:
> I have problems with this device too, hand_open value can't be changed. I tried a custom converter to change it but although i can persist it in the state with a custom external converter (curtainHandOpen on Zigbee, and hand_open in mqtt), that does not do anything on the device. Tried broadcasting to the device via mqtt broker directly. For a long while hand_open value was null. After converters, it's either true or false. But does not do anything to the device.
> 
> I remember it worked before, somehow it glitched. Also firmware/ota situation seems out of sync too - the firmware on the device is newer than the latest firmware in z2mqtt index.json. Using HA with z2mqtt addon

Hello I also have the same issue, hand_open can not be changed. Also when invert_cover is set to true, the state does not show correctly. It shows as the curtain is closed when the cover is actually open. This affects the control of the curtain, for example with a button press to open/close the curtain. 

Any updates regarding the issue? Thanks

closed 2025-06-09 by github-actions[bot]