Koenkk/zigbee2mqtt #24736: [Wrong device]: Livarno Lux (Lidl) HG08131C

RubenKelevra, 2024-11-12:
### Link

https://www.lidl.de/p/livarno-home-leuchtmittel-rgb-zigbee-smart-home/p100348095001

### Model

HG08131C

### Description

E27 RGBCT Light bulb

### Vendor

Livarno Lux

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/HG07834C.png

### Database entry

no idea how to access this, but this is the zigbee vendor and model: Zigbee Model: TS0505B Zigbee Manufacturer: _TZ3000_qd7hej8u

### Notes

Image is fine, but the light is misidentified as [HG07834C](https://www.zigbee2mqtt.io/devices/HG07834C.html#lidl-hg07834c). So the definition for this Model/Vendor Zigbee string needs to be extended to include the HG08131C.

Koenkk, 2024-11-12:
Could you make a PR to update https://github.com/Koenkk/zigbee-herdsman-converters/blob/ebfb2d55d31fffc3c72e075bdc0871a9c2a29fa8/src/devices/tuya.ts#L1622 ?

RubenKelevra, 2024-11-12:
Weird Github search didn't returned any result for those misused vendor strings as device identifications. Otherwise I would have already.

Thanks for the pointer, will do.

closed 2024-11-12 by RubenKelevra