Koenkk/zigbee2mqtt #24741: [New device support]: _TZE204_lpedvtvr

schnm00, 2024-11-12:
### Link

https://sl.aliexpress.ru/p?key=TXodrBx

### Database entry

{"id":32,"type":"Router","ieeeAddr":"0xa4c138871b22ed05","nwkAddr":49540,"manufId":4417,"manufName":"_TZE204_lpedvtvr","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"ϰ�.\u0013�\u0000�.e�\u0000�.\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1731402704557}

### Zigbee2MQTT version

1.41.0

### Comments

I have not been able to implement the external definition

Detailed info:
<html>
<body>
<span style="color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; display: inline !important; float: none;">Address: A4C13815F7B52427/1E16</span><br style="box-sizing: border-box; color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><br style="box-sizing: border-box; color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><b style="box-sizing: border-box; font-weight: var(--base-text-weight-semibold, 600); color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Structure</b><br style="box-sizing: border-box; color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><ul dir="auto" style="box-sizing: border-box; padding-left: 2em; margin-top: 0px; margin-bottom: var(--base-size-16); color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><li style="box-sizing: border-box;">Endpoint 1 (SMART_PLUG)<ul dir="auto" style="box-sizing: border-box; padding-left: 2em; margin-top: 0px; margin-bottom: 0px;"><li style="box-sizing: border-box;">Input<ul dir="auto" style="box-sizing: border-box; padding-left: 2em; margin-top: 0px; margin-bottom: 0px;"><li style="box-sizing: border-box;">EF00_ManufacturerSpecific<ul dir="auto" style="box-sizing: border-box; padding-left: 2em; margin-top: 0px; margin-bottom: 0px;"><li style="box-sizing: border-box;">0000_Custom: 0 [UNSIGNED_8_BIT_INTEGER]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0001_Custom: false [TUYA_BOOLEAN]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0002_Custom: 0 [TUYA_ENUMERATION_8_BIT]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0010_Custom: 225 [TUYA_32_BIT_INTEGER]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0012_Custom: 50 [TUYA_32_BIT_INTEGER]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0020_Custom: 1 [TUYA_ENUMERATION_8_BIT]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0022_Custom: 450 [TUYA_32_BIT_INTEGER]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0027_Custom: false [TUYA_BOOLEAN]</li><li style="box-sizing: border-box; margin-top: 0.25em;">002F_Custom: 1 [TUYA_ENUMERATION_8_BIT]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0030_Custom: 15 [TUYA_32_BIT_INTEGER]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0032_Custom: 245 [TUYA_32_BIT_INTEGER]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0065_Custom: -6 [TUYA_32_BIT_INTEGER]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0066_Custom: ByteArray [value=01 06 00 00 C8 0B 1E 00 C8 0D 1E 00 C8 11 1E 00 C8] [TUYA_RAW_OCTET]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0067_Custom: ByteArray [value=02 06 00 00 C8 0B 1E 00 C8 0D 1E 00 C8 11 1E 00 C8] [TUYA_RAW_OCTET]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0068_Custom: ByteArray [value=03 06 00 00 C8 0B 1E 00 C8 0D 1E 00 C8 11 1E 00 C8] [TUYA_RAW_OCTET]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0069_Custom: ByteArray [value=04 06 00 00 C8 0B 1E 00 C8 0D 1E 00 C8 11 1E 00 C8] [TUYA_RAW_OCTET]</li><li style="box-sizing: border-box; margin-top: 0.25em;">006A_Custom: ByteArray [value=05 06 00 00 C8 0B 1E 00 C8 0D 1E 00 C8 11 1E 00 C8] [TUYA_RAW_OCTET]</li><li style="box-sizing: border-box; margin-top: 0.25em;">006B_Custom: ByteArray [value=06 06 00 00 C8 0B 1E 00 C8 0D 1E 00 C8 11 1E 00 C8] [TUYA_RAW_OCTET]</li><li style="box-sizing: border-box; margin-top: 0.25em;">006C_Custom: ByteArray [value=07 06 00 00 C8 0B 1E 00 C8 0D 1E 00 C8 11 1E 00 C8] [TUYA_RAW_OCTET]</li><li style="box-sizing: border-box; margin-top: 0.25em;">006D_Custom: 255 [TUYA_32_BIT_INTEGER]</li><li style="box-sizing: border-box; margin-top: 0.25em;">006E_Custom: 5 [TUYA_32_BIT_INTEGER]</li><li style="box-sizing: border-box; margin-top: 0.25em;">006F_Custom: 450 [TUYA_32_BIT_INTEGER]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0070_Custom: 0 [TUYA_32_BIT_INTEGER]</li><li style="box-sizing: border-box; margin-top: 0.25em;">0071_Custom: 300 [TUYA_32_BIT_INTEGER]</li><li style="box-sizing: border-box; margin-top: 0.25em;">00 [... cut ...] ne !important; float: none;">Last update: Sun Nov 10 17:24:09 MSK 2024</span>
</body>
</html>

MOES ZigBee Smart Knob Thermostat - Star Ring Series:

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_lpedvtvr',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Paired, but not supported device

rebe94, 2024-11-25:
this thermostat looks great and seems to be good quality. Would be really nice to have some converter for that one 👍
[image]

alexsudakov, 2024-11-26:
+1 please add support:)

alexsudakov, 2024-12-03:
А есть понимание, с какой скоростью запросы в работу попадают? :)

magic1447, 2024-12-07:
Hello! Im also very interested in support for this device!

nholloh, 2024-12-14:
So I started on this, but given this is my first time _ever_ creating a custom ZHA quirk, I am struggling quite a bit. I was able to make sense of endpoints and clusters and already put some work into the attribute ids. However, The following things are not working yet:

1. The time is not automatically configured.
2. The occupied heating setpoint cannot be modified through HA; or more specifically is not reflecting on the thermostat.
3. The enabled mode is showing correctly, but updates are not reflecting on the thermostat.
4. I cannot see or control the child lock mode.
5. I am unsure about how to program the thermostat with a schedule.

Any help by a veteran zha quirk creator is greatly appreciated.

<details><summary>Current code</summary>
<p>

```py
"""Map from manufacturer to standard clusters for electric heating thermostats."""
import logging

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaThermostat,
    TuyaThermostatCluster,
    TuyaUserInterfaceCluster,
)

# info from https://github.com/zigpy/zha-device-handlers/pull/538#issuecomment-723334124
# https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L239
# and https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/common.js#L113
MOESBHT_TARGET_TEMP_ATTR = 0x0232  # [0,0,0,210] target room temp (decidegree)
MOESBHT_TEMPERATURE_ATTR = 0x0210  # [0,0,0,200] current room temp (decidegree)
#MOESBHT_SCHEDULE_MODE_ATTR = 0x0403  # [1] false [0] true   /!\ inverted
MOESBHT_MANUAL_MODE_ATTR = 0x0402  # [2] auto OR schedule, [1] temp_manual, [0] manual /!\ inverted
MOESBHT_ENABLED_ATTR = 0x0101  # [0] off [1] on
MOESBHT_RUNNING_MODE_ATTR = 0x042f  # [1] idle [0] heating /!\ inverted
MOESBHT_CHILD_LOCK_ATTR = 0x0127  # [0] unlocked [1] child-locked

_LOGGER = logging.getLogger(__name__)

class MoesBHTManufClusterTest(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of some electric heating thermostats."""

    attributes = {
        MOESBHT_TARGET_TEMP_ATTR: ("target_temperature", t.uint32_t, True),
        MOESBHT_TEMPERATURE_ATTR: ("temperature", t.uint32_t, True),
        MOESBHT_MANUAL_MODE_ATTR: ("manual_mode", t.uint8_t, True),
        MOESBHT_ENABLED_ATTR: ("enabled", t.uint8_t, True),
        MOESBHT_RUNNING_MODE_ATTR: ("running_mode", t.uint8_t, True),
        MOESBHT_CHILD_LOCK_ATTR: ("child_lock", t.uint8_t, True),
    }

    def _update_attribute(self, attrid, value):
        _LOGGER.info("### Updating attribute: attrid=%s, value=%s", attrid, value)
        _LOGGER.info("### Target temp %s", MOESBHT_TARGET_TEMP_ATTR)
        super()._update_attribute(attrid, value)
        if attrid == MOESBHT_TARGET_TEMP_ATTR:
            _LOGGER.info("### Updating attribute target temp: attrid=%s, value=%s", attrid, value)
            self.endpoint.device.thermostat_bus.listener_event(
                "temperature_change",
                "occupied_heating_setpoint",
[... 112 lines cut ...]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesBHTManufClusterTest,
                    MoesBHTThermostatTest,
                    MoesBHTUserInterfaceTest,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

</p>
</details>

archiespb, 2025-01-01:
+1!
Beautiful device must have support in z2m!

rebe94, 2025-01-02:
Recently someone has created external definition for this thermostat here:
https://github.com/Koenkk/zigbee2mqtt/issues/25038

I used it and it worked so definitely recommend it!

alexsudakov, 2025-01-29:
up

closed 2025-04-08 by github-actions[bot]