Koenkk/zigbee2mqtt #24742: [New device support]: Avatto Me168 new manufacturer id

K2ProjectImmo, 2024-11-12:
### Link

https://de.aliexpress.com/item/1005003952150617.html?spm=a2g0o.order_list.order_list_main.5.544d5c5f3oahGL&gatewayAdapt=glo2deu

### Database entry

{"id": 10, "type": "EndDevice", "ieeeAddr": "0x04cd152f10792222", "nwkAddr": 48191, "manufId": 441, "manufName": "lZKEA jakdfga", "powerSource": "Battery", "modelId": "TS0601", "epList": [1], "endpoints": [{"profId": 260, "epId": 1, "devId": 81, "inClusterList": [4, 5, 61184, 64529], "outClusterList": [25, 10], "clusters": {"genBasic": {"attributes": {"65281": 14980, "65282": {}, "65283": "some_value"}}}}, "appVersion": 77, "stackVersion": 3, "hwVersion": 1, "binds": [], "configuredReportings": [], "meta": {}}},"appVersion": 77, "stackVersion": 0, "hwVersion": 1, "dateCode": "", "zclVersion": 3, "interviewCompleted": true, "meta": {}, "lastSeen": 1731449781385

### Zigbee2MQTT version

1.41.0

### Comments

The Avatoo Me167 thermostat is no longer supported. Its manufacturer ID was changed from _TZE200_p3dbf6qs to _TZE284_p3dbf6qs, which Zigbee2mqtt does not recognize as a supported device. Working with an external converter also didn´t work.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_p3dbf6qs',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I´ve tried using the external converter definition from twhittock, but I guess it is not supported anymore. There are no exposes or other information when the device is added. Also, there is an error loading the external definition in the logs.

Koenkk, 2024-11-13:
Can you try to 

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE284_p3dbf6qs ` to `_TZE200_p3dbf6qs `
- Start Z2M and see if the device now works

K2ProjectImmo, 2024-11-13:
Thank you so much, I´ve tried it and it works perfectly! This saved me so much time.

Koenkk, 2024-11-14:
Added, you can change it back after the next z2m release

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-11-14 by Koenkk

dav71701home, 2024-11-18:
I have the same problem with:

> const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_o3x45p96',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

Is this also included in release 20.47.0?

mmuellr, 2024-11-18:
I had the same problem like @dav71701home 
I just got some TRV06 from here: https://aliexpress.com/item/1005005950102932.html

It's recogniced as:
```
Zigbee Model TS0601
Zigbee Manufacturer _TZE284_o3x45p96
Support status Not supported
```

I did this (but with file /homeassistant/zigbee2mqtt/database.db) 
I did what @Koenkk mentioned:
1. Stopped Z2M, 
2. replaced all occurances of _TZE284_o3x45p96 by _TZE200_p3dbf6qs in /homeassistant/zigbee2mqtt/database.db
3. Start Z2M

And it works for me

madzur, 2025-05-06:
I have the same exact issue with my device as @mmuellr.
When manually editing the DB my device works flawlessly. Upon Z2M restart, the device shows up as unsupported.
My device shows the following zigbee manufacturer: _TZE200_4aijvczq

Would be great if this ID could be added to the supported devices.
Thank you.

mmuellr, 2025-05-07:
You have to add `_TZE200_4aijvczq` to `src/devices/tuya.ts` as you can see [in this MR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8339). 
When you do this, the device will work permanently after the next release

madzur, 2025-05-11:
Thanks, it took a while for me to figure out how to submit a pull request, and it seems to have worked. Pleasure working with you :).

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9304