Koenkk/zigbee2mqtt #24743: [New device support]: Tuya 6 gang smart switch

simonardejr, 2024-11-12:
### Link

https://www.novadigitalsmart.com.br/produtos/interruptor-inteligente-zigbee-tecla-fisica-4x4-fzb-4-6

### Database entry

{"id":10,"type":"Router","ieeeAddr":"0xa4c138fea7a692a4","nwkAddr":16892,"manufId":4417,"manufName":"_TZE204_g4au0afs","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�i�.i�i�.i�i�.i�i�.i�i�.i�i�.i�i�.i�i�.i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74,"modelId":"TS0601","manufacturerName":"_TZE204_g4au0afs","powerSource":1,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1731454461297}

### Zigbee2MQTT version

1.41.0

### Comments

Hey everyone! I've bought this 6-gang switch, but can't make it work... I've searched the issues and found one that looks
similar to mine: https://github.com/Koenkk/zigbee2mqtt/issues/21225

It looks like this one: https://www.zigbee2mqtt.io/devices/TS0601_switch_6_gang.html

Can you guys help me with this? Thanks in advance!

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_g4au0afs',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

It doesn't show the switches

simonardejr, 2024-11-13:
> Can you try to
> 
>     * Stop Z2M
> 
>     * Open `data/database.db` with a text editor and change all occurrences of `_TZE204_g4au0afs ` to `_TZE200_mwvfvw8g`
> 
>     * Start Z2M and see if the device now works

Yeah! This worked! Now the device is identified by z2m and the switches are functioning properly!

Thanks!! I think we can close this 🎉

Koenkk, 2024-11-14:
Added, you can change it back after the next z2m release!

closed 2024-11-14 by Koenkk

gersonfer, 2025-01-14:
Same switch model, same Z2M version.

I did the same :
* Stop Z2M
* Open `data/database.db` with a text editor and change all occurrences of `_TZE204_g4au0afs` to`_TZE200_mwvfvw8g`
* Start Z2M

It works but when I stop Z2M,  the database.db is changed to  `_TZE204_g4au0afs ` again.